TrunkRecorder/trunk-recorder #346: home/pi/.gnuradio/prefs/vmcircbuf_default_factory: No such file or directory

rwcts, 2020-05-02:
pi@raspberrypi:~/trunk-recorder $ ./recorder
[2020-05-02 17:22:01.331092] (info)   Using Config file: ./config.json

[2020-05-02 17:22:01.332096] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------

[2020-05-02 17:22:01.332154] (info)   
-------------------------------------
SYSTEMS
-------------------------------------

[2020-05-02 17:22:01.332289] (info)   

System Number: 1
-------------------------------------

[2020-05-02 17:22:01.332337] (info)   Short Name: mySystemName
[2020-05-02 17:22:01.332383] (info)   System Type: p25
[2020-05-02 17:22:01.332415] (info)   Control Channels: 
[2020-05-02 17:22:01.332543] (info)     8.514250e+08
[2020-05-02 17:22:01.332633] (info)     8.514875e+08
[2020-05-02 17:22:01.332676] (info)     8.519750e+08
[2020-05-02 17:22:01.332714] (info)   API Key: 
[2020-05-02 17:22:01.332742] (info)   Broadcastify API Key: 2a2109we1-8r324-11fda-bcsab-fdsec
[2020-05-02 17:22:01.332781] (info)   Broadcastify Calls System ID: 99
[2020-05-02 17:22:01.332808] (info)   Upload Script: 
[2020-05-02 17:22:01.332833] (info)   Call Log: true
[2020-05-02 17:22:01.332883] (info)   Audio Archive: false
[2020-05-02 17:22:01.332910] (info)   Loading Talkgroups...
[2020-05-02 17:22:01.332936] (info)   Talkgroups File: 
[2020-05-02 17:22:01.332961] (info)   Loading Unit Tags...
[2020-05-02 17:22:01.332985] (info)   Unit Tags File: 
[2020-05-02 17:22:01.333010] (info)   Record Unknown Talkgroups: true
[2020-05-02 17:22:01.333036] (info)   Decode MDC: false
[2020-05-02 17:22:01.333062] (info)   Decode FSync: false
[2020-05-02 17:22:01.333087] (info)   Decode Star: false
[2020-05-02 17:22:01.333113] (info)   Decode TPS: false
[2020-05-02 17:22:01.333141] (info)   Talkgroup Display Format: Id
[2020-05-02 17:22:01.333173] (info)   Hide Encrypted Talkgroups: false
[2020-05-02 17:22:01.333200] (info)   Hide Unknown Talkgroups: false
[2020-05-02 17:22:01.333225] (info)   Minimum Call Duration (in seconds): 0
[2020-05-02 17:22:01.333255] (info)   
[2020-05-02 17:22:01.333284] (info)   

-------------------------------------
SOURCES
-------------------------------------

[2020-05-02 17:22:01.333365] (info)   Driver: osmosdr
[2020-05-02 17:22:01.333392] (info)   Center: 8.516500e+08
[2020-05-02 17:22:01.333435] (info)   Rate: 2048000
[2020-05-02 17:22:01.333478] (info)   Error: 0
[2020-05-02 17:22:01.333508] (info)   PPM Error: 69
[2020-05-02 17:22:01.333543] (info)   Gain: 35
[2020-05-02 17:22:01.333578] (info)   IF Gain: 0
[2020-05-02 17:22:01.333607] (info)   BB Gain: 0
[2020-05-02 17:22:01.333636] (info)   LNA Gain: 0
[2020-05-02 17:22:01.333664] (info)   PGA Gain: 0
[2020-05-02 17:22:01.333692] (info)   TIA Gain: 0
[2020-05-02 17:22:01.333720] (info)   MIX Gain: 0
[2020-05-02 17:22:01.333748] (info)   VGA1 Gain: 0
[2020-05-02 17:22:01.333776] (info)   VGA2 Gain: 0
[2020-05-02 17:22:01.333804] (info)   Squelch: 0
[2020-05-02 17:22:01.333831] (info)   Idle Silence: false
[2020-05-02 17:22:01.333857] (info)   Digital Recorders: 4
[2020-05-02 17:22:01.333888] (info)   Debug Recorder: false
[2020-05-02 17:22:01.333914] (info)   SigMF Recorders: 0
[2020-05-02 17:22:01.333940] (info)   Analog Recorders: 0
[... 92 lines cut ...]
[2020-05-02 17:24:26.000905] (error)   [mySystemName]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-05-02 17:24:29.008228] (error)   [mySystemName] Retuning to Control Channel: 8.514875e+08
[2020-05-02 17:24:29.008451] (info)   	 - System Source 0 - Min Freq: 8.506900e+08 Max Freq: 8.526100e+08
[2020-05-02 17:24:29.008668] (error)   [mySystemName]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-05-02 17:24:32.006916] (error)   [mySystemName] Retuning to Control Channel: 8.519750e+08
[2020-05-02 17:24:32.007143] (info)   	 - System Source 0 - Min Freq: 8.506900e+08 Max Freq: 8.526100e+08
[2020-05-02 17:24:32.007361] (error)   [mySystemName]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-05-02 17:24:35.001362] (error)   [mySystemName] Retuning to Control Channel: 8.514250e+08
[2020-05-02 17:24:35.001528] (info)   	 - System Source 0 - Min Freq: 8.506900e+08 Max Freq: 8.526100e+08
[2020-05-02 17:24:35.001640] (error)   [mySystemName]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-05-02 17:24:38.003834] (error)   [mySystemName] Retuning to Control Channel: 8.514875e+08
[2020-05-02 17:24:38.004013] (info)   	 - System Source 0 - Min Freq: 8.506900e+08 Max Freq: 8.526100e+08
[2020-05-02 17:24:38.004150] (error)   [mySystemName]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-05-02 17:24:41.003433] (error)   [mySystemName] Retuning to Control Channel: 8.519750e+08
[2020-05-02 17:24:41.003654] (info)   	 - System Source 0 - Min Freq: 8.506900e+08 Max Freq: 8.526100e+08
[2020-05-02 17:24:41.003899] (error)   [mySystemName]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-05-02 17:24:44.008962] (error)   [mySystemName] Retuning to Control Channel: 8.514250e+08
[2020-05-02 17:24:44.009337] (info)   	 - System Source 0 - Min Freq: 8.506900e+08 Max Freq: 8.526100e+08
[2020-05-02 17:24:44.009571] (error)   [mySystemName]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-05-02 17:24:47.005320] (error)   [mySystemName] Retuning to Control Channel: 8.514875e+08
[2020-05-02 17:24:47.005482] (info)   	 - System Source 0 - Min Freq: 8.506900e+08 Max Freq: 8.526100e+08
[2020-05-02 17:24:47.005595] (error)   [mySystemName]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-05-02 17:24:50.009910] (error)   [mySystemName] Retuning to Control Channel: 8.519750e+08
[2020-05-02 17:24:50.010093] (info)   	 - System Source 0 - Min Freq: 8.506900e+08 Max Freq: 8.526100e+08
[2020-05-02 17:24:50.010291] (error)   [mySystemName

rwcts, 2020-05-02:
I was able to find a line to run to change the buffer size however its still networking and giving the errors.

sudo sh -c 'echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb'

rwcts, 2020-05-05:
I have used some known good configuration on the Raspberrypi 4B and now I am getting further. It is now productive wav files! in the proper directory. But it is not uploading as it says it cannot open the file. I can play the wav files without problems. I assume this is due to some issues I want into with fskaac. anyone have thoughts on this? Below is the new sample output with the new config.json file but your will notice the error "[2020-05-04 22:59:11.458169] (info)   Error opening file /home/pi/trunk-recorder/GoodhueMore/2020/5/4/21001-1588651135_853100000.m4a" as the files being saved in that directory are .wav files and an associated .json file.

[2020-05-04 22:58:38.002123] (info)   Currently Active Calls: 0
[2020-05-04 22:58:38.002235] (info)   Recorders: 
[2020-05-04 22:58:38.002279] (info)   [ rtl=00,buflen=65536 ]  
[2020-05-04 22:58:38.002320] (info)   [ D0 ] State: inactive
[2020-05-04 22:58:38.002365] (info)   [ D1 ] State: inactive
[2020-05-04 22:58:38.002407] (info)   [ D2 ] State: inactive
[2020-05-04 22:58:38.002444] (info)   [ D3 ] State: inactive
[2020-05-04 22:58:55.895578] (info)   [GoodhueMore]	TG:      21001	Freq: 8.531000e+08	TG not in Talkgroup File 
[2020-05-04 22:58:55.895761] (info)   	- Starting P25 Recorder Num [0]	TG:      21001	Freq: 8.531000e+08 	TDMA: false	Slot: 0
[2020-05-04 22:58:55.896920] (info)   [GoodhueMore]	TG:      21001	Freq: 8.531000e+08	Starting Recorder on Src: rtl=00,buflen=65536
[2020-05-04 22:59:00.862346] (info)   [GoodhueMore]	TG:      21001	Freq: 8.529000e+08	Update Retuning - New Freq: 8.529000e+08	Elapsed: 5s 	Since update: 1s
[2020-05-04 22:59:11.004273] (info)   [GoodhueMore]	TG:      21001	Freq: 8.529000e+08	Ending Recorded Call - Last Update: 4s	Call Elapsed: 16
[2020-05-04 22:59:11.005398] (info)   	- Stopping P25 Recorder Num [0]	TG:      21001	Freq: 8.529000e+08 	TDMA: false	Slot: 0
sh: 1: fdkaac: not found
[2020-05-04 22:59:11.458169] (info)   Error opening file /home/pi/trunk-recorder/GoodhueMore/2020/5/4/21001-1588651135_853100000.m4a
[2020-05-04 22:59:11.458259] (error)   [GoodhueMore]	TG: 21001	Freq: 8.529000e+08	Broadcastify Audio Upload Error: Failed to open/read local data from file/application
[2020-05-04 22:59:24.114097] (info)   [GoodhueMore]	TG:       4304	Freq: 8.537000e+08	TG not in Talkgroup File 
[2020-05-04 22:59:24.114236] (info)   	- Starting P25 Recorder Num [0]	TG:       4304	Freq: 8.537000e+08 	TDMA: false	Slot: 0
[2020-05-04 22:59:24.114817] (info)   [GoodhueMore]	TG:       4304	Freq: 8.537000e+08	Starting Recorder on Src: rtl=00,buflen=65536
[2020-05-04 22:59:37.685346] (info)   [GoodhueMore]	TG:       4304	Freq: 8.527500e+08	Update Retuning - New Freq: 8.527500e+08	Elapsed: 13s 	Since update: 1s
[2020-05-04 22:59:44.009857] (info)   [GoodhueMore]	TG:       4304	Freq: 8.527500e+08	Ending Recorded Call - Last Update: 4s	Call Elapsed: 20
[2020-05-04 22:59:44.011972] (info)   	- Stopping P25 Recorder Num [0]	TG:       4304	Freq: 8.527500e+08 	TDMA: false	Slot: 0
sh: 1: fdkaac: not found
[2020-05-04 22:59:44.515323] (info)   Error opening file /home/pi/trunk-recorder/GoodhueMore/2020/5/4/4304-1588651164_853700000.m4a
[2020-05-04 22:59:44.515602] (error)   [GoodhueMore]	TG: 4304	Freq: 8.527500e+08	Broadcastify Audio Upload Error: Failed to open/read local data from file/application
[2020-05-04 22:59:52.397332] (info)   [GoodhueMore]	TG:      20476	Freq: 8.534000e+08	TG not in Talkgroup File 
[2020-05-04 22:59:52.397473] (info)   	- Starting P25 Recorder Num [0]	TG:      20476	Freq: 8.534000e+08 	TDMA: false	Slot: 0
[2020-05-04 22:59:52.397939] (info)   [GoodhueMore]	TG:      20476	Freq: 8.534000e+08	Starting Recorder on Src: rtl=00,buflen=65536
[2020-05-04 23:00:06.344352] (info)   [GoodhueMore]	TG:      20476	Freq: 8.531000e+08	Update Retuning - New Freq: 8.531000e+08	Elapsed: 14s 	Since update: 1s

hbrookeman, 2020-05-05:
make sure you have the following dependencies installed: libssl-dev libcurl4-openssl-dev fdkaac sox. The broadcastify calls system page outlines what you'll need

rwcts, 2020-05-05:
They are all installed with the exception of fdkaac of which I tried to make work however that may be the sole cause. I tried to "build" it but I could not find anything for Raspberrypi ARM with it.  I see very clearly on the braodcastify site in yellow highlight that it has to be done which is why I believe that may be the issue.  I used libfaac-dev (recalling from memory) and one other. I could not find any documentation on fdkaac for Raspberrypi that worked and what was available was very minimal.

hbrookeman, 2020-05-05:
Check the build info thats a few messages up here: https://gitter.im/trunk-recorder/Lobby

I believe this is what I followed to build on Raspberry pi

gofaster, 2020-05-05:
Build the fdkaac library and command line encoder from source:

[https://gist.github.com/gofaster/7581bc949ed4b789e89ab15815ec4676](url)

rwcts, 2020-05-06:
Thank you to the both of you. I will attempt to drudge through both links and report back.

rwcts, 2020-05-06:
I looked at both links but the one the fit the issue was hbrookeman. One of the git listings i think its wrong however I went past that but still having issues with those commands. here is the output of where the issues arose. All looked good with the instructions until I came to the /fdk-aac-cli-enc section. Toward the bottom is the issue that came up. It would not allow me to "make"

```
  CXX      libSYS/src/genericStds.lo
  CXX      libSYS/src/syslib_channelMapDescr.lo
  GEN      libfdk-aac.la
ar: `u' modifier ignored since `D' is the default (see `U')
pi@raspberrypi:~/fdk-aac $ sudo make install
make[1]: Entering directory '/home/pi/fdk-aac'
 /bin/mkdir -p '/usr/local/lib'
 /bin/bash ./libtool   --mode=install /usr/bin/install -c   libfdk-aac.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libfdk-aac.so.2.0.1 /usr/local/lib/libfdk-aac.so.2.0.1
libtool: install: (cd /usr/local/lib && { ln -s -f libfdk-aac.so.2.0.1 libfdk-aac.so.2 || { rm -f libfdk-aac.so.2 && ln -s libfdk-aac.so.2.0.1 libfdk-aac.so.2; }; })
libtool: install: (cd /usr/local/lib && { ln -s -f libfdk-aac.so.2.0.1 libfdk-aac.so || { rm -f libfdk-aac.so && ln -s libfdk-aac.so.2.0.1 libfdk-aac.so; }; })
libtool: install: /usr/bin/install -c .libs/libfdk-aac.lai /usr/local/lib/libfdk-aac.la
libtool: install: /usr/bin/install -c .libs/libfdk-aac.a /usr/local/lib/libfdk-aac.a
libtool: install: chmod 644 /usr/local/lib/libfdk-aac.a
libtool: install: ranlib /usr/local/lib/libfdk-aac.a
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /usr/local/lib
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/local/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
 /bin/mkdir -p '/usr/local/include/fdk-aac'
 /usr/bin/install -c -m 644 ./libSYS/include/machine_type.h ./libSYS/include/genericStds.h ./libSYS/include/FDK_audio.h ./libSYS/include/syslib_channelMapDescr.h ./libAACenc/include/aacenc_lib.h ./libAACdec/include/aacdecoder_lib.h '/usr/local/include/fdk-aac'
 /bin/mkdir -p '/usr/local/lib/pkgconfig'
 /usr/bin/install -c -m 644 fdk-aac.pc '/usr/local/lib/pkgconfig'
make[1]: Leaving directory '/home/pi/fdk-aac'
pi@raspberrypi:~/fdk-aac $ 
pi@raspberrypi:~/fdk-aac $ cd ~/fdk-aac-cli-enc
bash: cd: /home/pi/fdk-aac-cli-enc: No such file or directory
pi@raspberrypi:~/fdk-aac $ cd ..
pi@raspberrypi:~ $ ls
Desktop    Downloads  fdk-aac-cl-enc  Music     Public     trunk-recorder
Documents  fdk-aac    MagPi           Pictures  Templates  Videos
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ cd ~/fdk-aac-cli-enc
pi@raspberrypi:~/fdk-aac-cli-enc $ autoreconf -i
configure.ac:12: installing './compile'
configure.ac:47: installing './config.guess'
configure.ac:47: installing './config.sub'
configure.ac:10: installing './install-sh'
configure.ac:10: installing './missing'
Makefile.am: installing './depcomp'
pi@raspberrypi:~/fdk-aac-cli-enc $ make
make: *** No targets specified and no makefile found.  Stop.
```

rwcts, 2020-05-06:
I see that its likely that the ./configure was missing in the instructions. That seems, so far to be resolving this and moving forward. I will update the wiki if this resolved

extract the source, and cd to the source directory
2) autoreconf -i
3) ./configure
4) make
5) sudo make install

rwcts, 2020-05-06:
The system now works! I corrected the wiki on here to reflect the missing items. I started from scratch and setup the system again to make sure nothing was missing. The audio is not as clear as it was on the computer with ubuntu i had been using but i may be just ok with that and troubleshoot later. All is working as expected I can see that some novice users like myself will have trouble tuning in the config.json file with the proper parameters as they are specific to the system and equipment you will be using. Thanks again for those that helped me.

jwestyp, 2020-05-07:
If you want to try a more vanilla linux build, Ubuntu has an official image. I added a section to the installation wiki page to set it up completely as ubuntu 18.0.4. There are some decode issues with the Pi using a 32 bit OS.

rwcts, 2020-05-08:
j

> 
> 
> If you want to try a more vanilla linux build, Ubuntu has an official image. I added a section to the installation wiki page to set it up completely as ubuntu 18.0.4. There are some decode issues with the Pi using a 32 bit OS.

Thanks for the info. I tried Ubuntu 64 ARM on the pi however is still had several issues with fdkaac not being available in the repo.

closed 2020-05-08 by rwcts

jwestyp, 2020-05-08:
I was able to install fdkaac and Sox today. The ffmpeg step is not required
any more.

rwcts, 2020-05-08:
I have several Raspberrypi 4B's and will test this. Thank you!

jwestyp, 2020-05-08:
Mine is in production now at the moment
https://www.broadcastify.com/calls/systems/83