TrunkRecorder/trunk-recorder #347: Broadcastify Calls - ubuntu 18.04 getting error sending

rwcts, 2020-05-03:
Running Ubuntu 18.04 on laptop. Below is the output of the build with a few errors that may explain the issue but unsure why we are getting them. Also when getting prerequisites installed the libgnuradio-uhd version was showing 3.7.13 but needed to be 3.7.11 That is the only thing different from what was listed on the instructions.  Below will also show the output and errors when attempting to run as well as the config.json file.

**-----------------------------------------------START build log--------------------------------------**
broadcastify@Broadcastify-Calls:~/trunk-recorder$ git clone https://github.com/robotastic/trunk-recorder.git ~/trunk-recorder/
Cloning into '/home/broadcastify/trunk-recorder'...
remote: Enumerating objects: 45, done.
remote: Counting objects: 100% (45/45), done.
remote: Compressing objects: 100% (38/38), done.
remote: Total 7226 (delta 16), reused 14 (delta 7), pack-reused 7181
Receiving objects: 100% (7226/7226), 3.52 MiB | 5.33 MiB/s, done.
Resolving deltas: 100% (5115/5115), done.
broadcastify@Broadcastify-Calls:~/trunk-recorder$ cd ~/trunk-recorder
broadcastify@Broadcastify-Calls:~/trunk-recorder$ cmake .
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11
-- Found GNURADIO_RUNTIME: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so  
-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- Found CPPUNIT: /usr/lib/x86_64-linux-gnu/libcppunit.so;dl  
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found GNURADIO_RUNTIME: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so  
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: ANALOG
-- Checking for module 'gnuradio-analog'
--   Found gnuradio-analog, version 3.7.11
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-analog.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so;/usr/lib/x86_64-linux-gnu/libvolk.so
-- Found GNURADIO_ANALOG: /usr/lib/x86_64-linux-gnu/libgnuradio-analog.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so;/usr/lib/x86_64-linux-gnu/libvolk.so  
GNURADIO_ANALOG_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
-- Checking for module 'gnuradio-blocks'
--   Found gnuradio-blocks, version 3.7.11
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found GNURADIO_BLOCKS: /usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so  
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: DIGITAL
-- Checking for module 'gnuradio-digital'
--   Found gnuradio-digital, version 3.7.11
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-digital.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found GNURADIO_DIGITAL: /usr/lib/x86_64-linux-gnu/libgnuradio-digital.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so  
GNURADIO_DIGITAL_FOUND = TRUE
Checking for GNU Radio Module: FILTER
-- Checking for module 'gnuradio-filter'
--   Found gnuradio-filter, version 3.7.11
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-filter.so;/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found GNURADIO_FILTER: /usr/lib/x86_64-linux-gnu/libgnuradio-filter.so;/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so  
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: PMT
-- Checking for module 'gnuradio-runtime'
[... 334 lines cut ...]
[2020-05-03 11:50:30.720698] (info)   	 P25 Trunking two-stage decimator - Initial decimated rate: 128000 Second decimated rate: 32000 FA: 6250 FB: 16000 System Rate: 2048000
[2020-05-03 11:50:30.731868] (info)   	 P25 Trunking ARB - Initial Rate: 2048000 Resampled Rate: 32000 Initial Decimation: 16 System Rate: 24000 ARB Rate: 0.75
[2020-05-03 11:50:33.004352] (error)   [GHCS] Retuning to Control Channel: 8.514875e+08
[2020-05-03 11:50:33.004500] (info)   	 - System Source 0 - Min Freq: 8.504800e+08 Max Freq: 8.524000e+08
[2020-05-03 11:50:33.004641] (error)   [GHCS]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2020-05-03 11:50:36.006508] (error)   [GHCS] Retuning to Control Channel: 8.514250e+08
[2020-05-03 11:50:36.006706] (info)   	 - System Source 0 - Min Freq: 8.504800e+08 Max Freq: 8.524000e+08
[2020-05-03 11:50:36.006854] (error)   [GHCS]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-05-03 11:50:39.009276] (error)   [GHCS] Retuning to Control Channel: 8.514875e+08
[2020-05-03 11:50:39.009397] (info)   	 - System Source 0 - Min Freq: 8.504800e+08 Max Freq: 8.524000e+08
[2020-05-03 11:50:39.009527] (error)   [GHCS]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-05-03 11:50:42.009762] (error)   [GHCS] Retuning to Control Channel: 8.514250e+08
[2020-05-03 11:50:42.009901] (info)   	 - System Source 0 - Min Freq: 8.504800e+08 Max Freq: 8.524000e+08
[2020-05-03 11:50:42.010275] (error)   [GHCS]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-05-03 11:50:45.011586] (error)   [GHCS] Retuning to Control Channel: 8.514875e+08
[2020-05-03 11:50:45.011710] (info)   	 - System Source 0 - Min Freq: 8.504800e+08 Max Freq: 8.524000e+08
[2020-05-03 11:50:45.011822] (error)   [GHCS]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-05-03 11:50:48.011074] (error)   [GHCS] Retuning to Control Channel: 8.514250e+08
[2020-05-03 11:50:48.011202] (info)   	 - System Source 0 - Min Freq: 8.504800e+08 Max Freq: 8.524000e+08
[2020-05-03 11:50:48.011344] (error)   [GHCS]	 Control Channel Message Decode Rate: 1/sec, count:  3
^Z
[5]+  Stopped                 ./recorder
broadcastify@Broadcastify-Calls:~/trunk-recorder$ 

**----------------------------START run ./recorder ---------------------------------**

bctrainers, 2020-05-03:
Looking through the code, most of those compile notices/warnings are just that, informational warning or notice.

> [ 68%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/uploaders/call_uploader.cc.o
/home/broadcastify/trunk-recorder/trunk-recorder/main.cc: In function ‘void unit_check()’:
/home/broadcastify/trunk-recorder/trunk-recorder/main.cc:984:11: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
system(shell_command);

and

> [ 70%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/call_conventional.cc.o
/home/broadcastify/trunk-recorder/trunk-recorder/uploaders/call_uploader.cc: In function ‘void* upload_call_thread(void*)’:
/home/broadcastify/trunk-recorder/trunk-recorder/uploaders/call_uploader.cc:28:9: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
system(shell_command);

and

> [ 73%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/systems/p25_trunking.cc.o
/home/broadcastify/trunk-recorder/trunk-recorder/call.cc: In member function ‘void Call::end_call()’:
/home/broadcastify/trunk-recorder/trunk-recorder/call.cc:184:15: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
      system(shell_command.str().c_str());
      ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Ref Pull Request: https://github.com/robotastic/trunk-recorder/pull/329 

@EricTendian, it may be worthwhile doing a quick review of those three code segments in ref to the pull request that was merged in.

bctrainers, 2020-05-03:
Also, we need to know what system you are wanting to monitor, decode, and record. An RRDB link would be best, along with the system ID('s) that you're after. It's possible that the single RTLSDR in use cannot cover the entire spectrum needed.

Additionally, a PPM error of 69 is quite large for a SDR - do ensure that is that is indeed correct. Generally, if you're having PPM errors that large, it would be best to obtain another RTLSDR.

rwcts, 2020-05-03:
Thank you for the update. I also received some info from broadcastify as well and The example they list on their site wiki was changed some. They also said in credit to them just like your saying is that the PPM is different with each piece of hardware and needs to be checked. GQRX was recommended to play with and allow testing to see the best PPM setting for the RTL-SDR used. This worked and it should have been set at 0 as that was the best reception. And for historical notes. I used the NOAA weather frequency to test the best PPM. Its an interesting process but that worked well.

I did reset some of the original settings but below is the RRDB frequencies. This setup now seems to be working. Also this may NOT be good information but the two RTL-SDR dongles (aluminum cases) are different and received different results. One is the v3 with BIAS+HF that did NOT work no matter what I tried but then the one without a version number and does not have the BIAS+HF on the label worked with the attached settings.

In addition keep in mind that Radio Reference page references to have in the config file "modulation": "fsk4" however for me that does not work at all and it must be "modulation": "qpsk" This is likely different depending on the equipment used but was my experience.

3 (3) | 040 (28) | Goodhue Co. Simulcast | Goodhue | 851.425c | 851.4875a | 851.675 | 851.7125 | 851.975a | 852.450 | 852.750|852.900 | 853.100 | 853.400 | 853.675 | 853.700 | 854.3625 |  
https://www.radioreference.com/apps/db/?siteId=13268

Below is the new config file that is working.

{
	"sources": [{
		"center": 852290000.0,
		"rate": 2048000,
		"ppm": 0.0,
		"gain": 35.0,
		"digitalLevels": 3,
		"digitalRecorders": 4,
		"driver": "osmosdr",
		"device": "rtl=00,buflen=65536",
		"modulation": "qpsk"
	}],
	"systems": [{
		"control_channels": [851425000, 851487500, 851975000, 853400000],
		"type": "p25",
		"shortName": "GHCS",
		"broadcastifyApiKey": "REAL-KEY-HERE",
		"broadcastifySystemId": "99",
		"audioArchive": false
	}],
	"broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
	"controlWarnRate": 5,
	"callTimeout": 0.75
}

bctrainers, 2020-05-03:
Great to hear it's working now. 

I'm a bit curious about the RTLSDR hardware differences. I've quite a few RTLSDR's laying around from 2015 to 2019. So I probably have some sticks like what you describe. :)

rwcts, 2020-05-03:
I was able to do some more testing and the BIAS T+HF is working. I am unsure why since I had troubleshoot multiple times with the same results. However I have included the photos of inside front and back of both units as asked. Thanks! 
[image]
[image]

closed 2020-05-05 by rwcts