TrunkRecorder/trunk-recorder #353: Raspberry Pi 4B Trunk Recorder Build Error

SDWolf, 2020-05-04:
When I try to build the Trunk-Recorder folder with the cmake . command, I get errors.  I am not experienced with the Pi and unsure how to resolve this.

CMake Error at /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find CURL (missing: CURL_LIBRARY CURL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.13/Modules/FindCURL.cmake:74 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:106 (find_package)

-- Configuring incomplete, errors occurred!

kcwebby, 2020-05-04:
install libcurlpp-dev  had that issue recently.

EricTendian, 2020-05-06:
@SDWolf I've been trying to come up with a Dockerfile for Raspberry Pi that works, but I don't have one to test it on. If you don't mind, maybe you could try this Dockerfile (as a replacement to the current Dockerfile in this repo) and see if it works?

```dockerfile
FROM arm32v7/ubuntu:18.04 AS base

RUN apt-get update
RUN apt-get install -y gnuradio gnuradio-dev gr-osmosdr libhackrf-dev libuhd-dev libgmp-dev
RUN apt-get install -y git cmake build-essential pkg-config libboost-all-dev libusb-1.0-0.dev libssl-dev libcurl4-openssl-dev liborc-0.4-dev
RUN apt-get install -y ca-certificates expat libgomp1 fdkaac sox
# install necessary locales
RUN apt-get install -y locales \
    && echo "en_US.UTF-8 UTF-8" > /etc/locale.gen \
    && locale-gen

RUN apt-get autoremove -y && \
    apt-get clean -y

COPY . /src/trunk-recorder
RUN cd /src/trunk-recorder && cmake . && make
RUN mkdir /app && cp /src/trunk-recorder/recorder /app

RUN mkdir /app/media \
 && mkdir /app/config

ENV LC_ALL en_US.UTF-8
ENV LANG en_US.UTF-8
ENV LANGUAGE en_US.UTF-8

WORKDIR /app
CMD ["./recorder","--config=/app/config/config.json"]
```

SDWolf, 2020-05-07:
I'll try it out this weekend.  Thanks.

SDWolf, 2020-05-11:
Sorry, I was not able to get DockerFile to work.  I'm not sure where I am going wrong, I'm very new to linux command line and the rasperry pi.  I try to follow the instructions, but I always end up getting errors.

closed 2020-05-11 by SDWolf

EricTendian, 2020-05-11:
@SDWolf no problem, thanks for trying! 😄