TrunkRecorder/trunk-recorder #359: new commit (last night) won't compile.

kcwebby, 2020-05-19:
I was working on this machine last night, and prior to your latest commit, it compiled perfectly.

Did the dependencies change?

```
[ 84%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/recorders/p25conventional_recorder.cc.o
/home/jasonk/trunk-recorder/trunk-recorder/recorders/p25conventional_recorder.cc: In member function ‘virtual void p25conventional_recorder::start(Call*)’:
/home/jasonk/trunk-recorder/trunk-recorder/recorders/p25conventional_recorder.cc:67:7: error: ‘gr::op25_repeater::gardner_costas_cc::sptr p25_recorder::costas_clock’ is private within this context
       costas_clock->reset();
       ^~~~~~~~~~~~
In file included from /home/jasonk/trunk-recorder/trunk-recorder/recorders/p25conventional_recorder.cc:1:0:
/home/jasonk/trunk-recorder/trunk-recorder/recorders/p25_recorder.h:202:46: note: declared private here
   gr::op25_repeater::gardner_costas_cc::sptr costas_clock;
                                              ^~~~~~~~~~~~
CMakeFiles/recorder.dir/build.make:590: recipe for target 'CMakeFiles/recorder.dir/trunk-recorder/recorders/p25conventional_recorder.cc.o' failed
make[2]: *** [CMakeFiles/recorder.dir/trunk-recorder/recorders/p25conventional_recorder.cc.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/recorder.dir/all' failed
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

waterbwuk, 2020-05-19:
Was just coming here to post that. I'm experiencing the same problem on my end.

NoahJaehnert, 2020-05-19:
+1, getting the same issue on my end as well (Raspberry Pi 4).

EricTendian, 2020-05-19:
#358 has a fix for this, see https://github.com/robotastic/trunk-recorder/pull/358/commits/4365f1b7459eb612f241437b513a970839960889.

waterbwuk, 2020-05-19:
> #358 has a fix for this, see [4365f1b](https://github.com/robotastic/trunk-recorder/commit/4365f1b7459eb612f241437b513a970839960889).

Can confirm applying [4365f1b](https://github.com/robotastic/trunk-recorder/commit/4365f1b7459eb612f241437b513a970839960889) fixes the issue. Thank you!

robotastic, 2020-05-20:
Awesome! Thanks @EricTendian ! I made what I thought was just a tiny change without testing. I will go review that PR

EricTendian, 2020-05-22:
I think this can be closed since the fix was merged in.

robotastic, 2020-05-22:
Agreed! Just coming here to do that.

closed 2020-05-22 by robotastic