TrunkRecorder/trunk-recorder #361: Broadcastify Calls - Duplicate Call Handling

Ares4570, 2020-05-27:
Broadcastify handles duplicate calls by generating an error (Broadcastify Metadata Upload Error: SKIPPED---ALREADY-RECEIVED-THIS-CALL). Unfortunately, this error seems to prevent Trunk Recorder from deleting the source audio .wav, the converted m4a, or the metadata.

In a setup where talkgroups are sometimes simulcast, and sometimes not, it's necessary to have the same talkgroups across multiple systems which makes this a fairly common occurrence.

It would be very nice if Trunk Recorder could consider a "SKIPPED" error as a success.

leee, 2021-05-15:
I don't use this platform, so have no means of delving in and testing a patch for this. Would you be willing to test the following diff? You can save the contents of the following to a file and apply it by running `git apply [name of file].diff`, then build and test the change.

```
diff --git a/trunk-recorder/uploaders/broadcastify_uploader.cc b/trunk-recorder/uploaders/broadcastify_uploader.cc
index a53361f..4ee5f7d 100644
--- a/trunk-recorder/uploaders/broadcastify_uploader.cc
+++ b/trunk-recorder/uploaders/broadcastify_uploader.cc
@@ -221,6 +221,11 @@ int BroadcastifyUploader::upload(struct call_data_t *call) {
     std::string code = response_buffer.substr(0, spacepos);
     std::string message = response_buffer.substr(spacepos + 1);

+    if (code = "1" && message == "SKIPPED---ALREADY-RECEIVED-THIS-CALL") {
+      BOOST_LOG_TRIVIAL(info) << "[" << call->short_name << "]\tTG: " << call->talkgroup << "\tFreq: " << FormatFreq(call->freq) << "\tBroadcastify Upload Skipped: " << message;
+      return 0;
+    }
+
     if (code != "0") {
       BOOST_LOG_TRIVIAL(error) << "[" << call->short_name << "]\tTG: " << call->talkgroup << "\tFreq: " << FormatFreq(call->freq) << "\tBroadcastify Metadata Upload Error: " << message;
       return 1;
@@ -241,4 +246,4 @@ int BroadcastifyUploader::upload(struct call_data_t *call) {
   } else {
     return 1;
   }
-}
\ No newline at end of file
+}
```

Ares4570, 2021-05-16:
@leee Thanks for taking the time to write the diff. Unfortunately it fails to build - I get the following when I run "rebuild.sh" (manually updated the repo, then ran the diff beforehand):

```
[ 65%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/uploaders/broadcastify_uploader.cc.o
/home/sdr/trunk-recorder/trunk-recorder/uploaders/broadcastify_uploader.cc: In member function ‘virtual int BroadcastifyUploader::upload(call_data_t*)’:
/home/sdr/trunk-recorder/trunk-recorder/uploaders/broadcastify_uploader.cc:224:14: error: could not convert ‘code.std::__cxx11::basic_string<char>::operator=(((int)((char)((((const char*)"1") != 0) && std::operator==<char, std::char_traits<char>, std::allocator<char> >(message, ((const char*)"SKIPPED---ALREADY-RECEIVED-THIS-CALL"))))))’ from ‘std::__cxx11::basic_string<char>’ to ‘bool’
     if (code = "1" && message == "SKIPPED---ALREADY-RECEIVED-THIS-CALL") {
         ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/recorder.dir/build.make:158: recipe for target 'CMakeFiles/recorder.dir/trunk-recorder/uploaders/broadcastify_uploader.cc.o' failed
make[2]: *** [CMakeFiles/recorder.dir/trunk-recorder/uploaders/broadcastify_uploader.cc.o] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/recorder.dir/all' failed
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

leee, 2021-05-16:
@Ares4570 thank you, that's a common finger error 🙂 , here's a diff with an extra `=`. You can `git checkout .` and apply the following again:

```
diff --git a/trunk-recorder/uploaders/broadcastify_uploader.cc b/trunk-recorder/uploaders/broadcastify_uploader.cc
index a53361f..326903e 100644
--- a/trunk-recorder/uploaders/broadcastify_uploader.cc
+++ b/trunk-recorder/uploaders/broadcastify_uploader.cc
@@ -221,6 +221,11 @@ int BroadcastifyUploader::upload(struct call_data_t *call) {
     std::string code = response_buffer.substr(0, spacepos);
     std::string message = response_buffer.substr(spacepos + 1);

+    if (code == "1" && message == "SKIPPED---ALREADY-RECEIVED-THIS-CALL") {
+      BOOST_LOG_TRIVIAL(info) << "[" << call->short_name << "]\tTG: " << call->talkgroup << "\tFreq: " << FormatFreq(call->freq) << "\tBroadcastify Upload Skipped: " << message;
+      return 0;
+    }
+
     if (code != "0") {
       BOOST_LOG_TRIVIAL(error) << "[" << call->short_name << "]\tTG: " << call->talkgroup << "\tFreq: " << FormatFreq(call->freq) << "\tBroadcastify Metadata Upload Error: " << message;
       return 1;
@@ -241,4 +246,4 @@ int BroadcastifyUploader::upload(struct call_data_t *call) {
   } else {
     return 1;
   }
-}
\ No newline at end of file
+}
```

Ares4570, 2021-05-16:
@leee Awesome, thanks! That does compile, and seems to be working. Thank you very much!

I do see it's still creating folders as if it were going to keep them since the files are temporarily stored in these dirs before being uploaded to Broadcastify.

leee, 2021-05-16:
@Ares4570 I think cleanup of empty directories will have to be a different conversation (or transitioning to using `/tmp` or something, or similar).

But to confirm, calls that have already been received by this platform are being processed accordingly and cleared out? If so, I'll open up a PR for this.

Ares4570, 2021-05-16:
@leee Yup, agreed. I think a transition to /tmp would make sense but that's another conversion as mentioned.

Confirming this works. I'm seeing the duplicates mentioned in the trunk-recorder log, but nothing is being retained in the folders. Everything is being deleted.

leee, 2021-05-16:
@Ares4570 I just realized, this is probably as easy as setting `captureDir` to something in `/tmp`. I'll go submit a PR for this now.

Ares4570, 2021-05-16:
@leee 

Hah, yup. Obvious and easy. That worked just fine. Just updated my config and seems to be working just fine going to /tmp/trunk-recorder. Awesome.

closed 2021-05-17 by robotastic

rosecitytransit, 2021-05-19:
In some ways, it's nice to have the duplicate calls left behind if you're not sure who else is uploading the call. But it's true it's not really an error if you're only concerned about calls that Broadcastify doesn't (now) have. An alternative is to simply have a cron entry that deletes the previous day's directory.

Just FYI:
-You should be able attach .patch (or other) files to a comment here.
-I think "simulcast" in the first comment should actually be "patched". My understanding is that "simulcast" means the entire system/same exact signals (control and voice channels) is broadcast from multiple locations and appears as one site, while "patched" means a second site is broadcasting the same talk group but has it's own control channel.

PR #472 (2021-05-17): Treat Broadcastify duplicate skips as a success
Resolves #361

@Ares4570 tested.
commit 27a47fa0 (2021-05-17): Treat Broadcastify duplicate skips as a success

Resolves #361