TrunkRecorder/trunk-recorder #363: Cannot Find the Control Channel

afera15, 2020-05-27:
I am getting an error that I cannot find the control channel

```
[2020-05-27 17:42:32.737367] (info)   Using Config file: config.json

[2020-05-27 17:42:32.737936] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------

[2020-05-27 17:42:32.737965] (info)   
-------------------------------------
SYSTEMS
-------------------------------------

[2020-05-27 17:42:32.738007] (info)   

System Number: 1
-------------------------------------

[2020-05-27 17:42:32.738027] (info)   Short Name: MPSCS
[2020-05-27 17:42:32.738055] (info)   System Type: p25
[2020-05-27 17:42:32.738069] (info)   Control Channels: 
[2020-05-27 17:42:32.738138] (info)     8.531375e+08
[2020-05-27 17:42:32.738197] (info)     8.527625e+08
[2020-05-27 17:42:32.738217] (info)   API Key: 
[2020-05-27 17:42:32.738232] (info)   Broadcastify API Key: KEY
[2020-05-27 17:42:32.738246] (info)   Broadcastify Calls System ID: ##
[2020-05-27 17:42:32.738262] (info)   Upload Script: 
[2020-05-27 17:42:32.738275] (info)   Call Log: true
[2020-05-27 17:42:32.738293] (info)   Audio Archive: false
[2020-05-27 17:42:32.738310] (info)   Loading Talkgroups...
[2020-05-27 17:42:32.738327] (info)   Talkgroups File: 
[2020-05-27 17:42:32.738344] (info)   Loading Unit Tags...
[2020-05-27 17:42:32.738360] (info)   Unit Tags File: 
[2020-05-27 17:42:32.738373] (info)   Record Unknown Talkgroups: true
[2020-05-27 17:42:32.738389] (info)   Decode MDC: false
[2020-05-27 17:42:32.738401] (info)   Decode FSync: false
[2020-05-27 17:42:32.738417] (info)   Decode Star: false
[2020-05-27 17:42:32.738429] (info)   Decode TPS: false
[2020-05-27 17:42:32.738443] (info)   Talkgroup Display Format: Id
[2020-05-27 17:42:32.738461] (info)   Hide Encrypted Talkgroups: true
[2020-05-27 17:42:32.738476] (info)   Hide Unknown Talkgroups: false
[2020-05-27 17:42:32.738490] (info)   Minimum Call Duration (in seconds): 0
[2020-05-27 17:42:32.738506] (info)   
[2020-05-27 17:42:32.738519] (info)   

-------------------------------------
SOURCES
-------------------------------------

[2020-05-27 17:42:32.738568] (info)   Driver: osmosdr
[2020-05-27 17:42:32.738582] (info)   Center: 7.747812e+08
[2020-05-27 17:42:32.738604] (info)   Rate: 2400000
[2020-05-27 17:42:32.738625] (info)   Error: 0
[2020-05-27 17:42:32.738640] (info)   PPM Error: 0
[2020-05-27 17:42:32.738658] (info)   Auto gain control: false
[2020-05-27 17:42:32.738671] (info)   Gain: 35
[2020-05-27 17:42:32.738689] (info)   IF Gain: 0
[2020-05-27 17:42:32.738704] (info)   BB Gain: 0
[2020-05-27 17:42:32.738719] (info)   LNA Gain: 0
[2020-05-27 17:42:32.738736] (info)   PGA Gain: 0
[2020-05-27 17:42:32.738751] (info)   TIA Gain: 0
[2020-05-27 17:42:32.738765] (info)   MIX Gain: 0
[2020-05-27 17:42:32.738779] (info)   VGA1 Gain: 0
[2020-05-27 17:42:32.738794] (info)   VGA2 Gain: 0
[2020-05-27 17:42:32.738811] (info)   Squelch: 0
[2020-05-27 17:42:32.738826] (info)   Idle Silence: false
[2020-05-27 17:42:32.738838] (info)   Digital Recorders: 1
[2020-05-27 17:42:32.738855] (info)   Debug Recorder: false
[... 232 lines cut ...]
		"driver": "osmosdr",
		"device": "rtl=4,buflen=65536",
		"modulation": "fsk4"
	}],
	"systems": [{
		"control_channels": [853137500, 852762500],
		"type": "p25",
		"shortName": "MPSCS",
		"broadcastifyApiKey": "BROADCASTIFY-CALLS-KEY",
		"broadcastifySystemId": "##",
		"audioArchive": false,
		"hideEncrypted": true
	}],
	"broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
	"controlWarnRate": 5,
	"callTimeout": 0.75
}
```

An when I use GQRX:
[image]

I have tried playing with the Gain between 30 and 40 as well as PPM between 1 and -1

Any suggestions as I cannot figure this out

dkettmann, 2020-05-30:
Try moving the source that covers the control channel to be the first one in config.json.  I recently migrated a system that worked fine with the second source defined covering the CC which then broke when moving that same config.json on a recent build.  I think it is adjacent to #328 which I have a suspicion was mistakenly closed but I'm not confident in that at all. Worth a shot.

afera15, 2020-06-01:
I have pulled the most recent update and moved the first receiver to cover the control channel, no success

frameshift18, 2020-06-01:
Two things I see.

1. the center freqs on some of your sources are only 1mhz apart. but you have the rates at 2.4mhz. I would space out these sources so they don't overlap and in the process would need less dongles to cover the same spectrum. Less dongles are usually better.

2. try the other modulation type, qpsk.

dkettmann, 2020-06-01:
Thanks for trying my idea, recently bitten by it and perhaps seeing it everywhere now. 

Frameshift's points seem right and a little embarrassed missed them. I also see some of them with missing zeros in the center freq, even with the decimal .0 which I'm not used to some are missing one. I do think they are spaced so tight as to be able to cope with them missing maybe but maybe not.

More likely the culprit to me, I looked at the system at https://www.radioreference.com/apps/db/?sid=100 and I think the modulation should be qpsk for a P25 phase 1 which would make frame's fix #2 the one that has my bet.  Good luck!

afera15, 2020-06-01:
So I spaced them out to make sure they don't overlap and verified that the file looks good but I am still getting the same thing.  I have also tried changing to fsk4. Here is the current config:

```
{
	"sources": [{
		"center": 852200000,
		"rate": 2400000,
		"ppm": 0,
		"gain": 35.0,
		"digitalLevels": 8,
		"digitalRecorders": 13,
		"driver": "osmosdr",
		"device": "rtl=0,buflen=65536",
		"modulation": "qpsk"
	},{
		"center": 774781250,
		"rate": 2400000,
		"ppm": -1,
		"gain": 35.0,
		"digitalLevels": 8,
		"digitalRecorders": 1,
		"driver": "osmosdr",
		"device": "rtl=1,buflen=65536",
		"modulation": "qpsk"
	},{
		"center": 854812500,
		"rate": 2400000,
		"ppm": 0,
		"gain": 35.0,
		"digitalLevels": 8,
		"digitalRecorders": 3,
		"driver": "osmosdr",
		"device": "rtl=2,buflen=65536",
		"modulation": "qpsk"
	}],
	"systems": [{
		"control_channels": [853137500, 852762500],
		"type": "p25",
		"shortName": "MPSCS-GR",
		"broadcastifyApiKey": "KEY",
		"broadcastifySystemId": "ID",
		"audioArchive": false,
		"hideEncrypted": true
	}],
	"broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
	"controlWarnRate": 5,
	"callTimeout": 0.75
}
```

Any other ideas.  Thank you all for the help!!! <3

dkettmann, 2020-06-02:
With so many moving parts I'm prone to seeing what those more eperienced suggest but what happens if you run the following? In an attempt to simplify and it and walk before you crawl.  That config should run one stick and grab the control channel on either side of center. It will complain it doesn't have sources to give to the other channels but it will prove out a few things that *should* work IMO.

```
{
	"sources": [{
		"center": 853000000,
		"rate": 2400000,
		"ppm": 0,
		"gain": 35.0,
		"digitalLevels": 8,
		"digitalRecorders": 13,
		"driver": "osmosdr",
		"device": "rtl=0,buflen=65536",
		"modulation": "qpsk"
		}],
	"systems": [{
		"control_channels": [853137500, 852762500],
		"type": "p25",
		"shortName": "MPSCS-GR",
		"broadcastifyApiKey": "KEY",
		"broadcastifySystemId": "ID",
		"audioArchive": false,
		"hideEncrypted": true
	}],
	"broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
	"controlWarnRate": 5,
	"callTimeout": 0.75
}
```

afera15, 2020-06-02:
@dkettmann Thank you so much!!! It works now with the one channel.  So now I need to see what is wrong with the other two channels...

dkettmann, 2020-06-02:
So besides making the config simpler as I stated that change did one other thing.  It moved the "source" device to be the first (and only) device to work around the issue/bug I mentioned above.  Might be worth while to add the others below it and see if that (combined with correcting the qpsk) gets you where you want.  I'm considering maybe the new code didn't build in to a new executable and still hitting that? Again, I'd defer to someone with more experience if they want to pipe in. Good luck, hope that gets it.

afera15, 2020-06-17:
This can be closed

closed 2020-06-17 by afera15