TrunkRecorder/trunk-recorder #364: Regression: Audio files are deleted before script processing completes

riverar, 2020-05-31:
8ac74db11d60152415dd5c8ce6c58868b6bbf01e breaks scripts, as it introduces new code to delete audio files after the shell script is executed. But because the shell command is pushed into the background (via `&`), the system call is effectively async and the delete occurs before the shell script is finished.

https://github.com/robotastic/trunk-recorder/blob/8ac74db11d60152415dd5c8ce6c58868b6bbf01e/trunk-recorder/call.cc#L167-L169

https://github.com/robotastic/trunk-recorder/blob/8ac74db11d60152415dd5c8ce6c58868b6bbf01e/trunk-recorder/call.cc#L177-L192

riverar, 2020-08-03:
Pinging @robotastic for visibility

robotastic, 2020-08-03:
Thanks - I had missed this. There may not be an easy fix for this. The best approach would be to move file deletion to the script, so that the script handles deleting the file after it has done everything.

lennycitizen, 2021-09-03:
@robotastic any update on thinking here?

robotastic, 2021-09-03:
There maybe good news here - the `v4.0-beta` branch totally changes how call completion is handled and should fix this. Feel free to give this a try. I am still hunting down a few bugs, but it should be merged in soon.