TrunkRecorder/trunk-recorder #365: Docker status

robbiet480, 2020-06-05:
Hi @robotastic,

Thought I should create a issue to continue work from https://github.com/robotastic/docker-gnuradio/issues/2. Now that I've figured out the path to get multiple architectures building properly, i'm working on improving the entire trunk-recorder Docker ecosystem. This includes updating the two repos `Dockerfile` to follow best practices, slimming them down as much as possible and improving the Docker Compose story. Unrelated to trunk-recorder specifically, I'm also working on containerizing @chuot's `rdio-scanner` so that users can get a whole stack running with a single Docker Compose file and much faster too, especially for those on Raspberry Pi. 

I've created this issue because I was hoping to get the current lay of the land from you and @EricTendian. I see a lot of Docker issues in multiple repos but am unsure where my time is best spent so as to not duplicate work. So, a few questions to start:

1. Based on #349 it appears we need a new Dockerfile to support GNURadio 3.8. Is this correct? I see https://github.com/robotastic/trunk-recorder-docker-gr3.8, is this actually working?
2. What's the issue with Locales? I see you are setting everything to en_US.UTF-8. Why? Can we stop doing that and just use `C` or `C.UTF-8`?

Here's the current Dockerfile I'm using which works nicely and hews to expected Docker standards:

```Dockerfile
FROM robotastic/docker-gnuradio:latest

WORKDIR /src

COPY . .

RUN cmake . && make -j`nproc` && cp recorder /recorder

CMD ["/recorder"]
```

And the new and improved `docker-compose.yaml`, featuring rdio-scanner support too:

```yaml
version: '3'
volumes:
  rdio_database:
services:
  recorder:
    image: robbiet480/trunk-recorder:latest
    container_name: trunk-recorder
    restart: always
    devices:
      - /dev/bus/usb/001/003:/dev/bus/usb/001/003
    volumes:
      - ./media:/app/media
      - ./config:/app/config
      - ./ebrcs.csv:/app/ebrcs.csv
  rdio:
    image: robbiet480/rdio-scanner:latest
    container_name: rdio-scanner
    volumes:
      - ./rdio_config.json:/app/server/config.json
      - rdio_database:/app/server/database.sqlite
      - ./media:/calls
```

Excited to hear your thoughts and contribute to this fantastic project!

jquagga, 2020-06-05:
I started looking at the docker-compose file yesterday as well (but I'm not a home-assistant software developer so my skillset is substantially more limited :smile: ).  I didn't like that the [existing docker-compose.yml](https://github.com/robotastic/trunk-recorder-docker/blob/master/docker-compose.yml) ran as privileged.  I know is needs access to the sdr dongles, but I didn't want privileged.  So like you, I've been using:

```docker
devices:
      - "/dev/bus/usb:/dev/bus/usb"
```

I have multiple dongles, so pass the whole bus so they can all be accessed.  Not what I'd want ideally but I'm guessing the goal behind the docker-compose is to have a "run and go" method.  

I like that your version doesn't require a docker-build as it doesn't copy the talkgroups file in.  In terms of keeping with the "run and go" would it make sense to move the talkgroups .csv files into their own /csv directory (or whatever) and then pass that as a volume and update the configuration file to source them from there?  Then we could have multiple .csv and config.json could just pull them in from a directory.  

My final thought was that I wasn't sure if we wanted to continue to run the docker container as root.  That's generally not considered best practice as far as I know, but without that the user would have to configure permissions on the host to allow the docker user rw access to the sdr dongles.  So it's not "fire and forget".  

I would agree docker is the "easy" way to go with running trunk-recorder; mine is running on my Ryzen desktop but I've bought a NUC and will just copy a folder between the machines and docker-compose up when I move it to it's dedicated server.  In the past I've tried to build it natively on Pi's but was converted over to being a docker person since my last run at trunk-recorder.  Going docker this time was substantially easier.

robotastic, 2020-06-05:
@robbiet480 Thanks so much for looking into this! Improving Docker support has been on my list of things to do for a long time. I agree - once this is buttoned up, this should be the primary way of running Trunk Recorder. I can look at setting up an automated build pipeline so it automatically spits out new multi-arch images when there is a push to the repo. It may take a little EC2 time, but totally worth it. I actually just started messing around with buildx too, so all in on this approach.

More detailed:
1) The Dockerfile in TR needs to be updated for GR3.8. It can't hurt to goto 3.8. It will probably have the latest SDR drivers.
2) I am flexible on Locales, I think I just went with that because I was getting an error and needed to specify some Repo.
3) Do you mean from TR? I think we just need the recorder binary and can delete everything after we are done building. Def worth a shot.
4) Yea, I am seeing this in another project I am working on and compiling OpenCV. I don't think there are any good shortcuts. If we move it to an AWS EC2 instance though, it shouldn't be too big a problem. There are not too many commits to Trunk Recorder.

For the Pi - It could be interesting to look at https://www.balena.io as a way to push and manage Docker images on the Pi. 

Thanks again for digging into this!

robbiet480, 2020-06-05:
@jquagga Thanks for your feedback here. I knew I forgot to do something when building the new Dockerfile... will push a fix to run as non-root. I don’t believe the user will need to do anything with permissions to pass in the device but will check it out. As for the CSV thing, I like it this way, only because it’s the most flexible option. If users want a folder, they can mount a folder and update their config to point to the mounted path. 

@robotastic I’d suggest using a GitHub action to do builds since it’s just so easy to use. [This one looks perfect and is plug and play](https://github.com/crazy-max/ghaction-docker-buildx).

As for your responses, a few things:

- I tried copying over just the recorder binary but it failed to run due to missing boost or gnuradio libs. What’s weird is I built it off docker-gnuradio and copied it over to a second stage also FROM docker-gnuradio so I think there’s something wrong with the TR build that causes it to not actually be static.
- I was able to improve build times significantly by running one build at a time. You’d think this makes sense but it felt like it sped up too much for it to just be my computer not having enough resources. Almost like there’s something else that causes build to slow when trying for 3 architectures.
- OpenMHZ (off topic): I added a new system early this morning (EBRCS). @loganwilliams DMed me on Twitter a few minutes ago to tell me they also created a system with the same short name as me and reported there were some glitches now. May want to ensure that the short name is unique and possibly rename/delete their new system in the DB?

robotastic, 2020-06-07:
Cool - I didn't realize the GitHub actions included some compute time. I will work on getting that going, and that action you linked to looks like a good starting point. since it is running on Githubs compute, the long build time isn't the end of the world, esp since there are not that many commits.

I will go checkout the openmhz code. It was suppose to check for duplicate system shortnames... clearly broken though.

robbiet480, 2020-06-15:
@robotastic Did you get anywhere with determining if we can do static builds? That's probably the second biggest thing holding up progress & optimization, after getting GNURadio 3.8 building.

robbiet480, 2020-06-16:
https://github.com/robotastic/docker-gnuradio/pull/3

robbiet480, 2020-06-16:
#366 is ready for review. We should get that GNURadio Docker image merged before merging #366. Honestly though, I was thinking if its a good idea to continue building GNURadio and `trunk-recorder` separate, since we really sure be making sure they are versioned together I think. This is fine for now though.

robbiet480, 2020-06-16:
Oh yeah, we should absolutely attempt to get static builds going still. The latest image size (with GNURadio 3.8) is 1.92GB. If we got static builds going, we could probably get to under 100mb.

robotastic, 2020-06-16:
Thanks for patching up the Docker script - I just got something similar working on GitLab and was going to turn to this. I will go work on Static linking right now. I am not sure if we need to even worry about building GnuRadio. I think I was doing that because the version of gnuradio that used to be in the apt-get repo was so old. It is probably better to just install GR using apt-get. Maybe we could keep a bleeding edge version that would compile gnuradio from src nightly to test against.

robbiet480, 2020-06-16:
Would also agree that we should continue doing apt-get for GNURadio.

robotastic, 2020-06-16:
Ok... slight wrinkle... I have everything building staticly, except when it gets to all the gnuradio libs. and if you install them using apt-get, you only get shared libraries. It looks like if you want static libraries, you have to build grnuadio from source. You can use this flag: ENABLE_STATIC_LIBS
( from here: https://www.gnuradio.org/doc/doxygen/build_guide.html )
Check out the `static` branch....
I could also try to make everything static, except for libgnuradio. Not sure how much space that would save.

robbiet480, 2020-06-16:
Not the worst idea to build GNURadio from scratch _if_ it means we also can drop any GUI requirements. That would be a HUGE size savings and almost certainly get things starting up faster too.

robbiet480, 2020-06-16:
Just doing `apt-get install gnuradio` is the largest space hog by far, almost certainly due to installing QT. Here's all the packages it attempts to install:

adwaita-icon-theme alsa-topology-conf alsa-ucm-conf at-spi2-core autoconf
  automake autotools-dev binutils binutils-common binutils-x86-64-linux-gnu
  bladerf blt build-essential ca-certificates cmake cmake-data cpp cpp-8 cpp-9
  curl dbus dbus-user-session dconf-gsettings-backend dconf-service dirmngr
  dmsetup dpkg-dev fakeroot fdkaac ffmpeg file fontconfig fontconfig-config
  fonts-dejavu-core fonts-lyx freeglut3 g++ g++-9 gcc gcc-8 gcc-8-base gcc-9
  gcc-9-base gfortran gfortran-8 gfortran-9 gir1.2-glib-2.0 git git-man
  glib-networking glib-networking-common glib-networking-services gnupg
  gnupg-l10n gnupg-utils gnuradio gnuradio-dev gpg gpg-agent gpg-wks-client
  gpg-wks-server gpgconf gpgsm gr-fcdproplus gr-fosphor gr-iqbal gr-osmosdr
  gsettings-desktop-schemas gtk-update-icon-cache hackrf hicolor-icon-theme
  humanity-icon-theme i965-va-driver ibverbs-providers icu-devtools
  intel-media-va-driver javascript-common krb5-locales less libaacs0
  libairspy0 libairspyhf1 libalgorithm-diff-perl libalgorithm-diff-xs-perl
  libalgorithm-merge-perl libao-common libao4 libaom0 libapparmor1
  libarchive13 libargon2-1 libasan5 libasn1-8-heimdal libasound2
  libasound2-data libass9 libassuan0 libasyncns0 libatk-bridge2.0-0
  libatk1.0-0 libatk1.0-data libatomic1 libatspi2.0-0 libavahi-client3
  libavahi-common-data libavahi-common3 libavc1394-0 libavcodec58
  libavdevice58 libavfilter7 libavformat58 libavresample4 libavutil56
  libbdplus0 libbinutils libbladerf2 libblas3 libbluray2 libboost-all-dev
  libboost-atomic-dev libboost-atomic1.71-dev libboost-atomic1.71.0
  libboost-chrono-dev libboost-chrono1.71-dev libboost-chrono1.71.0
  libboost-container-dev libboost-container1.71-dev libboost-container1.71.0
  libboost-context-dev libboost-context1.71-dev libboost-context1.71.0
  libboost-coroutine-dev libboost-coroutine1.71-dev libboost-coroutine1.71.0
  libboost-date-time-dev libboost-date-time1.71-dev libboost-date-time1.71.0
  libboost-dev libboost-exception-dev libboost-exception1.71-dev
  libboost-fiber-dev libboost-fiber1.71-dev libboost-fiber1.71.0
  libboost-filesystem-dev libboost-filesystem1.71-dev
  libboost-filesystem1.71.0 libboost-graph-dev libboost-graph-parallel-dev
  libboost-graph-parallel1.71-dev libboost-graph-parallel1.71.0
  libboost-graph1.71-dev libboost-graph1.71.0 libboost-iostreams-dev
  libboost-iostreams1.71-dev libboost-iostreams1.71.0 libboost-locale-dev
  libboost-locale1.71-dev libboost-locale1.71.0 libboost-log-dev
  libboost-log1.71-dev libboost-log1.71.0 libboost-math-dev
  libboost-math1.71-dev libboost-math1.71.0 libboost-mpi-dev
  libboost-mpi-python-dev libboost-mpi-python1.71-dev
  libboost-mpi-python1.71.0 libboost-mpi1.71-dev libboost-mpi1.71.0
  libboost-numpy-dev libboost-numpy1.71-dev libboost-numpy1.71.0
  libboost-program-options-dev libboost-program-options1.71-dev
  libboost-program-options1.71.0 libboost-python-dev libboost-python1.71-dev
  libboost-python1.71.0 libboost-random-dev libboost-random1.71-dev
  libboost-random1.71.0 libboost-regex-dev libboost-regex1.71-dev
  libboost-regex1.71.0 libboost-serialization-dev
  libboost-serialization1.71-dev libboost-serialization1.71.0
  libboost-stacktrace-dev libboost-stacktrace1.71-dev
  libboost-stacktrace1.71.0 libboost-system-dev libboost-system1.71-dev
  libboost-system1.71.0 libboost-test-dev libboost-test1.71-dev
  libboost-test1.71.0 libboost-thread-dev libboost-thread1.71-dev
  libboost-thread1.71.0 libboost-timer-dev libboost-timer1.71-dev
  libboost-timer1.71.0 libboost-tools-dev libboost-type-erasure-dev
  libboost-type-erasure1.71-dev libboost-type-erasure1.71.0 libboost-wave-dev
  libboost-wave1.71-dev libboost-wave1.71.0 libboost1.71-dev
  libboost1.71-tools-dev libbrotli1 libbs2b0 libbsd0 libc-dev-bin libc6-dev
  libcaca0 libcaf-openmpi-3 libcairo-gobject2 libcairo2 libcanberra-gtk-module
  libcanberra-gtk0 libcanberra-gtk3-0 libcanberra-gtk3-module libcanberra0
  libcap2 libcbor0.6 libcc1-0 libcdio-cdda2 libcdio-paranoia2 libcdio18
  libchromaprint1 libcoarrays-dev libcoarrays-openmpi-dev libcodec2-0.9
  libcolord2 libcppunit-1.15-0 libcppunit-dev libcrypt-dev libcryptsetup12
  libctf-nobfd0 libctf0 libcups2 libcurl3-gnutls libcurl4 libcurl4-openssl-dev
  libdatrie1 libdbus-1-3 libdc1394-22 libdconf1 libdevmapper1.02.1
  libdouble-conversion3 libdpkg-perl libdrm-amdgpu1 libdrm-common
  libdrm-intel1 libdrm-nouveau2 libdrm-radeon1 libdrm2 libedit2 libegl-mesa0
  libegl1 libelf1 libepoxy0 liberror-perl libevdev2 libevent-2.1-7
  libevent-core-2.1-7 libevent-dev libevent-extra-2.1-7 libevent-openssl-2.1-7
  libevent-pthreads-2.1-7 libexpat1 libexpat1-dev libfabric1 libfakeroot
  libfdk-aac-dev libfdk-aac1 libfftw3-bin libfftw3-dev libfftw3-double3
[... 73 lines cut ...]
  mpi-default-dev netbase networkd-dispatcher ocl-icd-libopencl1 openmpi-bin
  openmpi-common openssh-client openssl osmo-sdr patch perl perl-modules-5.30
  pinentry-curses pkg-config publicsuffix python-matplotlib-data python2
  python2-minimal python2.7 python2.7-minimal python3 python3-bs4
  python3-cairo python3-certifi python3-chardet python3-click
  python3-click-plugins python3-colorama python3-cycler python3-dateutil
  python3-dbus python3-decorator python3-dev python3-distutils python3-gi
  python3-gi-cairo python3-html5lib python3-idna python3-kiwisolver
[... cut ...]
  soapyosmo-common0.7 soapysdr0.7-module-airspy soapysdr0.7-module-all
  soapysdr0.7-module-audio soapysdr0.7-module-bladerf
  soapysdr0.7-module-hackrf soapysdr0.7-module-lms7 soapysdr0.7-module-osmosdr
  soapysdr0.7-module-redpitaya soapysdr0.7-module-remote
  soapysdr0.7-module-rtlsdr soapysdr0.7-module-uhd sound-theme-freedesktop sox
  systemd systemd-sysv systemd-timesyncd tk8.6-blt2.5 ttf-bitstream-vera
  tzdata ubuntu-mono ucf udev uhd-host va-driver-all vdpau-driver-all
  x11-common xauth xdg-user-dirs xkb-data xz-utils zlib1g-dev
```

jquagga, 2020-06-21:
Ok, question on this one.  I was trying to help someone in the gitter get docker going on their pi which meant they were trying to use the updated guide in wiki.  I've been using the "legacy" trun-recorder-docker image but thought I'd give the new spin I go.  Using the default docker-compose now listed on the wiki (the one which runs as nobody and doesn't build a new image) with the :edge image as of today tosses an error that it cannot find config.json.

I think this is because the changes to the docker image now put the recorder binary into /recorder, but does not pass a ```--config=/app/config/config.json```.  So when the :edge image starts up, it looks for config.json in /config.json but it's not there in the example docker-compose.yml.  So I think either the docker-compose.yml on the wiki may need it's path to config.json adjusted, or the Dockerfile could use the ```--config=/app/config/config.json``` of CMD restored.  I don't think it matters which way it goes, they just have to line up.  

I have another docker image next to this one running audioplayer.php so I have to adjust paths in that to match the new docker setup.  Before I do that, I wanted to confirm which way this was going to go.

robbiet480, 2020-06-21:
Frick. Good catch @jquagga. Will submit a PR to hardcode looking for config at `/app/config.json` (really no reason for it to be `/app/config/config.json` I think, but open to discussion!)

Good reminder for me btw: @robotastic You still need to cut a new release of `trunk-recorder` so that the `latest` tag gets uploaded to Docker Hub. You should do that _after_ merging https://github.com/robotastic/docker-gnuradio/pull/4 and cutting a release there so that it also uploads the `latest` tag and `trunk-recorder` Docker gets built with GR3.8. (Also don’t forget to set the same Docker secrets on the docker-gnuradio repo too...) Thanks!

robbiet480, 2020-06-24:
@robotastic To get TR build going, ou just need to cut a 1.0 of gnuradio repo then restart the trunk-recorder build. Its trying to grab `:latest` tag but that tag doesn't exist yet.

closed 2020-08-04 by robotastic

hayden-t, 2021-03-24:
wheres the armv7 docker image at ?

robbiet480, 2021-03-25:
> wheres the armv7 docker image at ?

It's set to build but for some reason failing when building on top of the GNURadio image. Looking into it now.

robbiet480, 2021-03-25:
We're being bit by [this issue](https://gitlab.kitware.com/cmake/cmake/-/issues/20568) in CMake.

hayden-t, 2021-03-25:
any old docker images i can get ?

robbiet480, 2021-03-25:
Nope, armv7 was never properly built. Will have a fix late today probably.

hayden-t, 2021-03-25:
oh ok, that would be neat ! thanks

robbiet480, 2021-03-26:
@hayden-t @robotastic just merged #448 so in the next hour or so you should see valid armv7 images live on his Docker Hub as v3.3.2.

robbiet480, 2021-03-26:
@hayden-t `edge` is now live, @robotastic hasn't cut 3.3.2 yet so give `edge` a try and let me know! https://hub.docker.com/layers/robotastic/trunk-recorder/edge/images/sha256-835abe8b2103b8e8166faa3f63374606749eb5a34d21d8ca19460f1ee12f4280?context=explore

hayden-t, 2021-03-27:
ok i got it, and tried to run it, i will open support ticket seperately

hayden-t, 2021-03-28:
can you help https://github.com/robotastic/trunk-recorder/issues/450

jp83, 2021-04-30:
Sorry to tack on here, but I can't get docker usb permissions to work with the latest Docker images.  I have my own really old Docker build based on previous image that I overlaid and copied in my own config a year or two ago, but something underlying must have changed.  I previously had --privileged -v /dev/bus/usb:/dev/bus/usb.  I've tried --device /dev/bus/usb  and --device-cgroup-rule 'a *:* rwm'.  I keep getting the usual:

Using device #0 Generic RTL2832U OEM
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules

I understand from the FAQ to git rtl-sdr and build it, but shouldn't that be built into the image from Dockerfile?  What am I missing?  I happen to be running on a small thin client machine with Rancher OS, lightweight operating system.  For reference I just got openwebrx docker image running with my rtl-sdr out-of-the-box.  Please help.  Thanks.

leee, 2021-04-30:
Even though this issue was about Docker, the purpose and original use of this issue was for growing Docker support, not providing support on Docker issues. The comments in March probably could've been a new thread, so lets keep closed issues closed.

@jp83 as for your issue, I think it deserves a new issue of its own - would you be willing to create one? I also think referencing which openwebrx image you use (and the Dockerfile for it) could also be especially helpful for @robbiet480 or whoever wants to take up the fix.

Thanks!

jp83, 2021-04-30:
Understand, opened #461.