TrunkRecorder/trunk-recorder #368: Log Motorola P25 Patches

dotsam, 2020-06-09:
There's been code kicking around for 5 years now in p25_parser that can do basic logging of Motorola P25 patch messages that come through, but I'm hoping to be able to get these into the json output from Trunk Recoder.

https://github.com/robotastic/trunk-recorder/blob/2263cd224a43c8046a8d433f05a2471ed9f48c93/trunk-recorder/systems/p25_parser.cc#L241

There's code in boatbod's python app that seem to implement this already

https://github.com/boatbod/op25/blob/6f106847d78f8bedca805e39b2704195a03d7f9c/op25/gr-op25_repeater/apps/trunking.py#L262-L300 and https://github.com/boatbod/op25/blob/6f106847d78f8bedca805e39b2704195a03d7f9c/op25/gr-op25_repeater/apps/trunking.py#L446-L453

These patch messages seem to continuously stream through on the control channel, so there is seemingly a need to keep our own internal mapping of them, and add/update/remove them as they come through, and also expire them in cases where a "remove" message might be missed. This list will have to be referenced at the start of a new call, and if there's a patch active, the list of patched TGs attached.

Unfortunately, this all lies right at the edge of my C++ skills, so I was wondering if there isn't someone else who could take a run at this.

As for getting the data out into the call JSON, there's a under/unused `meta` property of the call object that could work perhaps?

kcwebby, 2020-06-09:
I second this, but add that we need to consider removal of the `Moto`  As Harris and other P25 Controllers probably don't like using the batwing name.  We should probably look up and find whatever the P25 Standard Code message is, and decode it there.

Another, probably more significant hurdle:
Since patch assignments are not done with every transmission, there would have to be some level of patch tracking added to TR to be able to follow patches.   Depending on if a unit keys up on one of the primary TG's that make up the patch, or on the patch TG itself, the Channel Grant message could look different.   I think it could do this, and it should.  It would help save recorders by only recording a transmission once, and then applying both TGID's to the json file.   

I could be wrong, its been a while since i looked.

Patch Grants happen (and stay until they are deleted)

A transmission on a TGID that's involved in a patch results in seperate Channel Grant messages with the Patch ID, and Each TGID that makes up the patch.   The grants are all to the same frequency, so essentially without a way to handle patches, if you have 5 TGID's in a patch, you are recording the same transmission 6 times.

aaknitt, 2021-10-15:
Thirding this.  Necessary feature to follow traffic on patched systems and systems that use patching for paging.  If anyone is willing to grant remote access on a system where patching is common I can start looking into this more.

robotastic, 2021-10-20:
yea - I also have to re-think a lot of the internal logic. Calls are separated by a single TG ID. Do patch calls always come in with a single Primary TG ID for the life of the patch? Or do they go out simultaniously on each of the TGs? In the example above, would there also be a transmission for TG 58911 that should be ignored because it is already being capture on the TG recording for 58909?

aaknitt, 2021-11-28:
Here's what I'd like to propose as a starting point for this.  If @robotastic  thinks this is an OK approach I can get started on it:

- Add a new MessageType enumeration (PATCH_ADD=10 ??) and some type of container to hold the decoded patch data to the TrunkMessage class (in parser.h) to allow patching info get get passed back up within the TrunkMessage class from p25_parser.cc
- Add a case in main.cc -> handle_message() to handle the patching messages (probably also add a call to this in the main loop to allow it to clear patches based on a timeout)
- Add a data structure of some type to the System class that would contain a list of the current patches on the system.  This list would be updated by new case in the handle_message() mentioned above.
- Should probably also figure out a good way to pass this data to the plugin system

Implementing what's proposed here wouldn't actually _do_ anything with the patching information yet, but once it's tracked by the System class it could then be used to add to the JSON, fire up recorders based on patching, etc.  One step at a time though.

robotastic, 2021-11-29:
@aaknitt This sounds like a solid approach to me. I also like that it will build things up slowly. We can get this out there and then get reports of what people are seeing on the local systems. Tracking and owning it at the system level makes sense. We could also add a bit to the Talkgroup Class too. If you are able to take a swing at this, I can help integrate it and clean things up.

aaknitt, 2021-11-30:
I've got a PR in for this first phase, feel free to suggest any changes or improvements and I can modify accordingly.  I think the next baby steps are to:

- Get a place for the patch data into the Call_Data_t structure
- Populate  the patch data into that structure in the Call_Concluder::create_call_data() function (from the System data)
- Get the patch data into the JSON file

I'll work on those next.

aaknitt, 2021-11-30:
That wasn't too hard...added to the PR so that the data makes it into the JSON file.  

Up next:

- Start a recorder in the scenario where the supergroup wouldn't normally get recorded (due to priority) but one of the talkgroups in the patch would
- Perhaps add an option to create duplicate audio files for patched talkgroups for use by uploadScripts or APIs?  Not sure how I feel about this...seems wasteful, but probably can't expect every ingesting service to handle the new JSON Data?

aaknitt, 2021-11-30:
To get trunk-recorder to start up a recorder based on patch data, here's what I'm thinking:

- In main.cc -> start_recorder(), check to see if the talkgroup is part of a current patch on the system
- If it is, find the talkgroup in the patch with the highest priority and use that priority for the next steps of determining whether the recording should be started or not.
- Two options at this point:  
  - Modify the priority of the "supergroup" talkgroup that's actually being sent over the air and use that talkgroup's info for recording (TGID, tag, etc.)
  - Use the higher priority talkgroup's (from the patch) information for recording (TGID, tag, etc.)

I think I like the first option a little better since it's the least amount of change to how trunk-recorder currently operates, but I think either would work.  Both cases require modification of the talkgroup data contained in the Call class which will require addition of some interfaces (setters) that don't currently exist in those classes.  

Regardless of which TGID is used for recording purposes (filename, tag, etc.), the JSON associated with the recording will have the patch info.  

Thoughts?

aaknitt, 2021-12-05:
> To get trunk-recorder to start up a recorder based on patch data, here's what I'm thinking:
>
> * In main.cc -> start_recorder(), check to see if the talkgroup is part of a current patch on the system [...]

Decided the first option is definitely a lot better for a number of reasons, so implemented that in the latest commit.

netsound, 2021-12-05:
> That wasn't too hard...added to the PR so that the data makes it into the JSON file.
>
> Up next:
>
>     * Start a recorder in the scenario where the supergroup wouldn't normally get recorded (due to [...]

I would really like the option to have it record files for each of the patched talkgroups.  I have a watcher looking for certain talkgroups getting recorded and sending the file after they are recorded.  IE Paging talkgroups..

aaknitt, 2021-12-05:
> I would really like the option to have it record files for each of the patched talkgroups. I have a watcher looking for certain talkgroups getting recorded and sending the file after they are recorded. IE Paging talkgroups..

This can definitely be handled by an uploadScript.  You could set up the script to check the JSON of each file that gets created and if the JSON includes a talkgroup you care about in the list of patched talkgroups, you can then handle as you please.

Having trunk-recorder itself create a separate audio file for each talkgroup in the patch is probably also possible but I think we'd definitely want it to be an option since that's probably not desired default behavior when uploading to OpenMHz, rdio-scanner, etc...would result in a whole bunch of duplicates being played back to back.

aaknitt, 2022-02-08:
Yeah my understanding is that Harris patches allocate a batch of TGIDs to use as the "supergroup" ID for a patch, whereas Motorola just uses one of the TGIDs that's in the patch as the "supergroup" ID.  

The trunk-recorder code already had parsing for the Motorola patch messages on the control channel built in under the hood, so using those messages to set up the current functionality was low-hanging fruit.  The control channel messages for Motorola patching use the Motorola manufacturer ID, so they're not standardized P25 messages, but part of the message set reserved for individual manufacturers to use for their own purposes.  

If Harris is using a different manufacturer-specific set of messages for system patches (as it would appear if you're saying that the current functionality isn't working on Harris systems), there will be some detective work necessary to tease out what those control channel messages are and get them parsed and added to the patch tracking code.  

I don't have any Harris systems near me.  In order to try to take a stab at figuring out their control channel messaging for system-level patches, I'd need either remote access to a trunk-recorder setup that's monitoring a Harris system or an IQ recording of control channel traffic on a Harris system when patches are present that I could replay and decode offline.  Or, if there are other open source projects that have Harris patch tracking in place, I might be able to look at their code to try to understand how it works.  

This also assumes that the patches are being done at the system level, not at the console level.  If that patches are done via console patching or multiselect, that's a different animal.  In that case the audio is actually going out on multiple frequencies/talkgroups on the system at the same time, and there wouldn't really be any control channel traffic indicating that a patch is in place.  As far as system controller is concerned it would be similar to keying up two radios on different talkgroups and talking into the mics of both of them at the same time.  Trunk-recorder wouldn't have any knowledge that a patch or multiselect is occurring in that scenario.  

Another question I've had is that if Motorola and Harris each use their own proprietary messaging for patches, how do subscriber units from different manufacturers interoperate on a given system?  I guess the subscriber units from each supplier must be able to handle the patch messages from the other's system?

blantonl, 2022-02-08:
This is using DSD

You can see the patch announcements.  They are ready and interpreted correctly by my APX8000

Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65140  TG=39046
Sync:+P25p1 NAC:981 TSDU        RFSS_STS_BCST        SysID=0D2 LRA=14 RFSS=14 Site=14 CC=770.28125 SC=70
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65140  TG=39045
Sync:+P25p1 NAC:981 TSDU        NET_STS_BCST         WACN ID=91F78 SysID=0D2 LRA=14 CC=770.28125 SC=70
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65140  TG=39047
Sync:+P25p1 NAC:981 TSDU        IDEN_UP              ID=0 Base=769.00625 Offset=+30 Spacing=12.5 BW=12.5
Sync:+P25p1 NAC:981 TSDU CRC FAIL
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65055  TG=38625
Sync:+P25p1 NAC:981 TSDU        SYS_SRV_BCST         SSA=2C2C01 SSS=2C2C01 RPL=1
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65055  TG=38626
Sync:+P25p1 NAC:981 TSDU        ADJ_STS_BCST         SysID=0D2 LRA=11 RFSS=11 Site=11 CC=769.10625 SC=70 CFVA=3: Networked
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65054  TG=38627
Sync:+P25p1 NAC:981 TSDU        RFSS_STS_BCST        SysID=0D2 LRA=14 RFSS=14 Site=14 CC=770.28125 SC=70
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65054  TG=38628
Sync:+P25p1 NAC:981 TSDU        NET_STS_BCST         WACN ID=91F78 SysID=0D2 LRA=14 CC=770.28125 SC=70
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65054  TG=38632
Sync:+P25p1 NAC:981 TSDU        IDEN_UP              ID=0 Base=769.00625 Offset=+30 Spacing=12.5 BW=12.5
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65050  TG=38629
Sync:+P25p1 NAC:981 TSDU        SYS_SRV_BCST         SSA=2C2C01 SSS=2C2C01 RPL=1
Sync:+P25p1 NAC:981 TSDU        U_REG_RSP            SysID=0D2 SID=6042847 Src=6042847 Resp=FAIL
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65050  TG=38630
Sync:+P25p1 NAC:981 TSDU        ADJ_STS_BCST         SysID=0D2 LRA=12 RFSS=12 Site=12 CC=773.93125 SC=70 CFVA=3: Networked
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65050  TG=38631
Sync:+P25p1 NAC:981 TSDU        RFSS_STS_BCST        SysID=0D2 LRA=14 RFSS=14 Site=14 CC=770.28125 SC=70
Sync:+P25p1 NAC:981 TSDU        GRP_AFF_RSP          (Global)  ATG=1 TG=38529 RID=6042847 Resp=FAIL
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65146  TG=39345
Sync:+P25p1 NAC:981 TSDU        NET_STS_BCST         WACN ID=91F78 SysID=0D2 LRA=14 CC=770.28125 SC=70
3259 group records saved; 1953 aliases
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65146  TG=39814
Sync:+P25p1 NAC:981 TSDU        IDEN_UP              ID=0 Base=769.00625 Offset=+30 Spacing=12.5 BW=12.5
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65144  TG=39359
Sync:+P25p1 NAC:981 TSDU        SYS_SRV_BCST         SSA=2C2C01 SSS=2C2C01 RPL=1
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65144  TG=39357
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT       PRI0 TG=38273 RID=6013221 ch=769.11875
Sync:+P25p1 NAC:981 TSDU        ADJ_STS_BCST         SysID=0D2 LRA=13 RFSS=13 Site=13 CC=769.35625 SC=70 CFVA=3: Networked
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT       PRI0 TG=38273 RID=6013221 ch=769.11875
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65140  TG=39046
Sync:+P25p1 NAC:981 TSDU        RFSS_STS_BCST        SysID=0D2 LRA=14 RFSS=14 Site=14 CC=770.28125 SC=70
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=38273 ch=769.11875  /  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65140  TG=39045
Sync:+P25p1 NAC:981 TSDU        NET_STS_BCST         WACN ID=91F78 SysID=0D2 LRA=14 CC=770.28125 SC=70
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=38273 ch=769.11875  /  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65140  TG=39047
Sync:+P25p1 NAC:981 TSDU        IDEN_UP              ID=0 Base=769.00625 Offset=+30 Spacing=12.5 BW=12.5
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=38273 ch=769.11875  /  TG=39877 ch=774.65625
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65055  TG=38625
Sync:+P25p1 NAC:981 TSDU        SYS_SRV_BCST         SSA=2C2C01 SSS=2C2C01 RPL=1
Sync:+P25p1 NAC:981 TSDU        GRP_V_CH_GRANT_UPDT  TG=38273 ch=769.11875  /  TG=39877 ch=774.65625
[... cut ...]
Sync:+P25p1 NAC:981 TSDU        ADJ_STS_BCST         SysID=0D2 LRA=13 RFSS=13 Site=13 CC=769.35625 SC=70 CFVA=3: Networked
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65140  TG=39046
Sync:+P25p1 NAC:981 TSDU        RFSS_STS_BCST        SysID=0D2 LRA=14 RFSS=14 Site=14 CC=770.28125 SC=70
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65140  TG=39045
Sync:+P25p1 NAC:981 TSDU        NET_STS_BCST         WACN ID=91F78 SysID=0D2 LRA=14 CC=770.28125 SC=70
Sync:+P25p1 NAC:981 TSDU        M/A-COM_P/SS_ANN     Patch=65140  TG=39047
Sync:+P25p1

aaknitt, 2022-02-08:
Is that using DSD or DSD+?  I'm not seeing where those print statements get generated in the DSD source, and DSD+ is closed source.  I may need to dig deeper into DSD, I just did a brief search. 

If you can grab a ~10 to 20 second WAV IQ recording of that control channel using SDRsharp or whatever, I can probably replay it and work with it, though seeing source code that already has it implemented would be easier.

blantonl, 2022-02-08:
It’s DSD Plus.

Interestingly enough another Harris system in the area is picking up the patches properly using 4.3

I’ll grab some IQ of this control channel.  Stay tuned.

blantonl, 2022-02-08:
See attached

https://www.lcblanton.com/recordings77028125.wav

aaknitt, 2022-02-09:
Thanks. Probably won’t get to this for a couple days, but I think I found what we need in the boatbod op25 code. Hopefully between that and this recording I can get something put together.

blantonl, 2022-02-09:
Sounds good – keep me posted on what you need on my end.

aaknitt, 2022-02-09:
Here's a branch for initial testing:

https://github.com/aaknitt/trunk-recorder/tree/macom-patch-follow

I haven't had luck getting the IQ recording to decode at all, I'm likely doing something wrong with the playback.  I'll mess with it some more when I get a chance.  

If this branch works, I'll need to do some cleanup and additions before submitting a PR, but this is a start to try it out anyway.

aaknitt, 2022-02-09:
@blantonl  are you sure that recording is of a P25 control channel?  It doesn't look or "sound" like the ones I have around here, but maybe it's a Phase 1 vs. Phase 2 thing?  

Recording:
[image]

Local Control Channel:
[image]

The sample rate of the recording (48 kHz) also seems lower than most baseband recordings (which would be at the sample rate of the SDR device being used).  A baseband recording would probably be better/easier to feed into trunk-recorder.

blantonl, 2022-02-09:
OK, so far this is looking good on this newer Harris system that is patched to an EDACS system.  I’m getting a list of the super patch and the associated patches in the attribute correctly for this Harris system.  This will allow me to go ahead and setup priority patch talkgroups based on what’s in that patched_talkgroups attribute field.

However, note that this use case on a different Harris system didn’t seem to work.  It reported the call on 65325 in Trunk-Recorder.

[Text  Description automatically generated]

Also, I can’t be sure – but it appears that there might be a problem with Source IDs getting confused or overwritten in memory as they are captured.  I might be chasing something random here, but something has appeared off with src IDs getting randomly mixed up.

Note my system is pretty complex, it’s capturing a bunch of different conv, convp25, 400 MHz Motorola, 150 MHz P25 Motorola, and Harris 700’s

https://github.com/aaknitt/trunk-recorder/tree/macom-patch-follow

blantonl, 2022-02-09:
Ok, the differences are in this screenshot

I’m seeing the following:

Patch/Simulselect Supergroup with associated subgroups
Patch Supergroup with associated subgroups

The Harris patch you just fixed in the macom-patch-follow branch seemed to fix Patch/simulselect Supergroup with associated subgroups

[Graphical user interface, text  Description automatically generated]
--
Lindsay C. Blanton III
CEO  - RadioReference.com / Broadcastify
Mobile:  +1 210-593-8685
Email:    ***@***.******@***.***>

https://github.com/aaknitt/trunk-recorder/tree/macom-patch-follow

aaknitt, 2022-02-10:
Interesting.  At least we're making progress.  Can you post a trunk-recorder log with the debug level set to "debug" in config.json?  Once set to debug, there should be some print statements that may help, and I may add some more if needed.

blantonl, 2022-02-10:
Which version should I use? The Macom test branch or prod release for logging?

blantonl, 2022-02-10:
These threads might have some insight 

https://forums.radioreference.com/threads/p25-patched-talkgroups.416095/
https://forums.radioreference.com/threads/p25-patch-supergroup-formula.344082/

aaknitt, 2022-02-10:
> Which version should I use? The Macom test branch or prod release for logging?

The Macom test branch

aaknitt, 2022-02-10:
Ok thanks.  A couple interesting things going on here, but I want to make sure I fully understand what I'm looking at.  

So this these log files are from two different systems, one of which (the 03

aaknitt, 2022-02-10:
Ok thanks.  I have an idea of what the issue might be with the short-duration patches, I'll take a swing at that.  

The patch following code doesn't touch source IDs at all, just which TGIDs are connected to each other.  It doesn't surprise me that there may be an issue with source IDs getting lost or messed up through an intersystem patch, but I think that likely warrants a separate issue.

blantonl, 2022-02-10:
Problem is I don't know what version of Trunk Recorder that node is running at.  They may be running at a version that added patch support in.... I'm going to do more testing.  Stay tuned.

aaknitt, 2022-02-11:
@blantonl I just pushed cleaned-up code to the test branch that also includes some checks for "0" talkgroups.  If you can grab that and give it a try to confirm that I didn't break anything during clean up and that the "0" issue is fixed, I can then submit a PR to get this code into the main repo/branch.

blantonl, 2022-02-11:
So far so good, except I'm still seeing some 0 talkgroups show up:

```
"patched_talkgroups": [0,11016,11017,11018,11019,11020,65347],
"patched_talkgroups": [0,11016,11017,11018,11019,11020,65347],
"patched_talkgroups": [0,8033,8034,65348],
"patched_talkgroups": [0,8033,8034,65348],
"patched_talkgroups": [8005,8033,8034,65332],
"patched_talkgroups": [8005,8033,8034,65332],
"patched_talkgroups": [8005,8033,8034,65332],
"patched_talkgroups": [8005,8033,8034,65332],
"patched_talkgroups": [8005,8033,8034,65332],
"patched_talkgroups": [8005,8033,8034,65332],
"patched_talkgroups": [8005,8033,8034,65332],
"patched_talkgroups": [8005,8033,8034,65332],
"patched_talkgroups": [8005,8033,8034,65332],
"patched_talkgroups": [8005,8033,8034,65332],
"patched_talkgroups": [0,8033,8034,65348],
```

aaknitt, 2022-02-12:
@blantonl can you post a "debug" level log file from when the "0" talkgroups were showing up?

aaknitt, 2022-02-12:
Also, are the "0" entries only happening on one of those two systems, or on both?

blantonl, 2022-02-13:
It's happening on the Harris system that uses the Motorola Patch method.  

https://www.radioreference.com/db-beta/sid/6587

Is there a flag I can run with TR to verify which branch I'm running and and code version so I can make I'm actually running your latest branch code?  I'll see about getting a debug dump today.

aaknitt, 2022-02-13:
> Is there a flag I can run with TR to verify which branch I'm running and and code version so I can make I'm actually running your latest branch code (or does it show on startup somewhere?)? 

I'm not sure there is, I think there's an open issue to try to get something like that in but I don't think it's been implemented yet.  

You can check the trunk-recorder/systems/system.cc file, you should see checks to prevent "0" talkgroups starting at line 485:
```
      if (0 != patch_data.sg){
        patch.second[patch_data.sg] = update_time;
      }
```
That'll be an indication that you have the latest _source_, but not that you've built with it.  If needed I can push a debug printout to ensure you're running my latest test code but won't be able to do that until tonight.

blantonl, 2022-02-13:
I'll check my build branch download and make sure that code is there.  Stay tuned.

blantonl, 2022-02-13:
I checked and I'm using the correct codebase so that's good.  

The systems are pretty quiet today, so I'll try to setup a dump log tomorrow on a separate machine to see if I can capture this behavior.  Stay tuned.

aaknitt, 2022-02-17:
@blantonl any update on this?  I don't want to submit a pull request if we still have a lurking bug out there.

blantonl, 2022-02-18:
Sorry, I've gotten so busy with personal items that I haven't had a chance to run the trace. I'll try to get a trace running before I begin some travel next week.  Stay tuned.

blantonl, 2022-02-19:
This debug level log file(s) should have captured the following patches with a zero 

"patched_talkgroups": [0,8033,8034,65350],
"patched_talkgroups": [0,8033,8034,65350],
"patched_talkgroups": [0,8033,8034,65350],
"patched_talkgroups": [0,12081,12098,65339],
"patched_talkgroups": [0,8033,8034,65350],

https://www.lcblanton.com/transfer/2021-02-19.tar.gz

aaknitt, 2022-02-19:
Thanks, that helped me find the typo causing the issue.  I'll get it fixed and get a PR submitted.

blantonl, 2022-02-19:
Excellent news.  This is good work for the community.  Nice work Andy.

robotastic, 2022-02-27:
Closing this out - I think we are good! Thanks @aaknitt

closed 2022-02-27 by robotastic