TrunkRecorder/trunk-recorder #369: Add title tag to m4a files?

dmonies327, 2020-06-09:
Is it possible to add a title tag to the m4a files created when uploading to openMHZ/broadcastify calls? I'd like to use these files with liquidsoap to stream to broadcastify as they are already balanced and there's no need to repackage them if liquidsoap can play/stream the m4a/aac files. I'd like them to have the TITLE tag on them, however.
If this isn't practical, does anyone know a good utility to quickly add the tag to the file that could be run in a separate script (similar to the way the encode_upload.sh does in the TR/LiquidSoap wiki writeup on RR)? I tried using id3v2 to add it, but it appears to just corrupt the files.

EricTendian, 2020-06-19:
There isn't that functionality currently available within trunk-recorder itself, since in https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/uploaders/call_uploader.cc#L20 we are just converting the wav to a m4a. However, it looks like fdkaac does have an argument `--title` so it probably would be possible to modify the upload logic to include whatever alpha tag is associated with the talkgroup if it exists.

robotastic, 2020-08-04:
This is a good idea and should be a pretty easy change. I will add this to my list of things.

natecarlson, 2020-11-26:
What content do you want in the tag?

Or in the ideal world should it be configurable, with placeholders for each of the major variables?

dmonies327, 2020-11-26:
The alpha-tag associated with the talk group would be ideal, but
configurable might be more flexible for the community.

robotastic, 2024-12-14:
I really have to add this - I thought I did at one point but I don't see it in the code

kd4e, 2024-12-14:
> I really have to add this - I thought I did at one point but I don't see it in the code

:+1:

robotastic, 2024-12-19:
Ok - it looks like it is working. It is in the dev/backlog branch, and will be merged in soon. Eventually I will look at how to create custom sox / ffmpeg so you can change parameters around.

closed 2024-12-19 by robotastic