TrunkRecorder/trunk-recorder #370: OSX fails at [100%] Linking CXX executable recorder

h00fty, 2020-06-11:
Tried compiling the latest build to be compatible with Broadcastify Calls and it fails on the final step..

attempted Brew and Macports. same issue on both.

`[100%] Linking CXX executable recorder
Undefined symbols for architecture x86_64:
  "boost::re_detail_107100::raw_storage::insert(unsigned long, unsigned long)", referenced from:
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_repeat(unsigned long, unsigned long) in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_alt() in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_perl_extension() in call_uploader.cc.o
  "boost::re_detail_107100::raw_storage::resize(unsigned long)", referenced from:
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse(char const*, char const*, unsigned int) in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_extended() in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_basic() in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_creator<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::finalize(char const*, char const*) in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_open_paren() in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_extended_escape() in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_repeat(unsigned long, unsigned long) in call_uploader.cc.o
      ...
  "boost::re_detail_107100::get_mem_block()", referenced from:
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_imp() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_all_states() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_startmark() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_alt() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_rep() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_char_repeat() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_set_repeat() in call_uploader.cc.o
      ...
  "boost::re_detail_107100::put_mem_block(void*)", referenced from:
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_imp() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::unwind_extra_block(bool) in call_uploader.cc.o
  "boost::re_detail_107100::verify_options(unsigned int, boost::regex_constants::_match_flags)", referenced from:
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_imp() in call_uploader.cc.o
  "boost::re_detail_107100::raise_runtime_error(std::runtime_error const&)", referenced from:
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::init() in call_uploader.cc.o
      void boost::re_detail_107100::raise_error<boost::regex_traits_wrapper<boost::regex_traits<char, boost::cpp_regex_traits<char> > > >(boost::regex_traits_wrapper<boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::error_type) in call_uploader.cc.o
  "boost::re_detail_107100::get_default_error_string(boost::regex_constants::error_type)", referenced from:
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::init() in call_uploader.cc.o
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::error_string(boost::regex_constants::error_type) const in call_uploader.cc.o
  "boost::re_detail_107100::cpp_regex_traits_char_layer<char>::init()", referenced from:
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::cpp_regex_traits_implementation(boost::re_detail_107100::cpp_regex_traits_base<char> const&) in call_uploader.cc.o
  "boost::re_detail_107100::lookup_default_collate_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::lookup_collatename(char const*, char const*) const in call_uploader.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [recorder] Error 1
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
make: *** [all] Error 2
`

robotastic, 2020-06-11:
Hmm... I just tried on my MacBook using MacPort and it worked fine. Are you sure Boost is installed?

h00fty, 2020-06-11:
Yes boost is installed. I followed the procedure again and received the same result. 

```
OLD-MBP:~ josh$ sudo port install gnuradio uhd gr-osmosdr
--->  Computing dependencies for gnuradio
--->  Cleaning gnuradio
--->  Computing dependencies for uhd
--->  Cleaning uhd
--->  Computing dependencies for gr-osmosdr
--->  Cleaning gr-osmosdr
--->  Scanning binaries for linking errors
--->  No broken files found.                             
--->  No broken ports found.
```

```
OLD-MBP:~ josh$ sudo port install cmake boost uhd libusb cppunit
--->  Computing dependencies for cmake
--->  Cleaning cmake
--->  Computing dependencies for boost
--->  Cleaning boost
--->  Computing dependencies for uhd
--->  Cleaning uhd
--->  Cleaning libusb
--->  Cleaning cppunit
--->  Scanning binaries for linking errors
--->  No broken files found.
--->  No broken ports found.
```

```
OLD-MBP:~ josh$ mkdir trunk-build
OLD-MBP:~ josh$ git clone https://github.com/robotastic/trunk-recorder.git
Cloning into 'trunk-recorder'...
remote: Enumerating objects: 101, done.
remote: Counting objects: 100% (101/101), done.
remote: Compressing objects: 100% (73/73), done.
remote: Total 7372 (delta 46), reused 52 (delta 26), pack-reused 7271
Receiving objects: 100% (7372/7372), 3.57 MiB | 1.06 MiB/s, done.
Resolving deltas: 100% (5215/5215), done.
OLD-MBP:~ josh$ cd trunk-build
OLD-MBP:trunk-build josh$ cmake ../trunk-recorder
-- The CXX compiler identification is AppleClang 9.1.0.9020039
-- The C compiler identification is AppleClang 9.1.0.9020039
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
CMake Warning (dev) at /opt/local/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (GnuradioRuntime).  This can
  lead to problems in calling code that expects `find_package` result
  variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /opt/local/share/cmake-3.17/Modules/FindPkgConfig.cmake:45 (find_package_handle_standard_args)
  cmake/Modules/FindGnuradioRuntime.cmake:1 (INCLUDE)
  CMakeLists.txt:69 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PkgConfig: /opt/local/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'gnuradio-runtime'
[... 421 lines cut ...]
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_rep() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_char_repeat() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_set_repeat() in call_uploader.cc.o
      ...
  "boost::re_detail_107100::put_mem_block(void*)", referenced from:
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_imp() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::unwind_extra_block(bool) in call_uploader.cc.o
  "boost::re_detail_107100::verify_options(unsigned int, boost::regex_constants::_match_flags)", referenced from:
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_imp() in call_uploader.cc.o
  "boost::re_detail_107100::raise_runtime_error(std::runtime_error const&)", referenced from:
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::init() in call_uploader.cc.o
      void boost::re_detail_107100::raise_error<boost::regex_traits_wrapper<boost::regex_traits<char, boost::cpp_regex_traits<char> > > >(boost::regex_traits_wrapper<boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::error_type) in call_uploader.cc.o
  "boost::re_detail_107100::get_default_error_string(boost::regex_constants::error_type)", referenced from:
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::init() in call_uploader.cc.o
[... cut ...]
  "boost::re_detail_107100::lookup_default_collate_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::lookup_collatename(char const*, char const*) const in call_uploader.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [recorder] Error 1
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
make: *** [all] Error 2
```

blantonl, 2020-07-18:
I'm seeing this behavior as well..

blantonl, 2020-07-18:
From what I can tell, it looks like a PMT Library linking problem specific to Mac OS X and Gnuradio

@robotastic robotastic - maybe you need to update your macports gnuradio install to reproduce?  

For the record, I'm using Brew, and I've also tried to revert to previous pulls/commits to no avail.

robotastic, 2020-07-18:
good call - I will do a MacPort update and see if I can reproduce it.

blantonl, 2020-07-18:
So, an update here.

my issue was actually I was missing pkgconfig as a prereq - it's not listed in the prereqs and after installing (brew install pkgconfg) it compiled just fine.

So, I'll see about updating the wiki to reflect that prepreq for mac os x

robotastic, 2020-07-18:
Ah cool!  yea, I just did a Port update and everything recompiled fine, so this makes sense.

NoahJaehnert, 2020-07-31:
Issue appears to still exist for Macports. 

`$ make
Scanning dependencies of target imbe_vocoder
[  1%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/aux_sub.cc.o
[  2%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/basicop2.cc.o
[  3%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/ch_decode.cc.o
[  4%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/ch_encode.cc.o
[  5%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/dc_rmv.cc.o
[  6%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/decode.cc.o
[  7%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/dsp_sub.cc.o
[  8%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/encode.cc.o
[  9%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/imbe_vocoder.cc.o
[ 10%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/math_sub.cc.o
[ 11%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/pe_lpf.cc.o
[ 12%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/pitch_est.cc.o
[ 13%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/pitch_ref.cc.o
[ 14%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/qnt_sub.cc.o
[ 15%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/rand_gen.cc.o
[ 16%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/sa_decode.cc.o
[ 17%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/sa_encode.cc.o
[ 18%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/sa_enh.cc.o
[ 19%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/tbls.cc.o
[ 20%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/uv_synt.cc.o
[ 21%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/v_synt.cc.o
[ 22%] Building CXX object lib/op25_repeater/lib/imbe_vocoder/CMakeFiles/imbe_vocoder.dir/v_uv_det.cc.o
[ 23%] Linking CXX static library libimbe_vocoder.a
[ 23%] Built target imbe_vocoder
Scanning dependencies of target gnuradio-op25_repeater
[ 24%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/ambe_encoder_sb_impl.cc.o
[ 25%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/dmr_bs_tx_bb_impl.cc.o
[ 26%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/ysf_tx_sb_impl.cc.o
[ 27%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/dstar_tx_sb_impl.cc.o
[ 28%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/vocoder_impl.cc.o
[ 29%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/gardner_costas_cc_impl.cc.o
[ 30%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25_frame_assembler_impl.cc.o
[ 31%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/fsk4_demod_ff_impl.cc.o
[ 32%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/fsk4_slicer_fb_impl.cc.o
[ 33%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/bch.cc.o
[ 34%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/rs.cc.o
[ 35%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25_framer.cc.o
[ 36%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p1_fdma.cc.o
[ 37%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p1_voice_encode.cc.o
[ 38%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p1_voice_decode.cc.o
[ 39%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_framer.cc.o
[ 40%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_isch.cc.o
[ 41%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_duid.cc.o
[ 42%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_sync.cc.o
[ 43%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_tdma.cc.o
[ 44%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/p25p2_vf.cc.o
[ 45%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/imbe_decoder.cc.o
[ 46%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/software_imbe_decoder.cc.o
[ 47%] Building C object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/ambe.c.o
[ 48%] Building C object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/mbelib.c.o
[ 50%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/ambe_encoder.cc.o
[ 51%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/rx_sync.cc.o
[ 52%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/hamming.cc.o
[... cut ...]
      __GLOBAL__sub_I_vocoder_impl.cc in vocoder_impl.cc.o
      __GLOBAL__sub_I_gardner_costas_cc_impl.cc in gardner_costas_cc_impl.cc.o
      __GLOBAL__sub_I_p25_frame_assembler_impl.cc in p25_frame_assembler_impl.cc.o
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/op25_repeater/lib/libgnuradio-op25_repeater.dylib] Error 1
make[1]: *** [lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/all] Error 2
make: *** [all] Error 2`

robotastic, 2020-08-01:
I am unable to recreate this. Try doing a full rebuild of all your port packages that can sometimes help. Do you have an old version of Xcode command line tools?

NoahJaehnert, 2020-08-01:
Just to clarify I'm using Brew, not MacPorts. So I don't think I can rebuild all port packages. 

Also, Xcode details as follows: 

$ xcode-select --version
xcode-select version 2373.

leee, 2020-08-01:
I just had this build on the following and it seems to run (other than a lot of `DEBUG: writing tag at sample n` and empty wavs, but I suppose one problem at a time):

```
~/src/github.com/robotastic/trunk-recorder(leee-cnv-iterate*) » sw_vers -productVersion
10.15.4
~/src/github.com/robotastic/trunk-recorder(leee-cnv-iterate*) » brew list --versions | grep gnuradio
gnuradio 3.7.13.4_14
```

You problem usually won't be with homebrew, but potentially your formula, linking, etc.

You could debug further by seeing for example if you even have gnuradio libraries where homebrew usually links libraries:

```
$ ls -la /usr/local/lib | grep gnuradio
```

And that should show you symlinks to gnuradio libraries in your Cellar. If there's not much or nothing there, then link the formula:

```
$ brew link gnuradio
```

FWIW though, the state of gnuradio in Homebrew is painful for a variety of reasons. I say this as a Homebrew user who intentionally tries to shy away from MacPorts (and my current macOS install doesn't have MacPorts on it... yet), but if for any reason I were forced to use gnuradio on macOS, I would just use either the bundled project (https://github.com/ktemkin/gnuradio-for-mac-without-macports) or MacPorts if I really had to.

NoahJaehnert, 2020-08-01:
> You could debug further by seeing for example if you even have gnuradio libraries where homebrew usually links libraries:
> ```$ ls -la /usr/local/lib | grep gnuradio```

Tested and all gnuradio libraries were symlinked correctly. 

> FWIW though, the state of gnuradio in Homebrew is painful for a variety of reasons. I say this as a Homebrew user who intentionally tries to shy away from MacPorts (and my current macOS install [...]

Last time I ran into this similar issue (on my Mac Mini), I ended up switching from Homebrew to Macports and that solved the issue for me (See here: https://github.com/robotastic/trunk-recorder/issues/277#issuecomment-582958492). But due to other work/dependencies, I can't really do that on my Mac laptop. 

Appreciate the advice on the full bundled project as well. I may try that route. But I'd really like to get this solved for the Brew side of things, as at least on my machine, the instructions in the Wiki for installing via Homebrew (https://github.com/robotastic/trunk-recorder/wiki/Installation#macos) don't actually work/result in this error :-(

leee, 2020-08-01:
Have you `make clean`'d and freshly `cmake && make`?

I'm not sure if `all`  and `correct`edness can be gauranteed here. For reference, I have 42 symlinks to `/usr/local/Cellar/gnuradio` in `/usr/local/lib`. Is `libgnuradio-pmt.dylib` in there?

Additionally, what about `/usr/local/opt`? What I have on my built `recorder`:

```
~/src/github.com/robotastic/trunk-recorder(leee-cnv-iterate*) » objdump -x recorder | grep gnuradio
0000000000000000         *UND*	__ZN8gnuradio6detail10sptr_magic18fetch_initial_sptrEPN2gr11basic_blockE
         name /usr/local/opt/gnuradio/lib/libgnuradio-runtime.3.7.13.4.dylib (offset 24)
         name /usr/local/opt/gnuradio/lib/libgnuradio-pmt.3.7.13.4.dylib (offset 24)
         name /usr/local/opt/gnuradio/lib/libgnuradio-filter.3.7.13.4.dylib (offset 24)
         name /usr/local/opt/gnuradio/lib/libgnuradio-fft.3.7.13.4.dylib (offset 24)
         name /usr/local/opt/gnuradio/lib/libgnuradio-digital.3.7.13.4.dylib (offset 24)
         name /usr/local/opt/gnuradio/lib/libgnuradio-analog.3.7.13.4.dylib (offset 24)
         name /usr/local/opt/gnuradio/lib/libvolk.1.4.dylib (offset 24)
         name /usr/local/opt/gnuradio/lib/libgnuradio-uhd.3.7.13.4.dylib (offset 24)
         name /usr/local/opt/gnuradio/lib/libgnuradio-blocks.3.7.13.4.dylib (offset 24)
         name /usr/local/opt/gr-osmosdr/lib/libgnuradio-osmosdr.0.1.4.dylib (offset 24)
         name @rpath/libgnuradio-op25_repeater.dylib (offset 24)
```

The instructions might need updating for something (although the question stands as to what that missing factor may be).

Even though the linking failure is related to gnuradio only, the instructions don't explicitly mention installing `boost` even though that's a `depends_on` for gnuradio.

robotastic, 2020-08-05:
Give it another try - I had previous been building with gr3.7, but when Port upgraded to gr3.8, I got the same error.

NoahJaehnert, 2020-08-05:
> Give it another try - I had previous been building with gr3.7, but when Port upgraded to gr3.8, I got the same error.

That did it! Thanks much.

leee, 2020-08-05:
Nice one! I wonder where you were getting gnuradio 3.8 from as Homebrew currently ships 3.7.13.4.

robotastic, 2020-08-05:
Awesome! That is weird if you weren't on 3.8. I wonder if some newer versions of Gr3.7 had some changes to libraries. Either way, I will take a win!

closed 2020-08-05 by robotastic

NoahJaehnert, 2020-08-05:
In my haste to answer this, I forgot that I had removed Brew and installed MacPorts instead (as this is what fixed this problem for me last time, as noted above). 

On MacPorts, TR compiles fine. 

Reinstalled Brew and the supporting/required prereqs for TR and... same issue still exists. Sorry for the mix up here. 

TR compile output below: 
```[ 61%] Linking CXX shared library libgnuradio-op25_repeater.dylib
Undefined symbols for architecture x86_64:
  "log4cpp::Appender::AppenderMapStorageInitializer::AppenderMapStorageInitializer()", referenced from:
      __GLOBAL__sub_I_ambe_encoder_sb_impl.cc in ambe_encoder_sb_impl.cc.o
      __GLOBAL__sub_I_dmr_bs_tx_bb_impl.cc in dmr_bs_tx_bb_impl.cc.o
      __GLOBAL__sub_I_ysf_tx_sb_impl.cc in ysf_tx_sb_impl.cc.o
      __GLOBAL__sub_I_dstar_tx_sb_impl.cc in dstar_tx_sb_impl.cc.o
      __GLOBAL__sub_I_vocoder_impl.cc in vocoder_impl.cc.o
      __GLOBAL__sub_I_gardner_costas_cc_impl.cc in gardner_costas_cc_impl.cc.o
      __GLOBAL__sub_I_p25_frame_assembler_impl.cc in p25_frame_assembler_impl.cc.o
      ...
  "log4cpp::Appender::AppenderMapStorageInitializer::~AppenderMapStorageInitializer()", referenced from:
      __GLOBAL__sub_I_ambe_encoder_sb_impl.cc in ambe_encoder_sb_impl.cc.o
      __GLOBAL__sub_I_dmr_bs_tx_bb_impl.cc in dmr_bs_tx_bb_impl.cc.o
      __GLOBAL__sub_I_ysf_tx_sb_impl.cc in ysf_tx_sb_impl.cc.o
      __GLOBAL__sub_I_dstar_tx_sb_impl.cc in dstar_tx_sb_impl.cc.o
      __GLOBAL__sub_I_vocoder_impl.cc in vocoder_impl.cc.o
      __GLOBAL__sub_I_gardner_costas_cc_impl.cc in gardner_costas_cc_impl.cc.o
      __GLOBAL__sub_I_p25_frame_assembler_impl.cc in p25_frame_assembler_impl.cc.o
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/op25_repeater/lib/libgnuradio-op25_repeater.dylib] Error 1
make[1]: *** [lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/all] Error 2
make: *** [all] Error 2```