TrunkRecorder/trunk-recorder #373: Is Trunk-Recorder able to use rtl_tcp as a Source?

alteredstate82, 2020-06-19:
Hello everyone!
[This](https://github.com/robotastic/trunk-recorder/issues/182) issue is still Open but I wanted to confirm if trunk-recorder would indeed work with an rtl_tcp source?  This would be useful setting up remote RPi SDR stations and have them pipe back to a central more robust machine or VM running trunk-recorder.

EricTendian, 2020-06-19:
Yes, I've used it before without issue. The config entry in that other issue is incorrect, the device string should be something like `rtl_tcp=1.2.3.4:1234` (for something running on 1.2.3.4 at port 1234).

alteredstate82, 2020-06-26:
> Yes, I've used it before without issue. The config entry in that other issue is incorrect, the device string should be something like `rtl_tcp=1.2.3.4:1234` (for something running on 1.2.3.4 at port 1234).

Thanks for the info!  I assume the same is true with Airspy SPY server, will it also work as a source with Trunk-Recorder?

EricTendian, 2020-06-26:
@alteredstate82 unfortunately SpyServer is a bit more challenging since gr-osmosdr doesn't natively support it. However, SoapySDR does, so if you've got all the proper dependencies installed (which is a discussion in itself) then this device string should work: `soapy=0,remote=1.2.3.4,driver=remote,remote:driver=airspy,remote:prot=tcp`

I've tested both as I wanted to try and get this project working in a Docker container on Windows, with decent success.

natecarlson, 2020-12-03:
@sally-yachts some info above regarding your question on Gitter.. want to give it a try and document in the wiki?

sally-yachts, 2020-12-31:
Been kicking the tires on this and initial testing suggests rtl_tcp through osmosdr is working just fine. I've got 3 RTL-SDRs from NooElec running on an rpi4, streaming across the network to a VM on another PC. Each is defined as a separate source in config.json just like you would if they were attached locally. Tested with the latest TR docker image. I'd like to poke at it a little bit more to make sure no weird behavior emerges before closing, but initial testing looks great.

closed 2021-04-28 by robotastic