TrunkRecorder/trunk-recorder #376: Repeated Segmentation Fault with P25 Phase 2 on Ubuntu 18.04

RossIV, 2020-06-24:
**Background:**  
T-R version 3.2  
Ubuntu 18.04 on Dell Optiplex 7040 Micro w/ i5-6600T, 16 GB RAM  
Monitoring [this system](https://www.radioreference.com/apps/db/?sid=9668) which has a mix of P25 Phase 1 and Phase 2 TGs.

**Issue:**
It seems that every time a P25 Phase 2 TG comes up to be recorded, TR experiences a seg fault and crashes. Note that it does not crash if the TG is not marked to be recorded - it shows "Not recording because of priority" as expected. P25 Phase 1 records just fine with no issues.

**Supporting Info:**
Last few lines of terminal output before the crash:
```
[2020-06-24 08:45:28.007883] (info)   [sog]	TG:        352	Freq: 8.536250e+08	Ending Recorded Call - Last Update: 4s	Call Elapsed: 17
[2020-06-24 08:45:28.008011] (info)   	- Stopping P25 Recorder Num [0]	TG:        352	Freq: 8.536250e+08 	TDMA: false	Slot: 0
[2020-06-24 08:46:51.445318] (info)   	- Starting P25 Recorder Num [0]	TG:        351	Freq: 8.538500e+08 	TDMA: false	Slot: 0
[2020-06-24 08:46:51.445501] (info)   [sog]	TG:        351	Freq: 8.538500e+08	Starting Recorder on Src: airspy
[2020-06-24 08:47:00.669471] (info)   [sog]	TG:        208	Freq: 8.526250e+08	TG not in Talkgroup File
[2020-06-24 08:47:00.669618] (info)   	 P25 Recorder - Adjust Channel Resampled Rate: 24000 System Rate: 24000 ARB Rate: 1
Segmentation fault (core dumped)
```

That's all the CLI output I get, despite having it set to run in debug mode.

I see there have been seg fault issues with low signal before [(this issue)](https://github.com/robotastic/trunk-recorder/issues/88), but as best as I can tell my signal levels are fine. They're not amazing, but not horrible either. I'll run through some of the steps referenced in that issue to see if I can gather more data, but figured I'd go ahead and get this out there in case anyone else had other suggestions/comments. Any help is appreciated!

bctrainers, 2020-06-24:
Could you upload both the core dump and the compiled recorder file to onedrive, dropbox,or some file hosting service? 

The core dump will help diagnose where its faulting in the code.

RossIV, 2020-06-24:
Done, here's the dump and the compiled recorder file
https://drive.google.com/drive/folders/1PciJl3LN61Nia8Auo3em0k9JJ877ybVc?usp=sharing

loganwilliams, 2020-06-29:
I am encountering this same issue with P25 Phase 2 on OS X 10.14.

bctrainers, 2020-06-29:
> Done, here's the dump and the compiled recorder file
> https://drive.google.com/drive/folders/1PciJl3LN61Nia8Auo3em0k9JJ877ybVc?usp=sharing

My apologies on the delay. Check the core dump now against the compiled recorder file. We shall see soon what is making it go belly up. :)

bctrainers, 2020-06-29:
Hrm interesting. Appears that the segfault happens when a TDMA toggle occurs, and the recorder is possibly expecting phase 1 and not phase 2, but it's coming in with phase 2 data. ( @robotastic )

```bash
Core was generated by `./recorder --config=config-sog.json'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000055e95904cbdc in p25_recorder::switch_tdma(bool) ()
[Current thread is 1 (LWP 2916)]
```

```bash
#0  0x000055e95904cbdc in p25_recorder::switch_tdma(bool) ()
#1  0x000055e95904d9cb in p25_recorder::start(Call*) ()
#2  0x000055e958f769c7 in start_recorder(Call*, TrunkMessage, System*) ()
#3  0x000055e958f78af7 in handle_call(TrunkMessage, System*) ()
#4  0x000055e958f7a1b8 in handle_message(std::vector<TrunkMessage, std::allocator<TrunkMessage> >, System*) ()
#5  0x000055e958f7a6e2 in monitor_messages() ()
#6  0x000055e958f71988 in main ()
```

```bash
#0  0x000055e95904cbdc in p25_recorder::switch_tdma(bool) ()
No symbol table info available.
#1  0x000055e95904d9cb in p25_recorder::start(Call*) ()
No symbol table info available.
#2  0x000055e958f769c7 in start_recorder(Call*, TrunkMessage, System*) ()
No symbol table info available.
#3  0x000055e958f78af7 in handle_call(TrunkMessage, System*) ()
No symbol table info available.
#4  0x000055e958f7a1b8 in handle_message(std::vector<TrunkMessage, std::allocator<TrunkMessage> >, System*) ()
No symbol table info available.
#5  0x000055e958f7a6e2 in monitor_messages() ()
No symbol table info available.
#6  0x000055e958f71988 in main ()
No symbol table info available.
```

loganwilliams, 2020-07-01:
Is there enough information from RossIV's core dump to debug this issue? Or would it be helpful to have another sample?

bctrainers, 2020-07-05:
> Is there enough information from RossIV's core dump to debug this issue? Or would it be helpful to have another sample?

More the merrier.

For those wishing to debug it locally, `apt install gdb` then `gdb dumpfile.core recorder` when it runs, `bt` and then `bt full` - one just shows more details than the other on the backtrace. :)

robotastic, 2020-07-05:
It looks like it is crashing in this function. It could be that some of the variables in here are being set incorrectly somewhere else. There isn't anything too complex happening here though... weird!
https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/recorders/p25_recorder.cc#L340

loganwilliams, 2020-07-06:
In my case I incorrectly had `"modulation": "fsk4"` instead of `qpsk` in my `config.json`.

For me, the function was segfaulting when `costas_clock->update_omega(omega);` is called, because `costas_clock` is only defined in `initialize_qpsk`, not in `initialize_fsk4`.

Kingscup20, 2020-09-05:
I am very new to all this but I am also receiving a segfault. On a RPi 4 with SDRplay. It works with lower sample rates (2000000 and lower) but not higher rates.

Program terminated with signal SIGSEGV, Segmentation fault.
#0  0xb6fb842c in ?? ()
[Current thread is 1 (LWP 26338)]

RossIV, 2020-09-19:
Looks like @loganwilliams may have my solution. I changed to qpsk and things appear stable now. Been running for over an hour compared to just a few minutes before.

Dygear, 2020-10-18:
Not sure if this is the same issue, but I'm also getting a seg fault. I am actually using qpsk and that's not helping. First time this has happened. Once my one line change was merged, I download the entire thing all over again and checked to see if it's working well. Compiled and rebuilt my SCPD receive site that's been having issues with the new 3.3 build. I got it to capture some traffic, but now it's seg faulting very quickly. After maybe 3 calls to the [`start_recorder()`](https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/main.cc#L598) function

```
pi@MimoSDR-MSU1:~/trunk-recorder $ gdb ./recorder 
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./recorder...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/pi/trunk-recorder/recorder 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[2020-10-18 18:04:42.851614] (info)   Using Config file: ./config.json

[2020-10-18 18:04:42.852756] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------

[2020-10-18 18:04:42.852823] (info)   
-------------------------------------
SYSTEMS
-------------------------------------

[2020-10-18 18:04:42.852943] (info)   

System Number: 1
-------------------------------------

[2020-10-18 18:04:42.852996] (info)   Short Name: SCPD
[2020-10-18 18:04:42.853027] (info)   System Type: p25
[2020-10-18 18:04:42.853060] (info)   Control Channels: 
[2020-10-18 18:04:42.853100] (info)     851.162500 MHz
[2020-10-18 18:04:42.853197] (info)     852.425000 MHz
[2020-10-18 18:04:42.853248] (info)     852.675000 MHz
[2020-10-18 18:04:42.853295] (info)     852.737500 MHz
[2020-10-18 18:04:42.853349] (info)   Modulation: qpsk
[2020-10-18 18:04:42.853378] (info)   Analog Recorder Maximum Deviation: 4000
[2020-10-18 18:04:42.853409] (info)   Filter Width: 1
[2020-10-18 18:04:42.853441] (info)   Squelch: 0
[2020-10-18 18:04:42.853474] (info)   API Key: 
[2020-10-18 18:04:42.853503] (info)   Broadcastify API Key: 
[2020-10-18 18:04:42.853532] (info)   Broadcastify Calls System ID: 0
[2020-10-18 18:04:42.853562] (info)   Upload Script: 
[2020-10-18 18:04:42.853591] (info)   Call Log: true
[2020-10-18 18:04:42.853630] (info)   Audio Archive: true
[2020-10-18 18:04:42.853660] (info)   Loading Talkgroups...
[2020-10-18 18:04:42.854358] (info)   Read 69 talkgroups.
[2020-10-18 18:04:42.854406] (info)   Talkgroups File: SCPD.csv
[2020-10-18 18:04:42.854437] (info)   Loading Unit Tags...
[2020-10-18 18:04:42.854465] (info)   Unit Tags File: 
[2020-10-18 18:04:42.854493] (info)   Record Unknown Talkgroups: true
[2020-10-18 18:04:42.854523] (info)   Decode MDC: false
[2020-10-18 18:04:42.854551] (info)   Decode FSync: false
[2020-10-18 18:04:42.854579] (info)   Decode Star: false
[2020-10-18 18:04:42.854608] (info)   Decode TPS: false
[2020-10-18 18:04:42.854639] (info)   Talkgroup Display Format: tag_id
[2020-10-18 18:04:42.854673] (info)   Hide Encrypted Talkgroups: false
[... 232 lines cut ...]
[New Thread 0x6fcc6dd0 (LWP 1730)]
[2020-10-18 18:04:46.001811] (error)   [SCPD] Retuning to Control Channel: 852.425000 MHz
[2020-10-18 18:04:46.002112] (info)   	 - System Source 0 - Min Freq: 850.770250 MHz Max Freq: 853.842250 MHz
[2020-10-18 18:04:46.002453] (error)   [SCPD]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2020-10-18 18:04:49.007724] (error)   [SCPD] Retuning to Control Channel: 852.675000 MHz
[2020-10-18 18:04:49.011225] (info)   	 - System Source 0 - Min Freq: 850.770250 MHz Max Freq: 853.842250 MHz
[2020-10-18 18:04:49.011610] (error)   [SCPD]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-10-18 18:04:52.010044] (error)   [SCPD]	 Control Channel Message Decode Rate: 3.33333/sec, count:  10
[2020-10-18 18:05:01.007426] (error)   [SCPD] Retuning to Control Channel: 852.737500 MHz
[2020-10-18 18:05:01.007743] (info)   	 - System Source 0 - Min Freq: 850.770250 MHz Max Freq: 853.842250 MHz
[2020-10-18 18:05:01.008871] (error)   [SCPD]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-10-18 18:05:01.958681] (info)   [SCPD]	TG:            SCPD 4th Pct (      3411)	Freq:   0.000000 MHz	Not Recording: no source covering Freq
[2020-10-18 18:05:02.566263] (info)   [SCPD]	TG:            SUNY PD CH 3 (        91)	Freq:   0.000000 MHz	Not Recording: no source covering Freq
[2020-10-18 18:05:02.940772] (info)   [SCPD]	TG:            SCPD 1st Pct (      3111)	Freq:   0.000000 MHz	Not Recording: no source covering Freq
[2020-10-18 18:05:03.517166] (info)   [SCPD]	Decoding System ID 3CE WACN: BEE00 NAC: 3C4
[2020-10-18 18:05:05.023183] (info)   	- Starting P25 Recorder Num [0]	TG:             MEDCOM East (       151)	Freq: 851.087500 MHz 	TDMA: false	Slot: 0	Mod: true
[2020-10-18 18:05:05.023618] (info)   [SCPD]	TG:             MEDCOM East (       151)	Freq: 851.087500 MHz	Starting Recorder on Src: rtl=MSU1

Thread 1 "recorder" received signal SIGSEGV, Segmentation fault.
0x0003e59c in start_recorder(Call*, TrunkMessage, System*) ()
(gdb) bt
#0  0x0003e59c in start_recorder(Call*, TrunkMessage, System*) ()
#1  0x00000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

bctrainers, 2020-10-18:
I assume `bt full` isn't showing anything more than what the generic `bt` command has dumped in gdb?

robotastic, 2020-10-19:
@Dygear I think I just fixed the crash you had. It looks like the PR you submitted didn't check to make sure there was a Talkgroup found for that TG number, so it sometimes checked for encryption on a NULL talkgroup. I just pushed a fix, give that a try.

Dygear, 2020-10-19:
> @Dygear I think I just fixed the crash you had. It looks like the PR you submitted didn't check to make sure there was a Talkgroup found for that TG number, so it sometimes checked for encryption on a NULL talkgroup. I just pushed a fix, give that a try.

Clearly when I wrote 

> The only other place this is checked is for A as in analog, so it shouldn't break anything else. 

What I actually meant was, I was going to break EVERYTHING. Lol, sorry about that. That didn't happen at all in my NCPD system and I'm pretty sure I'm not covering all of the talk groups in the talkgroup.csv file there. It did happen in the SCPD system tho. So that's interesting.

robotastic, 2020-10-20:
No Worries!! This program has gotten so complicated that there is no longer just a simple change. I keep having to relearn this with every change I make. It was a great idea - I will go add in a note in the docs.

closed 2020-10-20 by robotastic