TrunkRecorder/trunk-recorder #379: Tags on OpenMhz

tdduarte, 2020-07-12:
Hello,

Not sure if this a future feature but when i added a csv and unitTagsFile to my system, I am able to see that it is tagging the units in the json files on my end but i'm still seeing the src on OpenMhz.  Am i missing something here or does this not work on OpenMhz? 

Example local json file
"srcList": [ {"src": 4096, "time": 1594507209, "pos": 0.456125, "emergency": 0, "signal_system": "MDC1200", "tag": "Taunton Engine 1"} ],

Thanks!

bctrainers, 2020-07-12:
Don't believe OpenMHz supports unit tagging currently.

robotastic, 2020-08-03:
This isn't supported right now... but may not be too tricky to add. Weigh in folks if this is a feature you are interested in.

natecarlson, 2020-10-31:
I'd love to see it. In the ideal world I think this would all live on the player side, ideally with an easy way for authorized users to add unit tags as they are listening to calls.. but even just adding support for OpenMHZ to ingest and display the text field that gets passed through in the json from unitTagsFile on trunk recorder would be neat.

tdduarte, 2021-12-12:
Any more updates or interest in this?

robotastic, 2024-12-14:
Going to keep this open because it would be a good feature to add

PR #991 (2025-01-05): Dev/backlog
Fixes #988 by removing Stat Socket plugin to its own repo, here: https://github.com/TrunkRecorder/trunk-recorder-stat-socket

Also fixes #379 and adds Tag to the .m4a files that get created.
closed 2025-01-05 by robotastic