TrunkRecorder/trunk-recorder #380: OpenMHz losing feeds from Montgomery County TG 14832 and 15120 each night

gpslouis, 2020-07-12:
OpenMhz is losing feeds from Montgomery County, MD TG 14832 and 15120 each night.  I think this is a problem with trunk-recorder, not OpenMhz.

At about 1 a.m. each morning, TGs 14832 and 15120 no longer play any audio on OpenMhz.  The audio comes back later in the morning, usually between 5 a.m. and 8 a.m.

To see this, click on the following URL after about 1 a.m. CDT:
https://openmhz.com/system/mocomdps?filter-type=talkgroup&filter-code=14832,15120

I live in Montgomery County and listen to those talkgroups on my standalone scanner.  They are definitely active during the night. Not real active, but at least 5 to 10 transmissions per hour.

II think this problem might caused by "patching" of talkgroups during the night that is not fully handled by trunk-recorder, but I'm not sure. If patching is taking place, my standalone scanner is handling it just fine, though.

-- gpslouis

bctrainers, 2020-07-12:
Could you paste the logged output from Trunk Recorder? Logs are generally in the `./trunk-build/logs` folder as a datestamp.log file. You can do `ls -larth` in the logs directory to list all the files, and the newest one being at the bottom.  Feel free to upload that to this issue report, or paste the segment/time of concern. 

It'd also be a good idea to do `tail -f 07-12-2020_0000_06.log` or whatever the current and newest log file name is to watch for any skipped audio, the captured audio, and so forth.

gpslouis, 2020-07-12:
I am not the person running Trunk Recorder.  The person running Trunk Recorder is using it to feed the Montgomery County Smartnet II system to OpenMhz.   

And checking now, it seems that after many months of continuously providing that feed, the Montgomery County feed is no longer available from OpenMhz as of early in the morning on 7/10/2020.

closed 2020-07-18 by gpslouis