TrunkRecorder/trunk-recorder #382: fsk4 and qpsk - mixed systems

blantonl, 2020-07-25:
The documentation currently states:

**_modulation - the type of modulation that the system uses. The options are qpsk & fsk4. It is possible to have a mix of sources using fsk4 and qpsk demodulation._**

How does trunk recorder determine which source to use in cases where mixed types might exist in the same frequency range?

I have an interesting scenario where a conventional P25 repeater that I'm monitoring cannot be decoded with trunked recorder unless the source is set to qpsk.  However, 5 other conventional P25 repeaters can be decoded if the source is set to **either** qpsk or fsk4 - so, I've set the source to qpsk.  The problem is that periodically, after a few hours, TR simply "dies" and stops recording all P25 conventional transmissions without any errors etc.  if the source is set to qpsk, it just simply stops decoding after a period of time for those conventional P25 systems.

 I'm working around this by restarting trunk recorder every few hours in a cron job, but I'm hoping to see what/where this bug is.  This TR instance is also monitoring a bunch of other systems and has a bunch of other sources as well.  

My TR config is below.  The conventional P25 entry that can only be decoded using qpsk is 155.9325.

```
{
	"sources": [{
			"center": 408000000.0,
			"rate": 6000000,
                        "lnaGain": 15,
			"ifGain": 15,
                        "mixGain": 15,
			"digitalLevels": 3,
			"digitalRecorders": 10,
			"driver": "osmosdr",
			"device": "airspy",
			"modulation": "fsk4"
		},
		{
                        "center": 852200000.0,
                        "rate": 2048000,
                        "ppm": 42.0,
                        "gain": 38.0,
                        "digitalLevels": 3,
                        "digitalRecorders": 7,
                        "driver": "osmosdr",
                        "device": "rtl=1001,buflen=65536",
                        "modulation": "fsk4"
                },
                {
                        "center": 853000000.0,
                        "rate": 2048000,
                        "ppm": 74.0,
                        "gain": 38.0,
                        "digitalLevels": 3,
                        "digitalRecorders": 7,
                        "driver": "osmosdr",
                        "device": "rtl=1002,buflen=65536",
                        "modulation": "fsk4"
                },
		{
                        "center": 155300000.0,
                        "rate": 2048000,
                        "ppm": 68.0,
                        "squelch": -45,
                        "gain": 49.6,
                        "digitalLevels": 3,
                        "driver": "osmosdr",
                        "device": "rtl=1003,buflen=65536",
                        "modulation": "qpsk"
                }
	],
	"systems": [
		{
			"control_channels": [406362500],
			"type": "p25",
			"shortName": "lafb",
			"broadcastifyApiKey": "4c3e6b1c-xx",
			"broadcastifySystemId": "4",
			"audioArchive": false
		},
		{
			"control_channels": [407812500],
			"type": "p25",
			"shortName": "rafb",
			"broadcastifyApiKey": "4c3e6b1c-xx",
			"broadcastifySystemId": "5",
			"audioArchive": false
		},
		{
			"control_channels": [406112500],
			"type": "p25",
			"shortName": "cb",
			"broadcastifyApiKey": "4c3e6b1c-xx",
			"broadcastifySystemId": "6",
			"audioArchive": false
		},
		{
                        "control_channels": [851325000,851825000,853562500,853787500],
                        "type": "p25",
                        "shortName": "lcra800-1",
			"broadcastifyApiKey": "4c3e6b1c-324f-11ea-bd8b-0ecc8ab9ccec",
			"broadcastifySystemId": "3",
                        "audioArchive": false
                },
		{
                        "type": "conventionalP25",
                        "channels": [156000000, 155595000, 155122500, 154807500, 155932500],
                        "shortName": "bexar-surround-conv",
                        "delayCreateOutput": true,
			"broadcastifyApiKey": "4c3e6b1c-xx",
			"broadcastifySystemId": "8",
                        "audioArchive": false
                },
                {
                        "type": "conventional",
                        "channels": [155610000],
                        "shortName": "bexar-surround-aconv",
                        "delayCreateOutput": true,
			"broadcastifyApiKey": "4c3e6b1c-xx",
			"broadcastifySystemId": "20",
                        "audioArchive": false,
                        "minDuration": 0.75
                }
	],
	"broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
	"controlWarnRate": 5,
	"callTimeout": 0.75,
  "logFile": true
}
```

robotastic, 2020-07-25:
Interesting - I wonder if it is the same bug as listed here:
https://github.com/robotastic/trunk-recorder/issues/318

robotastic, 2020-09-19:
Making good progress on this - this is being tracked under #394 now and adds in other System level tuning.

closed 2020-09-19 by robotastic