TrunkRecorder/trunk-recorder #385: OPENMHZ - Filter on radio ID

dbuttry, 2020-08-01:
I would like to be able to filter on radio ID to follow my family member who is a LEO.  Any chance of being able to incorporate that?  Thanks!!!

robotastic, 2020-08-03:
Unfortunately it is probably something I can't do. Running that search on the DB would probably grind it to a halt if it was done a lot... I will can go check it out when I have a chance and see if I can come up without a good search/index strategy. It may not end up being as bad as I think.

blantonl, 2020-08-16:
I'm not sure if OpenMHZ is storing and indexing all radio IDs that are part of **conversations**, but certainly the conversation architecture that trunk-recorder has implemented makes is much harder to tackle this sort of individual radio ID indexing.  

This would reference the existing feature request I've opened to have an option to not record calls as conversations but instead as individual calls.  

https://github.com/robotastic/trunk-recorder/issues/326

"Conversation" handling is a rather arbitrary approach and results in significant difficulties in implementing data normalization schemes such as the OP request.

rosecitytransit, 2020-12-25:
Would it be possible to at least have a column of radio IDs from the call like I do on my sites?
http://www.rosecitytransit.org/radio/
http://kcmscanner.ddns.net/

Besides being able to follow certain employees, it helps in finding different parts of an incident once you know who's responding in order to get the whole picture. Now you basically have to listen to every conversation on that talk group. Broadcastify Calls does have a column but only ever seems to show one of the radios used.

natecarlson, 2020-12-25:
That's an easy change to the source code; I've done an ugly implementation of that on a fork of OpenMHz. Not sure if @robotastic would be interested in that upstream?

robotastic, 2024-12-14:
This may require a bit of work to enable with the new call processing method... if this is something you are really interested in, give it a thumbs up

robotastic, 2025-05-13:
Tracking in #813  - I will close this thread and move everything there

closed 2025-05-13 by robotastic