TrunkRecorder/trunk-recorder #391: Build fails on macOS 10.15.5

e850205, 2020-08-06:
Undefined symbols for architecture x86_64:
  "boost::re_detail_107100::raw_storage::insert(unsigned long, unsigned long)", referenced from:
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_repeat(unsigned long, unsigned long) in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_alt() in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_perl_extension() in call_uploader.cc.o
  "boost::re_detail_107100::raw_storage::resize(unsigned long)", referenced from:
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse(char const*, char const*, unsigned int) in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_extended() in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_basic() in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_creator<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::finalize(char const*, char const*) in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_open_paren() in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_extended_escape() in call_uploader.cc.o
      boost::re_detail_107100::basic_regex_parser<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::parse_repeat(unsigned long, unsigned long) in call_uploader.cc.o
      ...
  "boost::re_detail_107100::get_mem_block()", referenced from:
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_imp() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_all_states() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_startmark() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_endmark() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_alt() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_rep() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_char_repeat() in call_uploader.cc.o
      ...
  "boost::re_detail_107100::put_mem_block(void*)", referenced from:
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_imp() in call_uploader.cc.o
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::unwind_extra_block(bool) in call_uploader.cc.o
  "boost::re_detail_107100::verify_options(unsigned int, boost::regex_constants::_match_flags)", referenced from:
      boost::re_detail_107100::perl_matcher<char const*, std::__1::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::match_imp() in call_uploader.cc.o
  "boost::re_detail_107100::raise_runtime_error(std::runtime_error const&)", referenced from:
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::init() in call_uploader.cc.o
      void boost::re_detail_107100::raise_error<boost::regex_traits_wrapper<boost::regex_traits<char, boost::cpp_regex_traits<char> > > >(boost::regex_traits_wrapper<boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::error_type) in call_uploader.cc.o
  "boost::re_detail_107100::get_default_error_string(boost::regex_constants::error_type)", referenced from:
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::init() in call_uploader.cc.o
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::error_string(boost::regex_constants::error_type) const in call_uploader.cc.o
  "boost::re_detail_107100::cpp_regex_traits_char_layer<char>::init()", referenced from:
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::cpp_regex_traits_implementation(boost::re_detail_107100::cpp_regex_traits_base<char> const&) in call_uploader.cc.o
  "boost::re_detail_107100::lookup_default_collate_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      boost::re_detail_107100::cpp_regex_traits_implementation<char>::lookup_collatename(char const*, char const*) const in call_uploader.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [recorder] Error 1
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
make: *** [all] Error 2

robotastic, 2020-08-07:
Can you provide a little more information?
Have you installed it successfully before on this machine?
Are you using Brew or MacPorts? ( I have always done the MacPorts approach, if you can give that a try do that)
Make sure all your packages are up to date. This looks like some Boost weirdness.

robotastic, 2020-08-16:
Closing until more information is provided.

closed 2020-08-16 by robotastic