TrunkRecorder/trunk-recorder #393: Control channel intermittently fails to decode

pilotalex, 2020-08-11:
I'm having an issue where, intermittently, my control channel(s) will fail to decode, and the only fix seems to be control-c and restart ./recorder, hoping for success. Once it works, it works, until the next time I have to restart or shutdown the computer.

Ubuntu 18.04.4 LTS on a VMWare VM, running 3 RTL-SDR's. My PPM was calculated by using GQRX (on another VM) to tune to the control channel and adjusted to center the waterfall. It varies by 1-2 PPM each day, I'm assuming due to temperature drift. 

My only thought is that its some sort of frequency instability of the RTL-SDR, but that doesn't explain why it will sometimes succeed and others fail, and why once it starts, it runs indefinately. 

My config.json

{
        "sources": [{
                        "center": "854462500",
                        "rate": "2400000",
                        "ppm": "40",
                        "gain": "20",
                        "digitalRecorders": 2,
                        "digitalLevels": 8,
                        "driver": "osmosdr",
                        "device": "rtl=101010",
                        "modulation": "qpsk"
        },{
                        "center": "857950000",
                        "rate": "2400000",
                        "ppm": "35.5",
                        "gain": "20",
                        "digitalRecorders": 4,
                        "digitalLevels": 8,
                        "driver": "osmosdr",

Sample output of a failure

trunk-recorder@TRVM:~/trunk-build$ ./recorder
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[2020-08-11 11:45:55.881273] (info)   Using Config file: ./config.json

[2020-08-11 11:45:55.882225] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------

[2020-08-11 11:45:55.882734] (info)   
-------------------------------------
SYSTEMS
-------------------------------------

[2020-08-11 11:45:55.883194] (info)   

System Number: 1
-------------------------------------

[2020-08-11 11:45:55.883836] (info)   Short Name: bcem
[2020-08-11 11:45:55.884065] (info)   System Type: p25
[2020-08-11 11:45:55.884287] (info)   Control Channels: 
[2020-08-11 11:45:55.884486] (info)     859.962500 MHz
[2020-08-11 11:45:55.884708] (info)     859.937500 MHz
[2020-08-11 11:45:55.884920] (info)     858.962500 MHz
[2020-08-11 11:45:55.885120] (info)     858.937500 MHz
[2020-08-11 11:45:55.885311] (info)   API Key: 3cd9ede94d85ff497bf5f30f6e4020964b7c39092730ade731b6fb9dbc5887f4
[2020-08-11 11:45:55.885446] (info)   Broadcastify API Key: 09b3449a-d4ce-11ea-bd8b-0ecc8ab9ccec
[2020-08-11 11:45:55.885584] (info)   Broadcastify Calls System ID: 257
[2020-08-11 11:45:55.885714] (info)   Upload Script: 
[2020-08-11 11:45:55.885849] (info)   Call Log: true
[2020-08-11 11:45:55.885978] (info)   Audio Archive: true
[2020-08-11 11:45:55.886106] (info)   Loading Talkgroups...
[2020-08-11 11:45:55.886398] (error)   Malformed talkgroup entry at line 4.
[2020-08-11 11:45:55.886637] (error)   Malformed talkgroup entry at line 9.
[2020-08-11 11:45:55.887160] (error)   Malformed talkgroup entry at line 57.
[2020-08-11 11:45:55.887369] (error)   Malformed talkgroup entry at line 61.
[2020-08-11 11:45:55.887565] (error)   Warning: skipped 4 of 63 talkgroup entries! Invalid format?
[2020-08-11 11:45:55.887713] (error)   The format is very particular. See https://github.com/robotastic/trunk-recorder for example input.
[... 432 lines cut ...]
[2020-08-11 06:24:49.715334] (info)   Broadcastify Calls Server: https://api.broadcastify.com/call-upload
[2020-08-11 06:24:49.715355] (info)   Status Server: 
[2020-08-11 06:24:49.715374] (info)   Instance Key: 
[2020-08-11 06:24:49.715392] (info)   Instance Id: 
[2020-08-11 06:24:49.715412] (info)   Broadcast Signals: false
[2020-08-11 06:24:49.715433] (info)   Default Mode: digital
[2020-08-11 06:24:49.715460] (info)   Call Timeout (seconds): 3
[2020-08-11 06:24:49.715500] (info)   Log to File: true
[2020-08-11 06:24:49.715523] (info)   Control channel warning rate: 10
[2020-08-11 06:24:49.715547] (info)   Control channel retune limit: 0
[2020-08-11 06:24:49.715582] (info)   Frequency format: 1
[2020-08-11 06:24:49.715620] (info)   Status as String: true
[2020-08-11 06:24:49.715721] (info)   Log Level: info
[2020-08-11 06:24:49.715991] (info)   [bcem] Started with Control Channel: 859.962500 MHz
Decim: 25 Decim2:  4
[2020-08-11 06:24:49.716921] (info)   P25 Trunking two-stage decimator - Initial decimated rate: 96000 Second decimated rate: 24000 FA: 6250 FB: 12000 System Rate: 2400000
[2020-08-11 06:24:49.769897] (info)   P25 Trunking ARB - Initial Rate: 2400000 Resampled Rate: 24000 Initial Decimation: 25 System Rate: 24000 ARB Rate: 1
[2020-08-11 06:24:50.285525] (info)   [bcem] TG:        301 Freq:   0.000000 MHz Not Recording: no source covering Freq
[2020-08-11 06:24:50.605405] (info)   [bcem] Decoding System ID 34A WACN: BEE00 NAC: 340
[2020-08-11 06:25:46.916795] (info)   - Starting P25 Recorder Num [2] TG:        317 Freq: 856.962500 MHz TDMA: false Slot: 0
[2020-08-11 06:25:46.918072] (info)   [bcem] TG:        317 Freq: 856.962500 MHz Starting Recorder on Src: rtl=020202
[2020-08-11 06:25:57.009156] (info)   [bcem] TG:        317 Freq: 856.962500 MHz Ending Recorded Call - Last Update: 4s Call Elapsed: 11
[2020-08-11 06:25:57.010202] (info)   - Stopping P25 Recorder Num [2] TG:        317 Freq: 856.962500 MHz TDMA: false Slot: 0
[2020-08-11 06:25:58.782876] (info)   [bcem] TG: 317 Freq: 856.962500 MHz Broadcastify Upload Success - file size: 21032
[2020-08-11 06:25:59.462416] (info)   [bcem] TG: 317 Freq: 856.962500 MHz OpenMHz Upload Success - file size: 21032

e850205, 2020-08-14:
Same issue here. I just keep restarting the trunk recorder service until it works. Kind of annoying. I have no explanation for it.

robotastic, 2020-08-16:
This is generally a sign that the SDR is not tuned properly and reception is right on the margin. You might want to try a different sample rate, using error instead of ppm and adjusting gain.

Feel free to re-open if you can't get it fixed.

closed 2020-08-16 by robotastic