TrunkRecorder/trunk-recorder #394: Configure Recorders by System instead of source

robotastic, 2020-08-16:
Right now, Recorders are configured at start. This makes the overall design a lot simpler. The settings for recorders are determine the Source they are connected to. Recorders can be shared by any of the systems using that source for recording. 

Systems may have different configurations though, and it would be good to have the Recorder setting set by the system currently using it. To do this, Recorders either need to be:

- Permanently attached to the graph and dynamically configurable
or 
- Each time a recorder is need it is dynamically created and attached to the right Source and then destroyed when it is done. 

Trying the first option first. The second one would mean lots of starting and stopping of the graph.

robotastic, 2020-09-16:
Ok - I think this is doable. I have a branch: https://github.com/robotastic/trunk-recorder/tree/duper-recorder where I have the basics working. From a technical perspective, this should be doable. I got to do some integration work to do still so it can be configured by system. I will make it possible to configure squelch and modulation by system instead of source. heck, would could probably even do audio levels if folks are into that.

daryljones, 2020-09-16:
It would be very nice to have audio level adjustments by system.  Any
chance of adding CTCSS decode and gating?

robotastic, 2020-09-16:
I can definitely get audio levels added in! 

CTCSS decoding would probably be a different round of code surgery. It would be a little tricky for me to work on these because I don't have any local systems using it to work against. 

I just googled what gating is... is this the right thing? https://en.wikipedia.org/wiki/Noise_gate
That might be possible and not a huge lift. I would have to see if it is doable in GnuRadio. I would also have to find some local analog systems to test againt, but that might be easier.

daryljones, 2020-09-16:
The Wikipedia reference to "Noise Gate" describes what I was thinking about.

Thanks!

robotastic, 2020-09-19:
Ok - I think things should be working on the https://github.com/robotastic/trunk-recorder/tree/duper-recorder branch. Could someone give it a try? I don't have a local system with overlapping coverage. Instead of defining the modulation, squelch and levels at the source in the config file, define them under the system.

closed 2021-04-28 by robotastic