TrunkRecorder/trunk-recorder #396: Error - can't set XOR Mask for TDMA / parity data ... R-S symbol size

kcwebby, 2020-08-17:
Getting these on P2 Systems quite regularly, they might be two separate issues, but they might be linked.

```
[2020-08-16 20:26:18.298991] (info)   Error - can't set XOR Mask for TDMA
[2020-08-16 20:26:18.299027] (info)     - Starting P25 Recorder Num [9] TG:      30000  Freq: 7.720812e+08      TDMA: true      Slot: 0
[2020-08-16 20:26:18.299157] (info)   [sys_1]      TG:      30000  Freq: 7.720812e+08      Starting Recorder on Src: rtl=1,buflen=65536
thread[thread-per-block[509]: <block p25_frame_assembler (172)>]: reed-solomon: parity data contains information beyond R-S symbol size
thread[thread-per-block[493]: <block p25_frame_assembler (191)>]: reed-solomon: parity data contains information beyond R-S symbol size
```

robotastic, 2020-09-19:
I will try to pull in the latest OP25 code, perhaps there is an underlying bug there.

robotastic, 2021-04-28:
Let me know if this is still an issue - I can try to add additional debugging messages. I am not sure why the XOR Mask wouldn't be getting set.

closed 2021-04-28 by robotastic

MaxwellDPS, 2021-05-08:
Just saw this 

```
Decim: 16 Decim2:  4
[2021-05-08 10:45:42.253396] (info)      P25 Trunking two-stage decimator - Initial decimated rate: 128000 Second decimated rate: 32000 FA: 6250 FB: 16000 System Rate: 2048000
[2021-05-08 10:45:42.268079] (info)      P25 Trunking ARB - Initial Rate: 2048000 Resampled Rate: 32000 Initial Decimation: 16 System Rate: 24000 ARB Rate: 0.75
Allocating 15 zero-copy buffers
Allocating 15 zero-copy buffers
Allocating 15 zero-copy buffers
[2021-05-08 10:45:42.704295] (info)      P25 Recorder - Adjust Channel Resampled Rate: 32000 System Rate: 24000 ARB Rate: 0.75
[2021-05-08 10:45:42.704622] (info)   Error - can't set XOR Mask for TDMA
[2021-05-08 10:45:42.704647] (info)     - Starting P25 Recorder Num [4] TG:      12521  Freq: 8.577125e+08      TDMA: true                                                                                         Slot: 1  Mod: true
[2021-05-08 10:45:42.704705] (info)   [nesr]    TG:      12521  Freq: 8.577125e+08      Starting Recorder on Src: rtl=2
[2021-05-08 10:45:42.929243] (info)      P25 Recorder - Adjust Channel Resampled Rate: 32000 System Rate: 24000 ARB Rate: 0.75
[2021-05-08 10:45:42.929555] (info)   Error - can't set XOR Mask for TDMA
[2021-05-08 10:45:42.929575] (info)     - Starting P25 Recorder Num [8] TG:      12159  Freq: 8.594625e+08      TDMA: true                                                                                         Slot: 1  Mod: true
[2021-05-08 10:45:42.929673] (info)   [nesr]    TG:      12159  Freq: 8.594625e+08      Starting Recorder on Src: rtl=1
[2021-05-08 10:45:43.294929] (info)      P25 Recorder - Adjust Channel Resampled Rate: 32000 System Rate: 24000 ARB Rate: 0.75
[2021-05-08 10:45:43.295266] (info)   Error - can't set XOR Mask for TDMA
[2021-05-08 10:45:43.295289] (info)     - Starting P25 Recorder Num [9] TG:      12116  Freq: 8.594625e+08      TDMA: true                                                                                         Slot: 0  Mod: true
[2021-05-08 10:45:43.295339] (info)   [nesr]    TG:      12116  Freq: 8.594625e+08      Starting Recorder on Src: rtl=1
[2021-05-08 10:45:44.293213] (info)   [nesr]    Decoding System ID 144 WACN: BEE00 NAC: 143
[2021-05-08 10:45:46.006398] (info)   [nesr]    TG:      12159  Freq: 8.594625e+08      Ending Recorded Call - Last Update: 4s                                                                                     Call Elapsed: 4
[2021-05-08 10:45:46.006810] (info)     - Stopping P25 Recorder Num [8] TG:      12159  Freq: 8.594625e+08      TDMA: true                                                                                         Slot: 1
[2021-05-08 10:45:46.007102] (error)   wav error closing file
```