TrunkRecorder/trunk-recorder #397: TR Ignoring CSV Priority > 100

kcwebby, 2020-08-17:
I have several machines on the same system, where I'm trying to eliminate the duplicates, and I seem to be unable to get trunk-recorder to pick up (or ignore) the priority talkgroups.

I am seeing the TG File loading during TR startup.   Talkgroups are not ignored.

I've read/seen/chatted about something somewhere, that the priority calc is actually using the HEX field in the CSV ?

This is becoming bothersome for me, and costs storage/resources for me on a somewhat large scale.   I'd like to get this looked at, fixed ASAP.

Here's an example of my config

``
        "systems": [{
                "control_channels": [774456250,773693750],
                "type": "p25",
                "uploadScript": "016-houtx-03.txwarn.encode-upload.sh",
                "recordUnknown": false,
                "talkgroupsFile": "site0113.csv",
                "shortName": "sys_1.13"
          }],
```

And an example of my CSV file:

```
head trunk-build/site0113.csv
3,,D,PublicHealth,Ch 781 Public Health HFD Ch DB1,Interop,Interoperability  Zone 78,1
5,,D,CMOC 1,Ch 782 Houston Catastrophic Medical Operations Center CMOC 1 HFD Ch DB2,Interop,Interoperability  Zone 78,1
9,,D,HCCO1 Announ,Precinct 1  Announcement,Law Tac,Harris County Constables,2
10,,D,SctWhtColSta,Scott and White Hospital College Station,Hospital,Brazos County,10
11,,D,CMOC 2,Ch 783 Houston Catastrophic Medical Operations Center CMOC 2 HFD Ch DB3,Interop,Interoperability  Zone 78,1
13,,D,LifeFlt Ctrl,Hermann Life Flight Control,EMS-Tac,Air Medical,2
15,,D,CMOC 3,Ch 784 Houston Catastrophic Medical Operations Center CMOC 3 HFD Ch DB4,Interop,Interoperability  Zone 78,1
17,,D,CMOC 4,Ch 785 Houston Catastrophic Medical Operations Center CMOC 4 HFD Ch DB5,Interop,Interoperability  Zone 78,1
19,,D,CMOC 5,Ch 786 Houston Catastrophic Medical Operations Center CMOC 5 HFD Ch DB6,Interop,Interoperability  Zone 78,1
21,,D,CMOC 6,Ch 787 Houston Catastrophic Medical Operations Center CMOC 6 HFD Ch DB7,Interop,Interoperability  Zone 78,1
```

e850205, 2020-08-18:
Why would you leave the hex fields blank when it’s a known requirement?

kcwebby, 2020-08-20:
Well Sir -- Considering the code doesn't use it, why is it a requirement?

e850205, 2020-08-29:
I’m telling you the solution to the problem.

Dygear, 2020-10-19:
`10,,D,SctWhtColSta,Scott and White Hospital College Station,Hospital,Brazos County,10`

This line has the priority of 10. If you increase that to 100 it should ignore it.

closed 2021-04-28 by robotastic