TrunkRecorder/trunk-recorder #399: Failed to allocate zero-copy buffer

boydj, 2020-09-18:
After some period of time (10 hours or so), trunk-recorder starts failing with the following error messages and stops decoding:

```
trunk-recorder | Allocating 15 zero-copy buffers
trunk-recorder | Failed to allocate zero-copy buffer for transfer 0
trunk-recorder | Falling back to buffers in userspace
trunk-recorder | Failed to submit transfer 0
trunk-recorder | Please increase your allowed usbfs buffer size with the following command:
trunk-recorder | echo 0 > /sys/module/usbcore/parameters/usbfs_memory_mb
trunk-recorder | rtlsdr_read_async returned with -5

```

usbfs_memory_mb is already set to 0, it seems like there is plenty of free memory on the server still:

```
$ free -m
              total        used        free      shared  buff/cache   available
Mem:           7887        1106         212          96        6567        6405
Swap:          8095          45        8050
```

I am running robotastic/trunk-recorder:edge on Docker 19.03.12.

boydj, 2020-09-18:
config.json
```
{
	"sources": [{
		"center": "852731250",
		"rate": "2500000",
		"error": "0",
		"ppm": "0",
		"lnaGain": "47",
		"digitalRecorders": 7,
		"digitalLevels": 8,
		"driver": "osmosdr",
		"device": "rtl=10000001",
		"modulation": "qpsk"
	},{
		"center": "152302500",
		"rate": "2500000",
		"error": "0",
		"ppm": "0",
		"lnaGain": "47",
		"digitalRecorders": 1,
		"digitalLevels": 8,
		"driver": "osmosdr",
		"device": "rtl=10000002",
		"modulation": "fsk4"
	},{
		"center": "159112500",
		"rate": "2500000",
		"error": "0",
		"ppm": "0",
		"lnaGain": "47",
		"digitalRecorders": 2,
		"digitalLevels": 8,
		"driver": "osmosdr",
		"device": "rtl=10000003",
		"modulation": "fsk4"
	},{
		"center": "851806250",
		"rate": "2500000",
		"error": "-2400",
		"ppm": "0",
		"lnaGain": "47",
		"digitalRecorders": 3,
		"digitalLevels": 8,
		"driver": "osmosdr",
		"device": "rtl=10000005",
		"modulation": "qpsk"
	},{
		"center": "854118750",
		"rate": "2500000",
		"error": "-100",
		"lnaGain": "47",
		"digitalRecorders": 2,
		"digitalLevels": 8,
		"driver": "osmosdr",
		"device": "rtl=10000004",
		"modulation": "qpsk"
	}],
	"systems": [{
		"control_channels": [852612500,853912500,853050000],
		"type": "p25",
		"talkgroupsFile": "/app/trs_tg_6524.csv",
		"recordUnknown": true,
		"shortName": "loudoun",
		"broadcastifyApiKey": "XXX",
		"broadcastifySystemId": "208",
		"apiKey": "XXX",
		"callLog": true,
		"hideEncrypted": false,
		"hideUnknownTalkgroups": false
	},
 	{
		"control_channels": [152202500],
		"type": "p25",
		"talkgroupsFile": "/app/trs_tg_3783.csv",
		"recordUnknown": true,
		"shortName": "starsloco",
		"broadcastifyApiKey": "XXXX",
		"broadcastifySystemId": "291",
		"apiKey": "XXXXX",
		"callLog": true,
		"hideEncrypted": false,
		"hideUnknownTalkgroups": false
	},
	{
		"control_channels": [854437500,854362500],
		"type": "p25",
		"talkgroupsFile": "/app/trs_tg_10697.csv",
		"broadcastifyApiKey": "XXX",
		"broadcastifySystemId": "290",
		"apiKey": "XXXXX",
		"recordUnknown": true,
		"shortName": "mwaa",
		"callLog": true,
		"hideEncrypted": false,
		"hideUnknownTalkgroups": false
	}],
	"uploadServer": "https://api.openmhz.com",
	"broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
	"defaultMode": "digital",
	"captureDir": "/app/media",
	"callTimeout": "3",
	"logFile": "1",
	"frequencyFormat": "mhz"
}
```

robotastic, 2020-09-18:
Interesting - I have not seen this one before. It sounds like it is lower level than something inside trunk recorder, but I also could have built the Docker container wrong. I wonder if usbfs_memory_mb needs to be changed inside Docker and/or at the base OS.

Also give this trick a try:  It seems that when you have 5 or more RTLSDRs on one system you need to decrease the buffer size. I think it has something to do with the driver. Try adding buflen: "device": "rtl=serial_num,buflen=65536", there should be no space between buflen and the comma

boydj, 2020-09-18:
Checked inside the container, usbfs_memory_mb also returns as 0.

I will try to decrease the buffer size and report back.

boydj, 2020-09-28:
That seems to have fixed this problem, thank you.

boydj, 2020-10-05:
Setting buflen ends up have not fixed the issue, it has recurred.

After about 5 days of runtime I started receiving the same error again.

```
trunk-recorder | Allocating 15 zero-copy buffers
trunk-recorder | Failed to allocate zero-copy buffer for transfer 0
trunk-recorder | Falling back to buffers in userspace
trunk-recorder | Failed to submit transfer 0
trunk-recorder | Please increase your allowed usbfs buffer size with the following command:
trunk-recorder | echo 0 > /sys/module/usbcore/parameters/usbfs_memory_mb
trunk-recorder | rtlsdr_read_async returned with -5
```

jknepher, 2020-10-29:
I had to keep lowering buflen, and buflen=32768 made mine stable.  I have 6 sources on 2 separate usb hcd, and did not have to go down to 32768 until I added the 5th.  Looks like you also have 5 so this should help; at 65536, it would sometimes work, and sometimes fail just as you are seeing.

robotastic, 2021-04-28:
Closing this - reopen if it is still an issue.

closed 2021-04-28 by robotastic