TrunkRecorder/trunk-recorder #401: Mac OS X Compile Problem - log4cpp::Appender::AppenderMapStorageInitializer

blantonl, 2020-09-21:
I'm getting the following error compiling the latest version of TR

Dependencies were installed using Brew

```
[ 61%] Linking CXX shared library libgnuradio-op25_repeater.dylib
Undefined symbols for architecture x86_64:
  "log4cpp::Appender::AppenderMapStorageInitializer::AppenderMapStorageInitializer()", referenced from:
      __GLOBAL__sub_I_ambe_encoder_sb_impl.cc in ambe_encoder_sb_impl.cc.o
      __GLOBAL__sub_I_dmr_bs_tx_bb_impl.cc in dmr_bs_tx_bb_impl.cc.o
      __GLOBAL__sub_I_ysf_tx_sb_impl.cc in ysf_tx_sb_impl.cc.o
      __GLOBAL__sub_I_dstar_tx_sb_impl.cc in dstar_tx_sb_impl.cc.o
      __GLOBAL__sub_I_vocoder_impl.cc in vocoder_impl.cc.o
      __GLOBAL__sub_I_gardner_costas_cc_impl.cc in gardner_costas_cc_impl.cc.o
      __GLOBAL__sub_I_p25_frame_assembler_impl.cc in p25_frame_assembler_impl.cc.o
      ...
  "log4cpp::Appender::AppenderMapStorageInitializer::~AppenderMapStorageInitializer()", referenced from:
      __GLOBAL__sub_I_ambe_encoder_sb_impl.cc in ambe_encoder_sb_impl.cc.o
      __GLOBAL__sub_I_dmr_bs_tx_bb_impl.cc in dmr_bs_tx_bb_impl.cc.o
      __GLOBAL__sub_I_ysf_tx_sb_impl.cc in ysf_tx_sb_impl.cc.o
      __GLOBAL__sub_I_dstar_tx_sb_impl.cc in dstar_tx_sb_impl.cc.o
      __GLOBAL__sub_I_vocoder_impl.cc in vocoder_impl.cc.o
      __GLOBAL__sub_I_gardner_costas_cc_impl.cc in gardner_costas_cc_impl.cc.o
      __GLOBAL__sub_I_p25_frame_assembler_impl.cc in p25_frame_assembler_impl.cc.o
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/op25_repeater/lib/libgnuradio-op25_repeater.dylib] Error 1
make[1]: *** [lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/all] Error 2
make: *** [all] Error 2
```

robotastic, 2020-09-22:
Can you try rolling back to an earlier release to see if there is still an error? 
https://github.com/robotastic/trunk-recorder/releases

What version of GnuRadio are you using?

I don't test against brew - I am running MacPorts on my MacBook. It is less cool, but reliable.

blantonl, 2020-09-22:
It is gnuradio 3.8

I also installed using Mac ports and got the same problem

robotastic, 2020-09-22:
Hmm... Can you try installing Log4Cpp? It looks like it is having trouble finding this lib: https://formulae.brew.sh/formula/log4cpp
https://ports.macports.org/port/log4cpp/summary

Check /opt/local/lib/ for liblog4cpp.dylib

blantonl, 2020-09-22:
It is there and installed.  

```
THOR:lib lindsay$ ls -l | grep log4
-rwxr-xr-x    1 root      admin     258500 Sep 29  2017 liblog4cpp.5.dylib
-rw-r--r--    1 root      admin     682544 Sep 29  2017 liblog4cpp.a
lrwxr-xr-x    1 root      admin         18 Sep 29  2017 liblog4cpp.dylib -> liblog4cpp.5.dylib
THOR:lib lindsay$ pwd
/opt/local/lib
```

blantonl, 2020-09-26:
Note that I completely reinstalled the machine from scratch with Mac OS X 10.13.6 (High Sierra) and installed all prereqs using port and trunk recorder complied without any problems.

We can close this issue out.  Thanks for your patience!

robotastic, 2020-09-27:
Yea, sometimes libraries on the Mac just get messed up. Thanks for circling back with an update.

closed 2020-09-27 by robotastic