TrunkRecorder/trunk-recorder #403: Multiple Airspy Support Confirmation

blantonl, 2020-09-26:
Can anyone confirm that multiple Airspy support has been patched into the latest version of Trunk Recorder?

YagiBear, 2020-10-01:
I am looking for the same thing. I used docker to load my install and I am not sure how I could rebuild it with the older gnu radio library required for multiple airspy devices

gonzoblue, 2021-07-16:
would also like to know as well before I buy a second one...

leee, 2021-07-17:
trunk-recorder doesn't include gr-osmosdr in our own tree, and users install it from their usual package archive sections (e.g. debian main or ubuntu universe). Packages built by volunteer maintainers/developers for those "official/community" archives tend to stay as close as they can to upstream projects, with patches for platform build/install compatibility and not functional changes.

Likewise, trunk-recorder is downstream of several libraries and projects, including gr-osmosdr and op25. op25 is included in tree because a lot of changes were made to interface with trunk-recorder, that would never make it back upstream. gr-osmosdr hasn't been touched, and (relatively) small changes like this one likely aren't worth adding and forking off our own tree for.

There was some work and a patch suggested for multiple airspy support [in this upstream bug](https://osmocom.org/issues/5144).

If you're comfortable patching and building your own source, you can use the patchfile from the mailing list email linked in that bug (and hope it still applies cleanly and works for gr3.8 - maybe someone can confirm, or manually make the changes yourself). Or you could ask the maintainers of the package across the various platform archives to apply the patch and make available downstream, not as likely.

The easiest request would be to ask for the change to be taken in at the project.

When you use a patched gr-osmosdr (regardless of how it's made available, either by yourself, from a downstream package archive, or from the project itself upstream), trunk-recorder is agnostic to gr-osmosdr settings, and so configuring additional radios as specified by the patch will work out of the box.

If you wish to see this feature made available, asking for the feature in that upstream bug is the best course of action, and I think this issue can be closed. (@robotastic)

If someone tests and confirms that the patch works, then a temporary install note about osmosdr multiple airspy devices with instructions could be added to the wiki until the upstream project (hopefully, eventually) accepts that patch or similar.

blantonl, 2021-07-17:
I've successfully patched gr-osmosdr with this patch and then compiled trunk recorder

https://lists.osmocom.org/pipermail/osmocom-sdr/2016-April/001446.html

Works great with trunk recorder and is part of the Broadcastify Raspberry Pi Image which includes this compiled version.

https://www.broadcastify.com/rpi/

leee, 2021-07-17:
That's fantastic, and definitely a perk of being able to image and control a host from bottom to top!

Would you be willing to write up something for a section about this for inclusion in https://github.com/robotastic/trunk-recorder/wiki/Installation?

Also, letting the osmocom project know on the bug or on the mailing list that the patch works, and that you're using it in production would be fantastic for convincing them to take in the change.

blantonl, 2021-07-17:
I do need to write it up in the wiki.

Here are more details on what I did.  

https://forums.radioreference.com/threads/multiple-airspys-trunk-recorder.400218/

I'll put a task in my list of things to do to get this over to the wiki here for TR

leee, 2021-07-17:
Thank you so much! @robotastic I think this issue can be closed, but I look forward to the notes, and hopefully making it known that there are lots of hosts running muliple airspy radios will get that patch merged into the gr-osmosdr project!

robotastic, 2021-07-20:
Great work @blantonl !!!

closed 2021-07-20 by robotastic