TrunkRecorder/trunk-recorder #404: Talkgroup priority behavior does not match documentation

ginomiglio, 2020-09-27:
From the `talkgroupsFile` section of the README:

>The priority field specifies the number of recorders the system must have available to record a new call for the talkgroup. For example, a priority of 1, the highest means as long as at least a single recorder is available, the system will record the new call. If the priority is 2, the system would at least 2 free recorders to record the new call, and so on. If there is no priority set for a talkgroup entry, a prioity of 1 is assumed.

The code to get the # of available recorders exists, but it isn't used. The current behavior simply ignores talkgroups with a priority of 100 or greater:
https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/source.cc#L434-L458

As discussed on Gitter, my Raspberry Pi 4 can't quite keep up with all of San Francisco's radio traffic across two RTL-SDRs, so I'd like to be able to use the priority feature as documented to drop only-sort-of-interesting talkgroups when the system is very busy, but record those talkgroups when overall traffic is light.

robotastic, 2020-09-27:
I've added support in support for talkgroup priority with this commit: https://github.com/robotastic/trunk-recorder/commit/7c71e217da20c0ee21efa26c31dad6c047fbe891

Give the 'duper-recorder' branch a try. The way it should work is if the priority for a talkgroup is greater than the number of available recorders it will not be recorded. So if you have 4 recorders, set your important talkgroups to 1 and the less important ones as 3 or 4.

ginomiglio, 2020-10-13:
Fix has been merged - thanks!

closed 2020-10-13 by ginomiglio