TrunkRecorder/trunk-recorder #406: rtlsdr_read_async errors

JustApu, 2020-09-28:
I used to have one instance of trunk-recorder that monitored two sites from the [New Jersey Interoperability Communications System](https://www.radioreference.com/apps/db/?sid=7021). One site was received well and other not so well so I recently separated them so I could better position the recorder for the second site.

But, now, the original trunk-recorder system, with the one site that was always received well in the past, errors out after about five minutes of running with a single "rtlsdr_read_async returned with -5" followed by never-ending pairs of "rtlsdr_read_async returned with -2." If I exit trunk-recorder and restart it, it immediately starts working again but then errors about five minutes later.

I didn't need the buflen=65536 in the original configuration with eight RTL-SDRs but added it to this new config and am still getting the rtlsdr_read_async errors.

Thoughts?

[config.txt](https://github.com/robotastic/trunk-recorder/files/5289287/config.txt)

robotastic, 2020-10-01:
So, in googling the error, I found a thread from 2014 where I was having the same error. It looks like the problem was with LibUSB. I think it has been fixed since then though...
https://github.com/csete/gqrx/issues/162

There was a change the introduced starting and stopping a graph to retune a system. I wonder if this is causing it. This was the commit: fd9553a6716a2a86a6ef2eef2b82433aa831f043

Could you try going back to the version before it?
https://github.com/robotastic/trunk-recorder/tree/0d7e852a0ef206c120036ec89933d5a9c9c513ed

JustApu, 2020-10-01:
git reset --hard 0d7e852

```In file included from /home/apu/testing/trunk-recorder/trunk-recorder/recorders/p25conventional_recorder.cc:1:0:
/home/apu/testing/trunk-recorder/trunk-recorder/recorders/p25_recorder.h: In member function ‘virtual void p25conventional_recorder::start(Call*)’:
/home/apu/testing/trunk-recorder/trunk-recorder/recorders/p25_recorder.h:202:46: error: ‘gr::op25_repeater::gardner_costas_cc::sptr p25_recorder::costas_clock’ is private
   gr::op25_repeater::gardner_costas_cc::sptr costas_clock;
                                              ^
/home/apu/testing/trunk-recorder/trunk-recorder/recorders/p25conventional_recorder.cc:67:7: error: within this context
       costas_clock->reset();
       ^
```

JustApu, 2020-10-01:
BTW, I rolled back to [v3.1.4](https://github.com/robotastic/trunk-recorder/releases/tag/v3.1.4) and it seemed like it last longer but still ended up with the same errors.

Thoughts on this being hardware? Haven't really done anything except remove some dongles to move to the new system but maybe something happened in the process?

JustApu, 2020-10-22:
Looks like a failing USB hub. Things look better now.

closed 2020-10-22 by JustApu