TrunkRecorder/trunk-recorder #410: Docker exec error on RPI4

rpdale, 2020-10-09:
I've installed Ubuntu 18, 20, and Raspian and all give me the same result after I run docker:

Digest: sha256:76218b5f985d0bcf77cca1f5599aeed97097e26d9221d7a63b8ef70aee6647c5
Status: Downloaded newer image for robotastic/trunk-recorder:latest
standard_init_linux.go:211: exec user process caused "exec format error"

Any thoughts on where I'm going wrong?

ginomiglio, 2020-10-10:
Raspberry Pi 4 uses an ARM processor, and the `latest` tag doesn't have an arm64 image, just amd64. If you change `latest` to `edge` or `nightly` in the docker command it may work. (Assuming you're using the 64-bit versions of those distros.)

https://hub.docker.com/r/robotastic/trunk-recorder/tags

That being said, I had other problems getting the docker image to work on 64-bit Raspbian a couple of months ago, and I ended up just building from scratch. So I'm interested to see how it goes.

(@robotastic is that a setting somewhere, or do will we start getting arm64 builds when `latest` is updated to a newer revision? more and more people will probably have this issue over time)

rpdale, 2020-10-10:
Thanks - I do see "arm64" in that link for the nightly and edge, but getting this for both:

docker: no matching manifest for linux/arm/v7 in the manifest list entries.

ginomiglio, 2020-10-18:
I think ARM v7 is 32-bit only. You need a 64-bit OS. Are you currently running Raspbian?

The release version of Raspbian is 32-bit, but there's a beta 64-bit version available here:
https://www.raspberrypi.org/forums/viewtopic.php?t=275370

Although you might just want to install the 64-bit version of Ubuntu instead:
https://ubuntu.com/download/raspberry-pi

robotastic, 2020-10-18:
Unfortunately Docker's BuildX system didn't have ARM7 support when I checked last. I have to go dig into all of this again, it may have been added. I will try to find the best approach. The Pi4 has gotten serious enough to do some real work.

rpdale, 2020-11-28:
BestBuy had RPI4 kits on clearance so I'm taking advantage of that :) I was able to get docker running on Ubuntu 20 with RPI4 / 64 bit with "edge" instead of "latest"

However after modifying the docker example with /home/ubuntu instead of home/pi, and after putting config.json into /home/ubuntu/trunk-recorder, I get:

Failed parsing config: /app/config.json

ndrwstn, 2020-12-22:
Using edge (0702ccac61797969cb648b00aed90ac1ec82420598240a1e9124a1da2dc8ef6a), I'm struggling to even get that far. Google is finding a lot of common problems but nothing that is helpful to getting this container running.

[image]

On - Ubuntu 20.04.1 LT (aarch64) on RPi4B+ (4GB).

robbiet480, 2021-03-22:
@ndrwstn I'm working as of a few seconds ago with this:

```yaml
version: '3'
services:
  recorder:
    image: robotastic/trunk-recorder:nightly
    container_name: trunk-recorder
    restart: always
    devices:
      - /dev/bus/usb
    environment:
      - LC_ALL=en_US.UTF-8
      - LANG=en_US.UTF-8
      - LANGUAGE=en_US.UTF-8
    volumes:
      - ./media:/app/media
      - ./config/config.json:/app/config.json
      - ./ebrcs.csv:/app/ebrcs.csv
```

robbiet480, 2021-03-23:
Updates on this:

1. With #442, the `latest` tag is being built properly now whenever a new release is cut
2. With #445 we are now building for the following platforms:

- `linux/amd64`
- `linux/arm64`
- `linux/arm/v7` (new)

We are limited on what platforms we can build for because Ubuntu only has support for these. Could investigate changing the base image for GNURadio to something with broader support.

leee, 2021-07-17:
Since all these were merged, I think this can be closed. (@robotastic)

closed 2021-07-20 by robotastic