TrunkRecorder/trunk-recorder #412: Smartnet - Analog and Static

jliedtke6014, 2020-10-17:
Smartnet System with Analog TG's
End of each transmission has 3-10 seconds of static since upgrading to TR3.3
Squelch set to -60

Happens on Ubunutu 20 with TR3.3 and Ubuntu 18 with TR3.3

Example of what I am hearing   https://voca.ro/1phnqLku49J

robotastic, 2020-10-18:
Did you update your Config file? Is Squelch list under your systems? It sounds like the squelch value is not working, and stopping recording when the signal levels drop. Instead the recordings are stopping when the SmartNet control channel stops advertising the broadcasts.  The squelch value could also need to be adjusted.

jliedtke6014, 2020-10-18:
> Did you update your Config file? Is Squelch list under your systems? It sounds like the squelch value is not working, and stopping recording when the signal levels drop. Instead the recordings are [...]

I did and below is my config,  this is happening on both my machine and my remote machine both since updating to TR3.3.  
Same config I had before I upgraded other than moving the settings from the sources down to the systems.  I've tried the squelch as high as -10 and as low as -90 and still the same thing. But I think your comment is right on the control channel part.

Isn't there someone else saying it is as if the squelch is working like it is set to 0  its in the Gitter chat right now and I think there are one or two more.

`{
   "ver":2,
   "sources":[
      {
         "center":142192500,
         "rate":2400000,
         "error":0,
         "gain":50,
         "digitalRecorders":5,
         "analogRecorders":5,
         "driver":"osmosdr",
         "device":"rtl=0"
      }
   ],
   "systems":[
      {
         "control_channels":[
            142380000,
            142710000
         ],
         "type":"smartnet",
         "squelch":-60,
         "modulation":"fsk4",
         "digitalLevels":"12",
         "analogLevels":"28",
         "recordUnknown":true,
         "bandplan":"400_custom",
         "bandplanBase":141015000.0,
         "bandplanHigh":143145000.0,
         "bandplanSpacing":15000,
         "bandplanOffset":380,
         "talkgroupDisplayFormat":"tag_id",
         "callLog":true,
         "hideEncrypted":true,
         "talkgroupsFile":"trs_tg_2560.csv",
         "hideUnknownTalkgroups":false,
         "shortName":"fleetnet-pembroke",
         "uploadScript":"upload-call.sh 1"
      }
   ],
   "defaultMode":"digital",
   "captureDir":"/recordings",
   "callTimeout":"3",
   "logFile":"0",
   "logLevel":"info",
   "frequencyFormat":"mhz",
   "broadcastifyCallsServer":"https://api.broadcastify.com/call-upload"
}`

jliedtke6014, 2020-10-18:
> Did you update your Config file? Is Squelch list under your systems? It sounds like the squelch value is not working, and stopping recording when the signal levels drop. Instead the recordings are [...]

If you want access to my box to test anything yourself let me know. I have no issues giving you access.

robotastic, 2020-10-19:
That is very weird - your config looks good. I just gave it a try and it looks like it is passing the new Squelch down to the recorder and it is setting the block. It could be that you cannot change the Squelch threshold once a Gnuradio graph has started. I will dive into that code. I did add in a print statement, if you want to grab the latest copy of master, it should print out the Squelch value everytime a new recording is started.

jliedtke6014, 2020-10-19:
Ignore the API errors, its because I don't have it uploading to my rdio-scanner website right now.
[2020-10-19 08:42:03.006319] (info)   Currently Active Calls: 0
[2020-10-19 08:42:03.006553] (info)   Recorders:
[2020-10-19 08:42:03.006612] (info)   [ rtl=0 ]
[2020-10-19 08:42:03.006660] (info)   [ D0 ] State: inactive
[2020-10-19 08:42:03.006712] (info)   [ D1 ] State: inactive
[2020-10-19 08:42:03.006770] (info)   [ D2 ] State: inactive
[2020-10-19 08:42:03.006814] (info)   [ D3 ] State: inactive
[2020-10-19 08:42:03.006861] (info)   [ D4 ] State: inactive
[2020-10-19 08:42:03.006914] (info)   [ A5 ] State: inactive
[2020-10-19 08:42:03.006998] (info)   [ A6 ] State: inactive
[2020-10-19 08:42:03.007060] (info)   [ A7 ] State: inactive
[2020-10-19 08:42:03.007128] (info)   [ A8 ] State: inactive
[2020-10-19 08:42:03.007169] (info)   [ A9 ] State: inactive
[2020-10-19 08:43:42.553633] (error)   Setting squelch to: -60
[2020-10-19 08:43:42.554309] (info)   [fleetnet-pembroke]       TG:              Renfrew D1 (     35680)        Freq: 142.125000 MHz    Starting Recorder on Src: rtl=0
[2020-10-19 08:43:46.485375] (info)   [fleetnet-pembroke]       TG:              Renfrew D1 (     35680)        Freq: 141.435000 MHz    Update Retuning - New Freq: 141.435000 MHz      Elapsed: 4s     Since update: 2s
[2020-10-19 08:44:08.013329] (info)   [fleetnet-pembroke]       TG:              Renfrew D1 (     35680)        Freq: 141.435000 MHz    Ending Recorded Call - Last Update: 4s  Call Elapsed: 26
[2020-10-19 08:44:08.014098] (info)   Running upload script: ./upload-call.sh 1 /recordings/fleetnet-pembroke/2020/10/19/35680-1603111422_142125000.wav &
[2020-10-19 08:44:09.518702] (info)   [fleetnet-pembroke]       TG: 35680       Freq: 141.435000 MHz    Broadcastify Upload Success - file size: 309956
[2020-10-19 08:44:12.804805] (error)   Setting squelch to: -60
[2020-10-19 08:44:12.805207] (info)   [fleetnet-pembroke]       TG:              Renfrew D1 (     35680)        Freq: 141.675000 MHz    Starting Recorder on Src: rtl=0
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  175k  100    46  100  175k     30   115k  0:00:01  0:00:01 --:--:--  115k
Invalid API key for system 1 talkgroup 35680.
[2020-10-19 08:44:37.009122] (info)   [fleetnet-pembroke]       TG:              Renfrew D1 (     35680)        Freq: 141.675000 MHz    Ending Recorded Call - Last Update: 4s  Call Elapsed: 25
[2020-10-19 08:44:37.009845] (info)   Running upload script: ./upload-call.sh 1 /recordings/fleetnet-pembroke/2020/10/19/35680-1603111452_141675000.wav &
[2020-10-19 08:44:38.664529] (info)   [fleetnet-pembroke]       TG: 35680       Freq: 141.675000 MHz    Broadcastify Upload Success - file size: 294504
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  130k  100    46  100  130k     41   117k  0:00:01  0:00:01 --:--:--  117k
Invalid API key for system 1 talkgroup 35680.

robotastic, 2020-10-20:
I just tested things out on the digital side and it looks like the Squelch block  works fine if it is set dynamically. I think the values are getting passed in correctly. Could you try a value like -10 and see if the block stays locked? If it works, then nothing should be recorded. If that works, try going up by -10, til -20 and so on, until the block starts unlocking. I am wondering if the Squelch value changed between versions because some of the filtering in the analog recorder was changed slightly.

jliedtke6014, 2020-10-20:
> I just tested things out on the digital side and it looks like the Squelch block works fine if it is set dynamically. I think the values are getting passed in correctly. Could you try a value like [...]

-10 and -20 are ok  anything lower then -20 is static.
 So something must have changed.   Well that solves that I guess.

closed 2020-10-20 by jliedtke6014