TrunkRecorder/trunk-recorder #414: Anyone know what "output_index must be < noutputs" means?  Getting this error on the latest build.. not sure what im doing wrong.. Any ideas where to look?

tdduarte, 2020-10-18:
terminate called after throwing an instance of 'std::out_of_range'
  what():  output_index must be < noutputs
Aborted (core dumped)

Moved Modulation and squelch to  systems..

Thanks!

robotastic, 2020-10-18:
That probably means I messed something up... can you share your config file? I will try running it and see what is causing the crash.

tdduarte, 2020-10-18:
{
    "ver": 2,
    "sources": [{
        "center": 453550000,
        "rate": 2048000,
        "error": 0,
        "gain": 60,
        "debugRecorders": 0,
	"digitalRecoders": 1,
        "driver": "osmosdr",
	"device": "rtl=01"
}, {
        "center": 460100000,
        "rate": 2048000,
        "error": 0,
        "gain": 60,
        "debugRecorders": 0,
        "analogRecoders": 3,
        "driver": "osmosdr",
        "device": "rtl=02"
 }
 ],
    "systems": [
        {
            "type": "conventionalP25",
            "channels": [453550000],
            "shortName": "dartmouthP25",
            "apiKey": "",
            "talkgroupsFile": "dartmouthp25.csv",
            "broadcastifyApiKey": "",
	    "broadcastifySystemId": "",
            "callLog": true,
	    "digitalLevels": 14,
            "audioArchive": false,
            "modulation": "fsk4",
            "squelch": -30,
            "minDuration": 0
}, {
            "type": "conventional",
            "channels": [460287500, 460087500, 460062500],
            "shortName": "dartmouthfd",
            "apiKey": "",
            "talkgroupsFile": "dartmouthfd.csv",
            "broadcastifyApiKey": "",
	    "broadcastifySystemId": "",
            "callLog": true,
            "audioArchive": true,
            "minDuration": 3,
            "squelch": -30,
            "decodeMDC": true
}
],
  "uploadServer": "https://api.openmhz.com",
	"broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
	"controlWarnRate": 5,
	"callTimeout": 0.75

}

robotastic, 2020-10-18:
Thanks - It looks like there is problem with P25 Conventional in FSK4 mode. I will go see if I can sort it out quick.

tdduarte, 2020-10-18:
OK great thank you!

robotastic, 2020-10-19:
OK - so I think the latest commit should fix... or atleast stop it from crashing. I couldn't test if it is recording correctly. Let me know if it works.

tdduarte, 2020-10-19:
looks good now! Thank you for the quick fix!

Cole0209, 2020-10-19:
I just saw this issue after I made mine. It looks like something is still up with the conventional p25 recording. @tdduarte Are you seeing any issues with your traffic being split into 5-6 second files? #415

tdduarte, 2020-10-19:
@Cole0209 you maybe right.. I seem to be getting all the transmissions but yea nothing over 6 secs

tdduarte, 2020-10-19:
Analog seems fine

tdduarte, 2020-10-19:
P25 clips at 6 secs then starts a new file for sure seen a few mid transmission. Analog seems to be fine

robotastic, 2020-10-20:
I figured out a way to test locally - I am just using the channels of a trunking system as P25 Conventional. I am seeing the same thing. Weird! I will add in lots of print statements and get to the bottom of this. For some reason I think the squelch may not be working right...

robotastic, 2020-10-20:
OK! Well, I forgot the Squelch block in P25 Recorder. I just tested and... NOW it should be working. Let me know though, clearly I mess this up a bit.

tdduarte, 2020-10-20:
Looks good here now no more clipping... Just need to play with the gain and squelch setting for the analog side but happy i can now do analog and p25 on one source.   Thank you for all your efforts with this project!

Cole0209, 2020-10-21:
@robotastic After these changes it looks like the P25 radio ids don't reset with every transmission. I had one 5 second transmission with 90+ radio ids attached... Restarting trunk recorder seems to reset the count.

robotastic, 2020-10-22:
Agh! yep, when I got rid of the seperate recorder for P25 conventional calls it got rid of reseting the source list for each new recording. added that back in. Thanks for putting up with all the bugs and reporting them!

robotastic, 2020-10-31:
I think we solved this. Please reopen if not.

closed 2020-10-31 by robotastic