TrunkRecorder/trunk-recorder #416: Why?? Unable to setup a System to record, exiting...

mathisono, 2020-10-21:
Could someone shed some light on why im getting : (error)   Unable to setup a System to record, exiting...

increasing the debug level doesn't show any thing...

```
./recorder --config=config_SF_metro_hackrf.json
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[2020-10-21 00:37:12.773847] (info)   Using Config file: config_SF_metro_hackrf.json

[2020-10-21 00:37:12.774102] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------

[2020-10-21 00:37:12.774117] (info)   
-------------------------------------
SYSTEMS
-------------------------------------

[2020-10-21 00:37:12.774136] (info)   

System Number: 1
-------------------------------------

[2020-10-21 00:37:12.774147] (info)   Short Name: bpd.csv
[2020-10-21 00:37:12.774155] (info)   System Type: p25
[2020-10-21 00:37:12.774162] (info)   Control Channels: 
[2020-10-21 00:37:12.774176] (info)     7.742188e+02
[2020-10-21 00:37:12.774218] (error)   ! System Modulation Not Specified, could be fsk4 or qpsk, assuming qpsk
[2020-10-21 00:37:12.774227] (info)   Analog Recorder Maximum Deviation: 4000
[2020-10-21 00:37:12.774235] (info)   Filter Width: 1
[2020-10-21 00:37:12.774245] (info)   Squelch: -160
[2020-10-21 00:37:12.774254] (info)   API Key: 
[2020-10-21 00:37:12.774262] (info)   Broadcastify API Key: 
[2020-10-21 00:37:12.774270] (info)   Broadcastify Calls System ID: 0
[2020-10-21 00:37:12.774277] (info)   Upload Script: 
[2020-10-21 00:37:12.774284] (info)   Call Log: true
[2020-10-21 00:37:12.774293] (info)   Audio Archive: true
[2020-10-21 00:37:12.774301] (info)   Loading Talkgroups...
[2020-10-21 00:37:12.774361] (error)   Warning: skipped 1 of 12 talkgroup entries! Invalid format?
[2020-10-21 00:37:12.774372] (error)   The format is very particular. See https://github.com/robotastic/trunk-recorder for example input.
[2020-10-21 00:37:12.774386] (info)   Talkgroups File: bpd.csv
[2020-10-21 00:37:12.774394] (info)   Loading Unit Tags...
[2020-10-21 00:37:12.774402] (info)   Unit Tags File: 
[2020-10-21 00:37:12.774411] (info)   Record Unknown Talkgroups: true
[2020-10-21 00:37:12.774420] (info)   Decode MDC: false
[2020-10-21 00:37:12.774427] (info)   Decode FSync: false
[2020-10-21 00:37:12.774436] (info)   Decode Star: false
[2020-10-21 00:37:12.774444] (info)   Decode TPS: false
[2020-10-21 00:37:12.774453] (info)   Talkgroup Display Format: Id
[2020-10-21 00:37:12.774463] (info)   Hide Encrypted Talkgroups: false
[2020-10-21 00:37:12.774471] (info)   Hide Unknown Talkgroups: false
[2020-10-21 00:37:12.774479] (info)   Minimum Call Duration (in seconds): 0
[2020-10-21 00:37:12.774490] (info)   
[2020-10-21 00:37:12.774497] (info)   

-------------------------------------
SOURCES
-------------------------------------

[2020-10-21 00:37:12.774522] (info)   Driver: osmosdr
[2020-10-21 00:37:12.774531] (info)   Center: 7.721700e+06
[2020-10-21 00:37:12.774547] (info)   Rate: 6000000
[2020-10-21 00:37:12.774563] (info)   Error: 0
[2020-10-21 00:37:12.774572] (info)   PPM Error: 53
[2020-10-21 00:37:12.774584] (info)   Auto gain control: false
[2020-10-21 00:37:12.774593] (info)   Gain: 0
[2020-10-21 00:37:12.774602] (info)   IF Gain: 32
[2020-10-21 00:37:12.774614] (info)   BB Gain: 40
[... 91 lines cut ...]
            "center": 772170000.0,
	    "rate":  10000000.0,
            "ifGain": 32,
	    "ppm": 53,
            "bbGain": 40,
            "digitalRecorders": 5,
            "driver": "osmosdr",
 	    "device": "hackrf=4f4397"

        }
    ],
    "systems": [{
	"LogLevel": "fatal",
	"shortName": "ebrcstest",
	"apiKey": "undefined"
	"frequencyFormat": "mhz",
        "control_channels": [774.21875],
        "type": "conventionalP25",
        "talkgroupsFile": "bpd.csv",
        "shortName": "bpd",
	"modulation": "qpsk"
    }]
"uploadServer": "https://api.openmhz.com"
}
```

jonk6, 2020-10-21:
One thing that jumps out at me is

`[bpd.csv]	Started with Control Channel: 7.742188e+02`

Try changing your config file from

`"control_channels": [774.21875],`

to

`"control_channels": [774218750],`

like the rest of your frequency references.

mathisono, 2020-10-21:
it was ```"control_channels": [77421875.0],```
looks like all of the example config files have the wrong values!? 

Is this error ok? Or am I missing something else? 

```
[2020-10-21 10:41:11.005829] (error)   [ebrcstest]	 Control Channel Message Decode Rate: 1/sec, count:  3
```

jonk6, 2020-10-22:
`looks like all of the example config files have the wrong values!?`
Which example config files are you looking at? All of the config file examples in the trunk-recorder/examples directory have frequencies written as Hz values. Or do you mean that the actual values used in the sample config files don't work for you?

Yes, the `Control Channel Message Decode Rate: 1/sec, count:  3` message is a problem.  The control channel specified in your may be incorrect or you may not be receiving it well enough to reliably decide it. Double check that you're specifying the correct frequency for the control channel . Your comment value of 77421875.0 is 77.421875 MHz but based on the other parameters in your config file I'm guessing you want 774.21875 MHz which is 774218750.0. Also you  try adjusting your gains and antenna placement if you think you should be able to receive that system.

robotastic, 2020-10-31:
@mathisono Checking back it see if you got things working and if you spotted some specific errors in the example folder.

mathisono, 2020-11-01:
Antenna gain setting and signal reception isn't the problem here!!!! 

I'm rather confused, all of the example config files don't work and something here is crashing my hack rf???

```
{
    "ver": 2,
	 "sources": [
        {
            "center": 775000000,
            "rate":90000000,
            "ifGain": 40,
	    "ppm": 52,
            "bbGain": 40,
            "digitalRecorders": 5,
            "analogRecorders": 0,
            "debugRecorders": 0,
            "driver": "osmosdr",
 	    "modulation": "qpsk",
            "device": "hackrf=4f4397"

        }
    ],
    "systems": [
	{
        "control_channels":[772132000, 772268000],
        "type": "p25",
        "talkgroupsFile": "EBRCS_TG.csv",
        "shortName": "ebrcstest",
	"apiKey": "*****"
    }
],
"uploadServer": "https://api.openmhz.com"
}
```

> @mathisono Checking back it see if you got things working and if you spotted some specific errors in the example folder.

robotastic, 2020-11-01:
The example code is just an example. you will have to customize it for your local system.
Try running `hackrf_info` to make sure the HackRF is working.

In your config file, the rate is set at 90mhz, the HackRF maxes out at 20mhz. I think 8mhz is a standard config. I would give that a try. `8000000`

bctrainers, 2020-11-15:
> The example code is just an example. you will have to customize it for your local system.
> Try running `hackrf_info` to make sure the HackRF is working.
>
> In your config file, the rate is set at [...]

On the HackRF, it will have severe drift unless he obtains a TXCO module for it. It'll result in no-decode issues. 

The other portion is the bandwidth the HackRF uses against TR. Most systems do not need more than 10MHz (1000000), for the ones that do, 12.5MHz (12500000) of spectrum coverages does the job. For me, I'm just lazy and set it to 15000000 (15MHz). :)

robotastic, 2020-11-21:
Closing - reopen if there are any updates

closed 2020-11-21 by robotastic