TrunkRecorder/trunk-recorder #417: Control channel scanning across SDRs not working

kb2ear, 2020-11-07:
When scanning control channels at start or anytime if the control channel scanning crosses between SDR receivers the control channel will never lock on. This is across both trunking types P25 & SmartNet

natecarlson, 2020-11-27:
Thread from discussion on gitter:

Luke Berndt
@robotastic
Oct 29 20:44
@kb2ear Does it try to tune in the different freq in the list or is it not even trying to tune in some of the freq?

Scott Weis KB2EAR
@kb2ear
Oct 29 20:46
It appears to scan through the frequencies, but never gets a lock

Scott Weis KB2EAR
@kb2ear
Oct 29 20:54
the logs follow

Scott Weis KB2EAR
@kb2ear
Oct 29 21:01
The first file has the errors when the wrong frequency is on top.
https://ref020.dstargateway.org/ubr_uploads/controlchannelscan1.txt

Scott Weis KB2EAR
@kb2ear
Oct 29 21:01
the second is when the correct control channel is on the top of the list
https://ref020.dstargateway.org/ubr_uploads/controlchannelscan2.txt

Scott Weis KB2EAR
@kb2ear
Oct 29 21:04
Here is the config file

Scott Weis KB2EAR
@kb2ear
Oct 29 21:04
https://ref020.dstargateway.org/ubr_uploads/config.json.txt

Luke Berndt
@robotastic
Oct 29 21:05
I wonder if the problem has something to do with have the control channel move from one SDR to another. Could you try ordering it so that it tries some of the other 856 & 857MHz control channels before going to the correct one?

Scott Weis KB2EAR
@kb2ear
Oct 29 21:07
There are only 2 channels on 856/857

Scott Weis KB2EAR
@kb2ear
Oct 29 21:12
So at around 2am they moved the control channel from 769.543750 to 856.687500 and it went into the never ending scan loop until I moved the 856 up to the top. And around 7pm they moved back to the 769 frequency and I again went back into the loop. So it very well may be when crossing between SDRs

Scott Weis KB2EAR
@kb2ear
Oct 29 21:14
Also on an unrelated note my XG-25 radio takes forever to scan for the crontrol channel when it's in the 856 channel.

Scott Weis KB2EAR
@kb2ear
Nov 07 01:05
I can confirm that as soon as the control channel scan crosses between SDRs the control channel will NEVER lock on. It is true on both P25 and SmartNet systems.

Nate Carlson
@natecarlson
Nov 07 10:16
Interesting! I don't think I've seen a control channel change yet on my system.. sounds pretty frequent on yours?

Scott Weis KB2EAR
[... 168 lines cut ...]

Nate Carlson
@natecarlson
Nov 08 20:59
I see you posted a smartnet example too. So strange! Do you have multiple systems being monitored on one instance of TR here or are these separate instances? Just clutching at straws..

Nate Carlson
@natecarlson
Nov 08 20:59
I mean p25 example of course.

Scott Weis KB2EAR
@kb2ear
Nov 08 21:27
seperate instances on the same computer, but I can reproduce the issue on a smartnet system on a seperate system in another state.

Nate Carlson
@natecarlson
Nov 08 21:28
Very strange. Not sure if it makes any difference - but my system is on Ubuntu 20.04 with the packaged versions of rtl-sdr libraries (0.6.0), gnuradio 3.8.1.0~rc1, gr-osmosdr 0.2.0, and osmo-sdr 0.1.8.

Luke Berndt
@robotastic
Nov 09 21:15
Well - I just tested with my SmartNet system and had the same problem as @kb2ear - the return does not seem to work. I started with some bad channels on one RTL-SDR and then had the good channels later down the list on a different SDR and it failed. It is able to handle retunes on the same SDR. I really don't see anything obvious though that is wrong. One thing I could try is delete the old SmartNet parsing block and create a new one, each retune. I will go try this on a P25 system too, just to double check.

rosecitytransit, 2021-01-10:
Was #334 supposed to address this?

robotastic, 2024-12-14:
Support has been added - it maybe a little buggy, but it is there

closed 2024-12-14 by robotastic