TrunkRecorder/trunk-recorder #419: decodeMDC no longer works

Cole0209, 2020-11-14:
I noticed after the recent changes were made to conventional recorders decoding of MDC tones for the radio ID no longer works on standard conventional audio. Looking back in my systems it may have happened around the same time issues were being had with conventional P25 radio ids from issue #414. I don't have any systems using the other decode types so I am unsure if those have also stopped working

JoeGilkey, 2020-11-15:
I've checked in a fix for this. The new adjustable sample rate on outputs needed to be passed into the decoder.

robotastic, 2020-11-16:
Thanks @JoeGilkey  I have merged in the PR. @Cole0209 let us know if that does the trick. I don't have a local system to test with.

Cole0209, 2020-11-18:
Yup looks like that fixed it!

closed 2020-11-18 by Cole0209