TrunkRecorder/trunk-recorder #422: SDRplay support?

frameshift18, 2020-11-23:
Been poking around trying to get my old SDRplay to work with TR. After all the SoapySDRplay set up it seems the gain settings on the SDRplays are quite different than the RTL based SDRs. I have got TR running with my SDRplay but TR never finds the control channel. I'm thinking not being able to modify the various gain settings is the culprit. It does work with CubicSDR correctly though so I feel I have the driver and Soapy set up correctly.

I think this could be an easy fix if all we have to do is pass the correct gain settings to osmosdr/Soapy.

I would be happy to troubleshoot and test..

frameshift18, 2020-11-23:
After messing with the PPM adjustment I do get good decode in TR. Seems without explicitly setting the gain values it goes in to AGC mode.

wx-jc, 2020-12-28:
@frameshift18 did you happen to document the process to get TR working with an SDRPlay?  I have an RSPduo and haven't been successful yet.  The amount of hoops you have to jump though to get all of the drivers working correctly is ridiculous.  I think I am getting close, but not quite there yet.  I was going to see how much better it performs for analog channels.

frameshift18, 2020-12-28:
I have been meaning to put together some sort of guide. Honestly I've messed with so many things I'm not sure what actually got it working lol. I used [this](https://github.com/pothosware/SoapySDRPlay3) soapysdrplay module that requires the v3 api binary. I remember reading somewhere the newer SDRPlays are only supported on the v3 api. I will try to dig into my SDRPlay system and see what I can gather.

NoahJaehnert, 2021-01-03:
@frameshift18 I had the same problem with an SDRPlay RSP2. I eventually gave up on it and bought an Airspy. 

I still have an RSP2 and can test if you do come up with the steps/a guide. 

One thing that might help is if you could share your config, or at least what you set the two gains IFGR and RFGR to? 

Thanks,

-Noah

jasonkotecki, 2021-04-14:
@frameshift18 can you share your TR config file? I have an RSP1a that I have been trying to get to work with TR and think I'm real close. I'm getting the error messages below, but I also see a warning related to setting the gain. I've pasted my current config for you to compare.

[WARNING] set_gain() is deprecated - please use set_gain(name) instead

[2021-04-14 10:22:27.459012] (debug)   process_data_unit timeout
[2021-04-14 10:22:28.489953] (debug)   process_data_unit timeout
[2021-04-14 10:22:29.007669] (error)   [MPSCS]  There is only one control channel defined
[2021-04-14 10:22:29.007901] (error)   [MPSCS]   Control Channel Message Decode Rate: 0.666667/sec, count:  2

{
    "ver": 2,
    "sources": [{
        "center": 852993750,
        "rate": 8000000,
        "gain": -40,
        "antenna": "RX",
        "digitalLevels": 3,
        "digitalRecorders": 4,
        "driver": "osmosdr",
        "device": "soapy,driver=sdrplay"
    }],
    "systems": [{
        "shortName": "MPSCS",
        "modulation": "qpsk",
        "squelch": -40,
        "audioArchive": "false",
        "control_channels": [853250000],
        "delayCreateOutput": true,
        "type": "p25",
        "talkgroupsFile": "talkgroups.csv"
        }],

    "captureDir": "/data/tr-media",
    "logFile": true,
    "logLevel": "trace",
    "frequencyFormat": "mhz"
}

frameshift18, 2021-04-15:
I've noticed on my RSP (the original) the PPM needs adjusting to around 2 or 3 depending on temperature. Sometimes when starting from cold the XO will drift enough that once it warms up I lose the CC and will need to add .5 to 1 to the PPM to get it locked on the CC again. I suggest playing around with the PPM a bit and see if you get any decode.

Since I have the RSP I don't have as many options, such as antenna port, which I think is helping me since there is less to potentially be set wrong by the driver by default that TR can't currently change.

Also when you first run TR it will spit out a bunch of helpful information related to the SDR source. I get errors about no gain set but the Soapy driver defaults to AGC since TR doesn't seem to be able to set the individual gains currently. I tried poking around in the source code but the additional Soapy layer gets me lost lol.

  "ver": 2,
    "sources": [{
        "center": 854000000,
        "rate": 7000000,
        "ifGain": 0,
        "gain": 0,
        "ppm": 2,
        "digitalRecorders": 12,
	    "digitalLevels": 4,
	    "driver": "osmosdr",
        "device": "soapy,driver=sdrplay"
    }]

jkotecki, 2021-04-15:
@frameshift18 thank you for your reply and config. I tried setting the ppm and I get a 'Segmentation fault (core dumped)'
It looks like my AGC is OFF, but I'm not sure how to enable it.

[2021-04-15 15:57:28.734087] (error)   ! No Gain was specified! Things will probably not work
[2021-04-15 15:57:28.734108] (info)   Auto gain control is OFF

leee, 2021-07-17:
There hasn't been activity on this issue for a few months. If anyone wants to reopen this please do so, but for now @robotastic lets close this.

closed 2021-07-20 by robotastic

devicenull, 2021-11-05:
So, some notes from my efforts here:

* Use `"device": "soapy,driver=sdrplay,agc_setpoint=0"`
* I had to patch source.cc set_lna_gain (line 86) to set the gain for the "RFGR" stage, and not the "LNA" stage.
  * It would be nice if trunk_recorder let us specify a list of gain settings, so maybe something like "RFGR=10,IFGR=20", rather then having individual json keys for each type of gain ( see https://github.com/pothosware/SoapySDRPlay3/issues/35 )
* IFGR setting cannot be changed if AGC is enabled (so, setting the AGC mode needs to happen before trying to change this gain stage) 
  * I just hardcoded `cast_to_osmo_sptr(source_block)->set_gain(25, "IFGR", 0);` in set_lna_gain to get stuff working
* To get the AGC to actually disable, I had to enable it first, and then disable it I really don't understand this, but it works:
<pre>
    cast_to_osmo_sptr(source_block)->set_gain_mode(true);
    cast_to_osmo_sptr(source_block)->set_gain_mode(false);
</pre>

My config looks like this:
<pre>
        "sources": [{
                        "center": "772000000",
                        "rate": "10000000",
                        "error": "0",
                        "ppm": "0",
                        "lnaGain": 1,
                        "digitalRecorders": 5,
                        "digitalLevels": 8,
                        "driver": "osmosdr",
                        "device": "soapy,driver=sdrplay,agc_setpoint=0"
        }],
</pre>

W3AXL, 2022-05-17:
FYI I am unable to get gain control to work with my RSP1 from a fresh build of `trunk-recorder`.

My source config:
```json
"sources": [{
        "center": 857487500,
        "rate": 8000000,
        "error": 10000,
        "agc": false,
        "gain_settings": {
            "RFGR": 0,
            "IFGR": 25
        },
        "digitalRecorders": 16,
        "driver": "osmosdr",
        "device": "soapy,driver=sdrplay,agc_setpoint=0"
    }],
```

And the output when the device is initialized:
```
[2022-05-16 22:24:33.883161] (info)   Source Device: soapy,driver=sdrplay,agc_setpoint=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd rfspace soapy redpitaya 
[INFO] devIdx: 0
[INFO] SerNo: B0002P0001
[INFO] hwVer: 1
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
[INFO] Using format CF32.
[2022-05-16 22:24:33.936044] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2022-05-16 22:24:33.936079] (info)   Setting sample rate to: 8000000
[2022-05-16 22:24:33.936112] (info)   Actual sample rate: 8000000
[2022-05-16 22:24:33.936124] (info)   Tuning to 8.574975e+08
[2022-05-16 22:24:33.936202] (info)   Gain Stage: IFGR supported values: 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 
[2022-05-16 22:24:33.936225] (info)   Gain Stage: RFGR supported values: 0 1 2 3 
[2022-05-16 22:24:33.936237] (info)   Max Frequency: 8.614555e+08
[2022-05-16 22:24:33.936251] (info)   Min Frequency: 8.535195e+08
[2022-05-16 22:24:33.936272] (info)   LNA Gain set to: 0
[2022-05-16 22:24:33.936304] (info)   Auto gain control is ON
[2022-05-16 22:24:33.936330] (info)   Setting antenna to [RX]
[2022-05-16 22:24:33.939961] (info)      P25 Recorder two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 8000000
[2022-05-16 22:24:33.946239] (info)      P25 Recorder ARB - Initial Rate: 8000000 Resampled Rate: 25000 Initial Decimation: 80 ARB Rate: 0.96
```

As you can see, AGC is not disabled properly and the IF/RF gain reductions are not properly set.