TrunkRecorder/trunk-recorder #424: Setting the gain

rpdale, 2020-12-04:
How do I use gqrx to determine the best gain setting?

natecarlson, 2020-12-04:
+1, agree that better documentation around this would be helpful.

Also, better documentation on using GNU Radio Companion with the debug recorders to validation and fine tune.

rpdale, 2020-12-04:
Thanks - I'm trying a simulcast system and not getting the best reception, so not sure if it's gain - or the rate (see other thread :) ) or the antenna positioning...

melbournefan, 2020-12-06:
I have to agree on this one, the documentation can be confusing.

One trick I found was when I'm running the instance, because I run it on a remote server I have to use rtl_tcp. When I connect to rtl_tcp in the command terminal it shows a 'index' value, so when I found the correct gain in GQRX I went and looked at the 'index' in my SSH terminal and then used that in the config.

Hope this helps

closed 2021-04-28 by robotastic