TrunkRecorder/trunk-recorder #425: Rate errors

rpdale, 2020-12-04:
WIth the RTL-SDR V3 USB, I set up a RATE of 3100000 to cover the frequencies on a MPSCS P25 system. But I could never get the control channel in trunk-recorder even though gqrx was showing a good signal.

After a lot (LOT) of trial and error - I realized the rate couldn't higher than 2890000. I ended up needing another RTL, but if you are running a RPI this is something to consider. Is this something hardware related (so extra RTLs will do it) or is there something going wrong in the software?

(Side note - I also found out that MPSCS simulcast sites us a different modulation than MPSCS non-simulcast :) )

natecarlson, 2020-12-04:
From what I've read (can't recall where exactly!), it's generally not recommended to run the RTL with a rate wider than 2096000 - wouldn't take this as gospel though.  :)

rpdale, 2021-01-05:
Thanks - I learned that through trial and error :) Documentation.

closed 2021-01-05 by rpdale