TrunkRecorder/trunk-recorder #427: Upload retry support?

natecarlson, 2020-12-30:
I'm seeing occasional upload errors due to gateway time-outs/etc.. it'd be nice to have optional retries for these - both to handle server-side issues (like the below appears to be), and transient internet failures/etc.

Of course, this starts the discussion of if this should be handled internally, or if there should be a queuing mechanism and a second daemon that handles the actual uploads, to get it outside of the trunk-recorder threads... :)

```
[2020-12-30 07:22:50.414703]: [scarmer2]        TG: 326 Freq: 851.125000 MHz    OpenMHz Upload Error: <html>
<head><title>504 Gateway Time-out</title></head>
<body>
<center><h1>504 Gateway Time-out</h1></center>
<hr><center>nginx/1.15.8</center>
</body>
</html>
[2020-12-30 07:25:29.476250]: [scarmer2]        TG: 324 Freq: 852.700000 MHz    OpenMHz Upload Error: <html>
<head><title>504 Gateway Time-out</title></head>
<body>
<center><h1>504 Gateway Time-out</h1></center>
<hr><center>nginx/1.15.8</center>
</body>
</html>
```

robotastic, 2020-12-30:
I think it would be best to have it be a separate daemon, or even a separate program that just gets called. It could end up being a nice general purpose program for uploading that could also be used to upload a bunch of old files. Adding in the Plugin System could really help this out.

rosecitytransit, 2021-01-10:
How about removing the upload process from the recorder program completely and instead have it be a separate program activated via the upload-encode script?

robotastic, 2024-12-14:
Upload retries is supported in later versions

closed 2024-12-14 by robotastic