TrunkRecorder/trunk-recorder #428: Problem compiling latest 3.3 on Armbian 20.11.3 Stretch

mbt25, 2020-12-30:
I am running an Odroid N2 with an Airspy on Armbian 20.11.3 Stretch.   I am currently running either 3.1 or 3.2 Trunk Recorder.   I'm attempting to build Trunk Recorder 3.3

I've got through nearly the whole make process until:
[ 99%] Building CXX object CMakeFiles/recorder.dir/lib/gr_blocks/selector_impl.cc.o
[100%] Linking CXX executable recorder
/usr/bin/ld: CMakeFiles/recorder.dir/lib/gr_blocks/selector_impl.cc.o: undefined reference to symbol '_ZN7log4cpp14CategoryStreamD1Ev'
//usr/lib/aarch64-linux-gnu/liblog4cpp.so.5: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/recorder.dir/build.make:1134: recipe for target 'recorder' failed
make[2]: *** [recorder] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/recorder.dir/all' failed
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Any ideas?   Like i said, I have never had any issues building previous versions on this platform.

Mike

mbt25, 2021-12-26:
In my first post above, at some point I got TR compiled/installed/working.   But, now I'm running a newer version of Armbian Stretch (ya know, just the typical apt update / upgrade / dist upgrade over the course of time.

I followed the wiki instructions for installing the new version and I'm getting the same errors and cannot figure it out.

/usr/bin/ld: CMakeFiles/trunk-recorder.dir/lib/gr_blocks/selector_impl.cc.o: undefined reference to symbol '_ZN7log4cpp14CategoryStreamD1Ev'
//usr/lib/aarch64-linux-gnu/liblog4cpp.so.5: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/trunk-recorder.dir/build.make:1112: recipe for target 'trunk-recorder' failed
make[2]: *** [trunk-recorder] Error 1
CMakeFiles/Makefile2:105: recipe for target 'CMakeFiles/trunk-recorder.dir/all' failed
make[1]: *** [CMakeFiles/trunk-recorder.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

robotastic, 2024-12-14:
Closing because it is prob OBE - reopen if there are still issues

closed 2024-12-14 by robotastic