TrunkRecorder/trunk-recorder #430: Number of Recorders, Thread Resource Error

Dewey3, 2021-01-14:
Greetings all.  Is there a hard limit to the number of recorders that can be used in Trunk Recorder?  I cannot seem to go over 12 recorders without the program giving a thread resource error and aborting.  I was running 2 RTL dongles set to 5 recorders each, and do not have any problems.  When I increase to 5 recorders on 1 dongle and 6 recorders on the other dongle, Trunk Recorder will throw a warning, but continue to work.  If I increase to 6 recorders on each of the 2 dongles, I get the thread resource error and subsequent abort.  I added another RTL dongle for 3 dongles total, but now I cannot go above 3 recorders per dongle without getting the same thread resource error and abort.  I monitor a phase 2-TDMA system that can be very busy and can easily tie up 5 or 6 recorders per dongle when it's busy.

kb2ear, 2021-01-15:
Are you running on a raspberry pi?

Dewey3, 2021-01-15:
Yup... I apologize, I forgot to mention.  Learning on a Raspberry Pi 400 running the latest Raspberry Pi OS Lite Image (1/11/2021).  Once I get the hang of things and am comfortable, my plan is to move everything over to a headless Raspberry Pi 4-8GB.

wx-jc, 2021-01-15:
Raspberry Pis don't really have the horsepower for that many recorders most likely.  Are you monitoring the CPU usage?  

There are people running this on decent computers with much higher than 11.

kb2ear, 2021-01-15:
You are going to need a more robust machine for anything more than maybe two RTL-SDR devices and a very inactive system.  Nothing before an i5. Multithreading is very important.

Dewey3, 2021-01-15:
So based on the responses here thus far, even if I do move over to an 8GB Raspberry Pi 4, I should not expect anything more since this appear to be more processor abilities than anything else ???  Thanks all.

kd4e, 2021-01-15:
Might this gain you some processing resources? https://software.intel.com/content/www/us/en/develop/articles/raspberry-pi-4-and-intel-neural-compute-stick-2-setup.html I'm only just learning how this helps with certain applications, but not with others.

kb2ear, 2021-01-15:
@Dewey3 Correct, the pi400 has a slightly better processor then the pi4 so you may even see worse performance. But 11 is the same number I get on a pi4 2gb.

@kd4e The will not improve anything related to processor threads available on the pi.

Dewey3, 2021-01-17:
Thanks much @kb2ear , that makes sense.  From all that I've heard concerning the Raspberry Pi, the 400's processor is supposed to be the same as the 4, just a later revision that they were able to clock a little higher than the 4 due to the 400's huge built in heatsink.

closed 2021-01-17 by Dewey3

kb2ear, 2021-01-29:
I have recently discovered that if I run:

ulimit -s 4096
before I start recorder I can have more than 11 recorder and I have tested up to 25.

bctrainers, 2021-01-29:
I completely forgot about ulimit and setting stack sizes (that's the -s in this instance).  

You may need to edit `/etc/security/limits.conf` to make such a setting permanent, however. 

```bash
*    soft    stack     4096
```
 would be what to put in to that.

Dewey3, 2021-01-29:
Thanks @kb2ear and @bctrainers !!!  My 3 RTL-SDR recorders were set at 4, 4, and 3, because of what appeared to be a hard limit of 11 recorders on my Pi 400.  I ran the ulimit command, changed the recorders to 7, 7, and 7, and everything is running smoothly.  For now, I'm PuTTY'd in from another computer and monitoring the CPU with HTOP, and it doesn't look like a problem.  In the end, I'm thinking I may dial it back to 5, 5, and 5, for a total of 15 recorders, but it's great knowing that I was able to break through my Pi 400's 11 recorder glass ceiling!  Thank you both again very much!

... now, back to chasing my "No such file or directory" audioplayer.php problem ...