TrunkRecorder/trunk-recorder #431: Airspy R2 Rate problem

rpdale, 2021-01-15:
Using gqrx I can select 2.5M, 3M, 6M, or 10M rates. Trunk-recorder only allows 2.5M and 10M. Any idea how to enable those other two options?

 - Rob

gonzoblue, 2021-07-16:
I have an airspy mini and have been able to use 6M and 3M, haven't tried any others. Just info, not sure if it helps anything.

leee, 2021-07-17:
What was trunk-recorder doing/reporting that didn't allow for 3 or 6 Msps? It seems like this other user can do it, so I think this issue can be closed for now (@robotastic) - but if you have any error messages/logs to share definitely open this issue up again.

robotastic, 2021-07-20:
Agreed - Trunk Recorder should allow for any value to be passed through to the driver. Let me know if there is a deeper issue though.

closed 2021-07-20 by robotastic