TrunkRecorder/trunk-recorder #434: audioplayer.php - what am I doing wrong?

Dewey3, 2021-01-25:
Greetings again to all.  Thanks to the great help here, I am now comfortable with Trunk Recorder on my current Raspberry Pi 400.  (My ultimate plan is to get everything "just right", and move it all over to a headless Raspberry Pi 4... which I also now know based on the help here that I will be restricted to 11 recorders).  The problem I am now having for the moment is trying to figure out audioplayer.  I've changed the php file line 2 to wav and the config file name on line 8 to the name of my config file.  When I run the php -S command, I have to use the IP address of the Pi400 rather than the 127.0.0.1 loopback, or the browsers on my other computers never run audioplayer.  When I do use my Pi400 IP address, audioplayer runs fine and looks great, but always returns "No such file or directory" when I attempt to download or listen to a recording.  I don't think it's a permissions or path issue since all the recordings get listed, the individual files are just "not found" when I click on an individual file to play it or download it.  I've also tried audioplayer-v2, but I'm not getting past "Undefined index: QUERY_STRING" on line 16, so for now, I'm just focusing on trying to get audioplayer to work.

rosecitytransit, 2021-02-06:
Can you see in your browser or possibly the teminial output the address it's trying to use for the audio files?  Looking at [PHP: Built-in web server - Manual](https://www.php.net/manual/en/features.commandline.webserver.php) one possibility is that PHP isn't looking in the right place.

As for the "Undefined index: QUERY_STRING", that'll happen if you're running the file from the command line because it doesn't use HTTP query strings. Try just commenting the line out as the if/else statements following it populate default values.

Dewey3, 2021-02-06:
Thanks much @rosecitytransit .  I am still learning Linux and all of the associated applications, so I'm sure it's something I'm missing somewhere.  I very strongly believe that it is a permissions issue somewhere that I am missing.

When I run `audioplayer.php`, it's showing the working directories and all files.  When I set the filter, that works correctly and I see the filter get set on the terminal where Trunk Recorder is running.  The problem is anytime I attempt to play or download a file from my laptop (`php -S 192.168.1.100:8080` at the terminal, and `192.168.1.100:8080/audioplayer.php` on the laptop), the terminal displays all of the correct file path and name information, but is followed by `No such file or directory`.  I've gone as far as to change the permissions of the top level recordings directory to 777 (& 0777), but that's not working.

When I run `audioplayer-v2.php` on my laptop, I get populated drop-downs for `'Day` and `Channel(s)`, but the `Month` drop-down never gets populated and I never get any file listings.  I did comment out the `Undefined index: QUERY_STRING` line, and while that resolved the "QUERY_STRING" error (thank you), nothing else has changed.

rosecitytransit, 2021-02-07:
If you want to temporarily set it up where I could log into it (either directly or via something like TeamViewer), I'd be willing to have a look at it. [E-mail me](audioplayer@jasonmchuff.net)

closed 2021-04-28 by robotastic