TrunkRecorder/trunk-recorder #435: MDC1200 number format

frameshift18, 2021-02-01:
Messing with MDC decoding here and it looks like the src id that is written into the JSON file is a decimal representation of the decoded MDC src id.

So for example a MDC unit id of 2216 will get written to the JSON file as 8726.

I haven't looked at where else the MDC unit id is used but could it just be converted to a string somewhere in the mdc decode functions?

robotastic, 2021-03-09:
Interesting! I do not have any MDC systems near me, so I can't really test. It would be this line here... 
https://github.com/robotastic/trunk-recorder/blob/master/lib/gr_blocks/decoders/signal_decoder_sink_impl.cc#L80
I think if you change the `x` to a `d` it should fix it. Let me know if that works.

Nokoa, 2021-03-13:
I noticed that the MDC unit ids decoded from TrunkRecorder are different than what my Motorola radio is decoding.

rml596, 2021-07-22:
I get the MDC message fine in the logs 
```json
{"type":"MDC1200","timestamp":"1626985908","op":"01","arg":"00","unitID":"0495","ex0":"00","ex1":"00","ex2":"00","ex3":"00"}
```
But it looks like that string is converted to an INT in the Call JSON
 ```json
{
"freq": 1.58835e+08,
"start_time": 1626985902,
"stop_time": 1626985920,
"emergency": 0,
"duplex": 0,
"mode": 0,
"priority": 0,
"call_length": 17.9928,
"talkgroup": 2,
"srcList": [ {"src": 1173, "time": 1626985900, "pos": 4.681438, "emergency": 0, "signal_system": "MDC1200", "tag": ""} ],
"freqList": [ ]
}
```

`1173` to hex is `0x495`, the unit number that comes over the radio is `495`. 

Where is the Call JSON built? I can look at playing around with the code.

zero48, 2021-07-23:
Few years algo I did a MDC1200 encoder (microcontoller bases) which worked
correctly with Motorola radios de coser (display) and Cimarrón profesional
decoders and always sent 4 characters un HEX from 0000 to DEEE if this info
may help , never un decimal , Also did GESTAR which was in decimal 00000 yo
16383

leee, 2021-08-30:
> If the code you are working on is in git just tell me which branch to build and I'm happy to test for you. I'm just getting going with trunk-recorder so I can restart and rebuild this as many times as I want without angering my non-existent listeners.

You can find v4.0 on branch `4.0-beta`. You can also find a tagged image for the beta branch on Docker Hub as well. ​v4 provides support for extensions/plugins using Boost, and moves around or even reimplements some extensions such as the uploaders. You should install using `make install` instead of moving files around, or use the `install.sh` script.

> My little hack works, but for some of the front ends I need to do things like sed the json and I'm not a huge fan.

I'm also a fan of piping things around for quick queries, but if you're building out an application for this, it seems you'd be better off using also having that application parse json and cast values. I'd imagine you can do this server-side, but you could possibly do this just as well client-side in-browser.

> Honestly a generic "iWantHex=true" and just convert nothing would be nice, or even better a per item and/or per system option to not convert would be great. I can imagine even just in my local area that some departments use hex and some use decimal with the unit number. I'll decode a few more and see how needed that is.
> 
> I don't think this is a high priority request, but I don't think it's incredibly difficult to address either, so I'd love to hear the maintainer's thoughts.

Not disagreeing with you, I'd much rather work on hex values across the board, but the use of decimal values is pretty strongly ingrained already both for ingested values and outputs, that working on this at the core is indeed closer to being "incredibly difficult to address" than it is trivial.

Applying some sort of "output transformation filter" would also require a lot of handling, not just config file handling additions and parsing judgement, but also output management for every single thing that gets consumed.

A simpler solution (which leans on the user) that solves your problem right now would be to have an `uploadScript` that modifies call jsons after the fact, loading the json, enumerating through `srcList` and replacing values in-situ or creating a new json. But if you're already handling the jsons, you might as well just integrate it into your application, right?

robotastic, 2021-09-02:
Yea, the right way to do this would be with input/output format converters. This was sort of started with some of the frequency output. The trick would be making sure it is done the same everywhere. If this could be done in a compact way,  I would be supportive of merging it in if someone wanted to tackle it. It would probably have to default to Decimal to keep things backwards compatible. I think I would also have to check on the OpenMHz side to make sure Hex numbers are supported. Some of the parsers maybe looking for decimal integers and would fall over if they saw any Alpha.

ZeroChaos-, 2021-09-02:
so after some quick testing, seding a hex representation into the json is a bad idea as it expects unquoted numbers and hex may violate that.  adding quotes makes the json right, but then rdio scanner (what I'm testing with because it looked easy to set up) can't read the id since it's not expecting a string.  I still think offering hex would be a great addition, but it looks to be a bit more painful as the whole ecosystem has to support it.

On the bright side, if you are a reader looking to solve your problems, I can highly recommend the unitTagsFile. I quite literally add "binary,hex" for each of the units I want to convert and the tag shows in the logs.  As a short term solution, this helps more than a little.

closed 2024-12-14 by robotastic