TrunkRecorder/trunk-recorder #437: Help with config file

shaffj01, 2021-02-21:
Hi!

We are attempting to setup trunk-recorder for use on openMHz and are having some issue with the config file. Being that I'm just the idea guy on this project, is anyone available to assist and provide some step by step guidance for us?

Thanks
Jeremy

rosecitytransit, 2021-03-02:
What system/site are you wanting to record? Did you get the program installed OK?

shaffj01, 2021-03-04:
The program is installed, but we were having some issues with teh
config.json file.

It is Lancaster County, PA Public safety (specifically fire & EMS) that we
are trying to record.

rosecitytransit, 2021-03-05:
I take it it's [this system](https://www.radioreference.com/apps/db/?sid=7641)? Which site? Also what type of SDR(s) are you using (it looks like all of the frequencies are very close together so 1 RTL SDR would work)?

According to [this page](https://wiki.radioreference.com/index.php/Programming_the_Lancaster_County_(PA%29_TRS) it appears to need a custom band plan. Looking at the README file, the code might need a change since trunk-recorder only seems to support that for SmartNet (analog) systems and not P25 digital ones.  I'd try ignoring that and see what happens.

A start at a config file:
````
{
    "ver": 2,
    "sources": [
        {
            "center": 476.740625,
            "rate": 1536000.0,
            "error": 0,
            "gain": 24,
            "digitalRecorders": 5,
            "analogRecorders": 0,
            "driver": "osmosdr",
            "device": "rtl=1"
        }
    ],
    "systems": [{
        "control_channels": [477.2125, 477.475],
        "type": "p25",
        "talkgroupsFile": "lcwc.csv",
        "shortName": "LCWC",
        "modulation": "fsk4"
    }]
}
````

kb2ear, 2021-04-21:
Did you ever get up and running, I can help you with a config if not. If you are up and running please close the issue on github.

closed 2021-04-28 by robotastic