TrunkRecorder/trunk-recorder #438: Recordings cutting off early on some transmissions with HackRF one and SmartNet 800Mhz Trunked System

jcummingsg2, 2021-03-03:
Configured for a SmartNet 800Mhz Trunk System - https://www.radioreference.com/apps/db/?sid=604   System is working well however sometimes the recording files are cut off and ended short even though the trunk channel stays active with additional transmissions for that conversation.  Seems like it may be related to retune from the logs and even though the control channel tracks is successfully with the correct duration, the recording does not seem to include all of the audio.  

Example:  (trunk-recorder log file for one transmission)
[2021-03-02 17:37:25.827372]: [kcers]   Started with Control Channel: 853.425000 MHz
[2021-03-02 17:37:25.827623]: SmartNet Trunking - SysNum: 0
[2021-03-02 17:37:25.827655]: Control channel: 853.425024 MHz
[2021-03-02 17:37:25.827809]:    smartnet Trunking two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 16000000
[2021-03-02 17:37:25.833448]:    smartnet Trunking ARB - Initial Rate: 16000000 Resampled Rate: 25000 Initial Decimation: 160 System Rate: 18000 ARB Rate: 0.72
...
[2021-03-02 14:29:09.339128]: [kcers]   TG:                  KCSO N (     23152)        Freq: 855.437500 MHz    Starting Recorder on Src: hackrf=26b468dc2e7d848f
[2021-03-02 14:29:16.898275]: [kcers]   TG:                  KCSO N (     23152)        Freq: 854.287500 MHz    Update Retuning - New Freq: 854.287500 MHz      Elapsed: 7s     Since update: 4s
[2021-03-02 14:29:20.672932]: [kcers]   TG:                  KCSO N (     23152)        Freq: 852.862500 MHz    Update Retuning - New Freq: 852.862500 MHz      Elapsed: 11s    Since update: 2s
[2021-03-02 14:29:26.011854]: [kcers]   TG:                  KCSO N (     23152)        Freq: 852.862500 MHz    Ending Recorded Call - Last Update: 6s  Call Elapsed: 17
[2021-03-02 14:29:26.012152]: Running upload script: ./encode-local-sys-1.sh /home/n7qob/trunk-build/kcers/2021/3/2/23152-1614724149_855437500.wav &
[2021-03-02 14:29:27.102472]: [kcers]   TG: 23152       Freq: 852.862500 MHz    Broadcastify Upload Success - file size: 71216
--

Control Channel indicates duration of the entire exchange is 17 seconds.  The recording file is only 5 seconds and only includes 2 transmission exchanges.  Its obvious from the recording that there is an additional response that is not captured in the recording.

[image]

[image]

[image]

MP3 File Example for this log.
[23152-1614724149_855437500.zip](https://github.com/robotastic/trunk-recorder/files/6073051/23152-1614724149_855437500.zip)

Running the latest version of Trunk-Recorder on Ubuntu 18.04 LTS with HackRFOne

bctrainers, 2021-03-03:
Paste your `config.json` file into a code block, please. 

Does your HackRF have a TXCO installed? If not, you're possibly suffering from light to severe frequency drift on the HackRF. It will cause clipped audio.

jcummingsg2, 2021-03-04:
Yes, I am running the TXCO installed on the HackRF board.

Config file used in my setup:

{
    "ver": 2,
    "sources": [{
        "center": 854000000,
        "rate": 16000000,
        "error": 0,
        "ppm": 0,
        "gain": 14,
        "ifGain": 24,
        "bbGain": 32,
        "analogRecorders": 24,
        "driver": "osmosdr",
        "device": "hackrf=26b468dc2e7d848f"
    }],
  "systems": [
    {
      "control_channels": [ 853425000, 853450000, 851812500, 856887500],
      "type": "smartnet",
      "bandplan": "800_reband",
          "modulation": "fsk4",
          "squelch": -42,
          "analogLevels": 8,
      "recordUnknown": "true",
          "talkgroupsFile": "tg_604.csv",
      "talkgroupDisplayFormat": "tag_id",
      "shortName": "kcers",
      "minDuration": 0,
          "broadcastifyApiKey": "xxxxxxxx",
          "broadcastifySystemId": "xxxxxxxx",
          "hideUnknownTalkgroups": "true",
          "uploadScript": "encode-local-sys-1.sh"
    }
  ],
  "broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
  "frequencyFormat": "mhz",
  "callTimeout": 5,
  "defaultMode": "analog",
  "controlWarnRate": 10,
  "logFile": "true",
  "logLevel": "info",
  "debugRecorderAddress": "xxxxxxxx",
  "debugRecorder": "false"
}

rosecitytransit, 2021-03-05:
I haven't tried it, but there's a debug mode where you can record or watch the spectrum as trunk-recorder gets it. If you don't want to use GNURadio, I believe SDR# or the like should be able to attach to the interface.  Looks like you'll need to put your real IP address in debugRecorderAddress to connect to it from a different computer.

gofaster, 2021-03-05:
It is possible that the received signal levels of the missing recordings were below the configured squelch level.  The analog recorder depends on the squelch to admit rx to the rest of the decode and audio file recording flowgraph.  Even if the tuned passband is on the freq defined by the control channel, if rx power is not enough to break squelch, nothing gets recorded.

0lav, 2021-03-17:
I get the same issue on a particular p25 phase 2system. I have tried switching out dongles with ones I know have not drifted but I still get some recordings that do that. I haven't figured out the reason yet. Can some talk groups switch to encrypted mid conversation? That was my only other guess.

kd4e, 2021-03-18:
Digital voice communications has always been a little quirky.

If the signal is strong then suddenly gets weak, or is interfered with, 
you can get garbled audio.

On 3/17/21 7:51 PM, 0lav wrote:
> I get the same issue on a particular p25 phase 2system. I have tried
> switching out dongles with ones I know have not drifted but I still
> get some recordings that do that. I haven't figured out the [...]

jcummingsg2, 2022-01-14:
Closing the issue as the version 4.x release changed the way end of calls are handled and seems to be working much better.  No longer have recordings cut off and seem to have complete calls captured with latest 4.2.2 release and have been running it stable for a few weeks now.  Thanks for all the work on the codebase.

closed 2022-01-14 by jcummingsg2