TrunkRecorder/trunk-recorder #440: Unitscript failure

rpdale, 2021-03-11:
I’m getting a syntax error on Line 11 (“elif”) and I’m no Bash expert - but I think elif has to have a test parameter behind it. And I don’t think “!=“ is used in Bash either?

Also the readme seems to imply that I just put the script into the ~/trunk-recorder directory, but it couldn’t find it when I ran recorder. It worked with the fully delimited name (after I first noticed the file was executable -but that’s when the syntax errors popped up.)

robotastic, 2021-03-12:
Great catches - It was a little too eager to merge things in. @rosecitytransit is going to build out the example script a little more. `unitScript` and `OnUnitChange` are the same thing. I changed the name to make it a little clearer what the parameter was enabling.

that is interesting... it may not like relative directories. let me go check that out.

The conversation on this feature is here:
https://github.com/robotastic/trunk-recorder/pull/181

rpdale, 2021-03-12:
Excellent! I started a little Bash research myself just so I can play more later. Eventually I’d like to figure out how to mark or otherwise pull aside transmissions coming from my radio for “post-event review.”

rosecitytransit, 2021-03-12:
Great to hear people are already trying to use the feature. It makes me wish I could have gotten it in 3 years ago, but oh well.

As for the config entry, you probably need to include `./` at the start for relative paths, just like if you were running a script directly in the terminal (I didn't put any presumptions about the path in the code; I'll add a note to the readme entry).

If no one else does it, I do indeed plan to get the script finished and tested, either tonight or in the next few days.

As for saving transmissions from a specific radio, that's entirely possible.  On one of my installs, I have the radio IDs being put in the file name so that would be an easy `find` command. An `if` statement in this script could act on seeing a specific radio, but I'm not sure you could use it to automatically save calls since it doesn't get the file names (neither the frequency nor the precise time stamp).

My next goal is an option to create daily call log files (instead of hundreds of JSON files); this could be easily searched for radio IDs. For now, I think you'd need a script that reads every JSON file and searches through the list of radios.

rosecitytransit, 2021-03-13:
Still not working (getting a sed error) but here is what I have now. I noticed that @CrimeIsDown tried unit logging in #411 : b2987a2be0a5866a6e6008eee7b4830d7274076a
````
#!/bin/bash

CAPTUREDIR="test" #fill this in with the path used in config.json, no ending /

# Creates a list of radio IDs in a file named "radiolist.csv" located in the
# shortName directory along side the recordings, and logs of radio activity to
# "radiolog.csv" located in each day's recordings directory

# file format: radioID, timestamp, action, talkgroup
# for radiolist.csv, acknowledgment response timestamp in 5th column when it was
# seen after other action

# Feel free to customize this script; to use for multiple systems, include in each
# system's config.json section

# NOTE: For efficiency, this script does not check for/create the radio log files,
# so they MUST exist beforehand. Run "echo > radiolist.csv" where you are keeping
# it or them, and set up a daily cron task of &&

# sed script based on https://stackoverflow.com/a/49852337

printf -v TRDATE '%(%Y/%-m/%-d)T'
printf -v NOWTIME '%(%s)T'
if [ ! "$3" == "ackresp" ]; then
  sed -i -e 's/^'${2}',.*/{s/'${2}','${NOWTIME}','${3}','${4}',/;:a;n;ba;q}' -e '$a'${2}','${NOWTIME}','${3}','${4}',' $CAPTUREDIR/$1/radiolist.csv
else
  sed -i -e 's/^'${2}',\([0-9]*\),\([on|join|off|call]\),\([0-9]*\),\([0-9]*\)/{s/'${2}',\1,\2,\3,'${NOWTIME}'/;:a;n;ba;q}' -e '$a'${2}','${NOWTIME}',ackresp,,' $CAPTUREDIR/$1/radiolist.csv
fi
echo "$2,$NOWTIME,$3,$4" >> $CAPTUREDIR/$1/$TRDATE/radiolog.csv
````

rosecitytransit, 2021-03-13:
OK, I think this is working. Go ahead and try it, and if so, I'll submit a pull request:
```bash
#!/bin/bash

CAPTUREDIR="test" #fill this in with the path used in config.json, no ending /

# Creates a list of radio IDs in a file named "radiolist.csv" located in the
# shortName directory along side the recordings, and logs of radio activity to
# "radiolog.csv" located in each day's recordings directory

# file format: radioID, timestamp, action, talkgroup
# for radiolist.csv, acknowledgment response timestamp in 5th column when it
# was seen after other action

# Feel free to customize this script; to use for multiple systems, include in
# each system's config.json section

# NOTE: You need to run "echo > radiolist.csv" where the file(s) is going to be
# beforehand as sed doesn't work on empty files, and to capture actions before
# trunk-recorder makes the daily directory upon recording the first call,
# set up a daily cron task of mkdir -p <capturedir>/

# sed usage based on https://stackoverflow.com/a/49852337

printf -v TRDATE '%(%Y/%-m/%-d)T'
printf -v NOWTIME '%(%s)T'
if [ ! "$3" == "ackresp" ]; then
  sed -i -e '/^'${2}',.*/{s//'${2}','${NOWTIME}','${3}','${4}',/;:a;n;ba;q}' -e '$a'${2}','${NOWTIME}','${3}','${4}',' $CAPTUREDIR/$1/radiolist.csv
else
  sed -i -e '/^'${2}',\([0-9]*\),ackresp,,/{s//'${2}','${NOWTIME}',\1,ackresp,,/;:a;n;ba;q}' -e '/^'${2}',\([0-9]*\),\([a-z]*\),\([0-9]*\),\([0-9]*\)/{s//'${2}',\1,\2,\3,'${NOWTIME}'/;:a;n;ba;q}' -e '$a'${2}','${NOWTIME}',ackresp,,' $CAPTUREDIR/$1/radiolist.csv
fi
#echo "$2,$NOWTIME,$3,$4" >> $CAPTUREDIR/$1/$TRDATE/radiolog.csv
```

rpdale, 2021-03-13:
It’s ***@***.***:~/mpscs $ cat radiolist.csv9383269,1615640130,call,1237,1000002,1615640228,call,1237,9508328,1615640165,call,12627,9383794,1615640162,call,1237,9505391,1615640174,call,12627,1001708,1615640216,call,11923,9383549,1615640195,call,1237,10233231,1615640213,call,11923,9380225,1615640230,call,1237,9230049,1615640233,call,1258, From: Jason McHuffSent: Saturday, March 13, 2021 12:15 AMTo: robotastic/trunk-recorderCc: rpdale; AuthorSubject: Re: [robotastic/trunk-recorder] Unitscript failure (#440) OK, I think this is working. Go ahead and try it, and if so, I'll submit a pull request:#!/bin/bash CAPTUREDIR="test" #fill this in with the path used in config.json, no ending / # Creates a list of radio IDs in a file named "radiolist.csv" located in the# shortName directory along side the recordings, and logs of radio activity to# "radiolog.csv" located in each day's recordings directory # file format: radioID, timestamp, action, talkgroup# for radiolist.csv, acknowledgment response timestamp in 5th column when it# was seen after other action # Feel free to customize this script; to use for multiple systems, include in# each system's config.json section # NOTE: You need to run "echo > radiolist.csv" where the file(s) is going to be# beforehand as sed doesn't work on empty files, and to capture actions before# trunk-recorder makes the daily directory upon recording the first call,# set up a daily cron task of mkdir -p <capturedir>/ # sed usage based on https://stackoverflow.com/a/49852337 printf -v TRDATE '%(%Y/%-m/%-d)T'printf -v NOWTIME '%(%s)T'if [ ! "$3" == "ackresp" ]; then  sed -i -e '/^'${2}',.*/{s//'${2}','${NOWTIME}','${3}','${4}',/;:a;n;ba;q}' -e '$a'${2}','${NOWTIME}','${3}','${4}',' $CAPTUREDIR/$1/radiolist.csvelse  sed -i -e '/^'${2}',\([0-9]*\),ackresp,,/{s//'${2}','${NOWTIME}',\1,ackresp,,/;:a;n;ba;q}' -e '/^'${2}',\([0-9]*\),\([a-z]*\),\([0-9]*\),\([0-9]*\)/{s//'${2}',\1,\2,\3,'${NOWTIME}'/;:a;n;ba;q}' -e '$a'${2}','${NOWTIME}',ackresp,,' $CAPTUREDIR/$1/radiolist.csvfi#echo "$2,$NOWTIME,$3,$4" >> $CAPTUREDIR/$1/$TRDATE/radiolog.csv

rpdale, 2021-03-13:
I'm not sure if it's a script issue or software issue... but I'm only getting "calls" - no joins or any other status so far.

[image]
[image]

rosecitytransit, 2021-03-13:
Are you using the latest one, in the new pull request, and is this in
radiolist.csv or radiolog.csv?

rosecitytransit, 2021-03-13:
I should add that if you're only getting call entries, then it makes me think the script isn't being called correctly, since it's done differently for calls (as the program, of course, actually processes calls unlike with the other actions).

@robotastic or whoever, am I doing it correctly? The code flow is: 
```C++
unit_registration(sys->get_unit_script(), message.source);

void unit_registration(string unit_script, long unit) {
  unit_affiliations[unit] = 0;

  if (unit_script.length() != 0) {
    char   shell_command[200];
    sprintf(shell_command, "%s %li on &", unit_script.c_str(), unit);
    int rc = system(shell_command);
  }
}
```
I know I could pass `sys` like `handle_call` does, but I don't need the entire system object, just the script name.

rpdale, 2021-03-13:
I updated the script to the file versus the reply here and same response... Only calls are showing up in both .csv files

In addition - my reception isn't perfect, so may be to blame - but occasionally getting these errors where it appears to be using the radio ID as the system name (should be 'mpscs')

/media/pi/F4D9-02F5/9384148/2021/3/13/radiolog.csv:

rosecitytransit, 2021-03-13:
> /media/pi/F4D9-02F5/9384148/2021/3/13/radiolog.csv:

Presumably, the shortName is not getting passed to the script, which causes the following parameters to shift over (I'm betting that file doesn't include the radio IDs). My guess is that trunk-recorder is decoding a system ID which does not match the existing one that the shortName has been applied to, and creates a new system. But unless the directory structure already exists, I don't see how the file is getting created since the `echo` command needs it and the script doesn't create it.

If you could attach or link to the trunk-recorder log file, that might be helpful.

rpdale, 2021-03-13:
There's nothing terribly exciting that I see in the logfile before that happens. 9239955 is a valid MPSCS radio ID, as well as 10233242.

[2021-03-13 18:41:09.594562] (info)   	- Starting P25 Recorder Num [5]	TG:       1258	Freq: 7.707062e+08 	TDMA: false	Slot: 0	Mod: false
[2021-03-13 18:41:09.596169] (info)   [mpscs]	TG:       1258	Freq: 7.707062e+08	Starting Recorder on Src: rtl=2
[2021-03-13 18:41:24.006429] (info)   Currently Active Calls: 1
[2021-03-13 18:41:24.006850] (info)   TG:       1258 Freq: 7.707062e+08 Elapsed: 15 State: recording
[2021-03-13 18:41:24.007164] (info)   	[ 5 ] State: active
[2021-03-13 18:41:24.007278] (info)   Recorders: 
[2021-03-13 18:41:24.007378] (info)   [ rtl=1 ]  
[2021-03-13 18:41:24.007480] (info)   [ D0 ] State: inactive
[2021-03-13 18:41:24.007571] (info)   [ D1 ] State: inactive
[2021-03-13 18:41:24.007662] (info)   [ D2 ] State: inactive
[2021-03-13 18:41:24.007748] (info)   [ D3 ] State: inactive
[2021-03-13 18:41:24.007839] (info)   [ D4 ] State: inactive
[2021-03-13 18:41:24.007931] (info)   [ rtl=2 ]  
[2021-03-13 18:41:24.008020] (info)   [ D5 ] State: active
[2021-03-13 18:41:24.008106] (info)   [ D6 ] State: inactive
[2021-03-13 18:41:24.008195] (info)   [ D7 ] State: inactive
[2021-03-13 18:41:43.006907] (info)   [mpscs]	TG:       1258	Freq: 7.707062e+08	Ending Recorded Call - Last Update: 4sCall Elapsed: 34
[2021-03-13 18:41:43.150012] (info)   	- Stopping P25 Recorder Num [5]	TG:       1258	Freq: 7.707062e+08 	TDMA: false	Slot: 0
[2021-03-13 18:41:45.023567] (info)   [mpscs]	TG: 1258	Freq: 7.707062e+08	Broadcastify Upload Success - file size: 105311
[2021-03-13 18:42:04.009000] (info)   [mpscs]	TG:       1258	Freq: 7.707062e+08	TG not in Talkgroup File 
[2021-03-13 18:42:04.009660] (info)   	- Starting P25 Recorder Num [5]	TG:       1258	Freq: 7.707062e+08 	TDMA: false	Slot: 0	Mod: false
[2021-03-13 18:42:04.010406] (info)   [mpscs]	TG:       1258	Freq: 7.707062e+08	Starting Recorder on Src: rtl=2
[2021-03-13 18:42:04.779131] (info)   [mpscs]	TG:      12627	Freq: 7.713812e+08	TG not in Talkgroup File 
[2021-03-13 18:42:04.779398] (info)   	- Starting P25 Recorder Num [6]	TG:      12627	Freq: 7.713812e+08 	TDMA: false	Slot: 0	Mod: false
[2021-03-13 18:42:04.779529] (info)   [mpscs]	TG:      12627	Freq: 7.713812e+08	Starting Recorder on Src: rtl=2
[2021-03-13 18:42:10.148803] (info)   [mpscs]	TG:      12627	Freq: 7.697188e+08	Update Retuning - New Freq: 7.697188e+08	Elapsed: 6s 	Since update: 2s
sed: can't read /media/pi/F4D9-02F5/9239955/radiolist.csv: No such file or directory
/home/pi/trunk-recorder/unit-script.sh: line 14: /media/pi/F4D9-02F5/9239955/2021/3/13/radiolog.csv: No such file or directory
sed: can't read /media/pi/F4D9-02F5/9239955/radiolist.csv: No such file or directory
/home/pi/trunk-recorder/unit-script.sh: line 14: /media/pi/F4D9-02F5/9239955/2021/3/13/radiolog.csv: No such file or directory

9232043,1615677035,call,1258,
9230852,1615679266,call,1258,
9230049,1615676568,call,1041,
9230047,1615678128,call,1041,
9230904,1615676735,call,1041,

[2021-03-13 18:34:36.982562] (error)   [mpscs]	TG: 12627	Freq: 7.713812e+08	Broadcastify Metadata Upload Error: SKIPPED---ALREADY-RECEIVED-THIS-CALL
[2021-03-13 18:34:42.018009] (info)   Currently Active Calls: 1
[2021-03-13 18:34:42.018310] (info)   TG:       1258 Freq: 7.707062e+08 Elapsed: 24 State: recording
[2021-03-13 18:34:42.027607] (info)   	[ 5 ] State: active
[2021-03-13 18:34:42.027989] (info)   Recorders: 
[2021-03-13 18:34:42.028166] (info)   [ rtl=1 ]  
[2021-03-13 18:34:42.028278] (info)   [ D0 ] State: inactive
[2021-03-13 18:34:42.028382] (info)   [ D1 ] State: inactive
[2021-03-13 18:34:42.028484] (info)   [ D2 ] State: inactive
[2021-03-13 18:34:42.028612] (info)   [ D3 ] State: inactive
[2021-03-13 18:34:42.028743] (info)   [ D4 ] State: inactive
[2021-03-13 18:34:42.028856] (info)   [ rtl=2 ]  
[2021-03-13 18:34:42.028966] (info)   [ D5 ] State: active
[2021-03-13 18:34:42.029068] (info)   [ D6 ] State: inactive
[2021-03-13 18:34:42.029163] (info)   [ D7 ] State: inactive
[2021-03-13 18:35:03.003593] (info)   [mpscs]	TG:       1258	Freq: 7.707062e+08	Ending Recorded Call - Last Update: 4sCall Elapsed: 45
[2021-03-13 18:35:03.149963] (info)   	- Stopping P25 Recorder Num [5]	TG:       1258	Freq: 7.707062e+08 	TDMA: false	Slot: 0
[2021-03-13 18:35:06.965822] (info)   [mpscs]	TG: 1258	Freq: 7.707062e+08	Broadcastify Upload Success - file size: 197351
sed: can't read /media/pi/F4D9-02F5/10233242/radiolist.csv: No such file or directory
/home/pi/trunk-recorder/unit-script.sh: line 14: /media/pi/F4D9-02F5/10233242/2021/3/13/radiolog.csv: No such file or directory
sed: can't read /media/pi/F4D9-02F5/10233242/radiolist.csv: No such file or directory
/home/pi/trunk-recorder/unit-script.sh: line 14: /media/pi/F4D9-02F5/10233242/2021/3/13/radiolog.csv: No such file or directory
[2021-03-13 18:36:10.283197] (info)   [mpscs]	TG:       1237	Freq: 8.519750e+08	TG not in Talkgroup File 
[2021-03-13 18:36:10.283577] (info)   	- Starting P25 Recorder Num [0]	TG:       1237	Freq: 8.519750e+08 	TDMA: false	Slot: 0	Mod: false
[2021-03-13 18:36:10.284078] (info)   [mpscs]	TG:       1237	Freq: 8.519750e+08	Starting Recorder on Src: rtl=1
[2021-03-13 18:36:17.658789] (info)   [mpscs]	TG:       1237	Freq: 8.528500e+08	Update Retuning - New Freq: 8.528500e+08	Elapsed: 7s 	Since update: 3s
[2021-03-13 18:36:27.001575] (info)   [mpscs]	TG:       1237	Freq: 8.528500e+08	Ending Recorded Call - Last Update: 4sCall Elapsed: 17

10233298,1615672318,call,1258,
10233216,1615672775,call,1258,
10233287,1615675381,call,11923,
10233220,1615675335,call,1258,
10233232,1615675386,call,1258,
10233250,1615675414,call,1258,
10233208,1615679416,call,1258,
10233288,1615676001,call,1258,
10233234,1615676297,call,1258,
10233281,1615677576,call,11923,
10233240,1615678646,call,1258,

rosecitytransit, 2021-03-14:
A thought occurred to me: it's NOT a decoding issue (I'm not even another
system is set up when a different System ID is seen), but that the errors
are all the joins, etc and what's happening is that I'm not accessing the
shortName correctly so it's not getting passed.

For now, if you wanted you could replace the $1 in the script with your
actual shortName. To get the data into the correct columns, you'd have to
shift the parameters over when it's not a call.

rosecitytransit, 2021-03-16:
I only just now realized that I'm not even trying to pass shortName (`sys->get_short_name()`) at all... (my brain was thinking `sys->get_unit_script()` does that). Oops.

I think I have the issue fixed in the new pull request (#441) if you can try that. Just replace trunk-recorder/trunk-recorder/main.cc with [this one](https://github.com/robotastic/trunk-recorder/blob/7569c98f398930ee6b258ce7bf53e1a40b408279/trunk-recorder/main.cc)

rpdale, 2021-03-16:
No change with a rebuild. It is getting things right most of the time, but still occasionally using the radio ID as the system name.

And radiolist.csv no longer has anything at all in it. The file is empty. radiolog.csv is filling up (other than when it gets the radio ID in there instead.)

sed: can't read /media/pi/F4D9-02F5/9230099/radiolist.csv: No such file or directory
/home/pi/trunk-recorder/unit-script.sh: line 14: /media/pi/F4D9-02F5/9230099/2021/3/16/radiolog.csv: No such file or directory

rosecitytransit, 2021-03-21:
Finally getting back to this.

In trunk-recorder/trunk-recorder/main.cc, do you find `sys->get_unit_script(), sys->get_short_name()`? I'm pretty sure I'm calling both the same way, so I'm not sure why one would show up and not the other.

If radiolist.csv is completely empty, open the file, press enter a couple times to generate a blank line or two and save it. As I said, sed won't work on empty files.

Try replacing occurrences of `$1` in unit-script.sh with your actual short name, mpscs.

I have an extra SDR so I'm going to work on getting it set up and tested myself.

rpdale, 2021-03-21:
I hardcoded the unit-script.sh file and that changed things a bit… I am now seeing on/join/add instead of just calls in the log. But something doesn’t look right still. radiolist.csv:1199991,1616336409,call,1191,0,1616336425,call,1191,on,1616336492,,,join,1616336492,11923,, radiolog.csv:0,1616336252,call,11910,1616336252,call,11910,1616336253,call,1191on,1616336298,,on,1616336298,,join,1616336298,11922,join,1616336298,11922,1199991,1616336408,call,11911199991,1616336408,call,11911199991,1616336409,call,1191   From: Jason McHuffSent: Sunday, March 21, 2021 1:58 AMTo: robotastic/trunk-recorderCc: rpdale; AuthorSubject: Re: [robotastic/trunk-recorder] Unitscript failure (#440) Finally getting back to this.In trunk-recorder/trunk-recorder/main.cc, do you find sys->get_unit_script(), sys->get_short_name()? I'm pretty sure I'm calling both the same way, so I'm not sure why one would show up and not the other.If radiolist.csv is completely empty, open the file, press enter a couple times to generate a blank line or two and save it. As I said, sed won't work on empty files.Try replacing occurrences of $1 in unit-script.sh with your actual short name, mpscs.I have an extra SDR so I'm going to work on getting it set up and tested myself.

rosecitytransit, 2021-03-21:
What doesn't look right? It's a little hard for me to tell since the line breaks didn't come through, but it looks OK.

rpdale, 2021-03-21:
Sorry - I forgot emailing replies isn't so pretty :) radiolog.csv:

[image]

rosecitytransit, 2021-03-22:
I got the new version of trunk-recorder set up and running! Yay!

So in unit-script.sh, remove the `\1,` in the second sed call. So `'${NOWTIME}',\1,ackresp` should be `'${NOWTIME}',ackresp`. Other then that, it's working for me, even without hard coding the shortName. I'll also make it so messages with radio ID of 0 are ignored.

radiolist.csv:
```
11823,1616409075,join,1437,
20728,1616409077,join,1257,
224,1616409078,call,703,
13553,1616409078,call,1579,
15409,1616409080,ackresp,,
22272,1616409094,call,1763,
222,1616409085,call,471,
18216,1616409092,ackresp,,
```

radiolog.csv
```
11823,1616409075,on,
11823,1616409075,on,
0,1616409075,call,1687
0,1616409075,call,471
11823,1616409075,join,1437
0,1616409075,call,1687
0,1616409075,call,471
0,1616409075,call,1687
0,1616409075,call,471
20728,1616409076,on,
20728,1616409076,join,1257
20728,1616409076,join,1257
```

rpdale, 2021-03-23:
Here's with that edit.

[image]

rosecitytransit, 2021-03-23:
Weird, because the on/off/join messages which look messed up on yours are working fine for me:
```
12340,1616412500,ackresp,,
12344,1616412500,ackresp,,
12328,1616412504,call,1193,
13496,1616413276,call,1579,
12320,1616471353,join,1809,
16169,1616471357,join,1809,
15312,1616471357,off,,
220,1616471408,call,803,
22270,1616471403,call,1777,
12894,1616471364,on,,
14603,1616471364,call,803,
17253,1616471369,join,2219,
12787,1616471377,ackresp,,
10425,1616471377,call,1579,
```

If you want, I could log in to your system (either directly or via remote desktop program) and look at it.

closed 2021-03-28 by robotastic

rpdale, 2021-03-28:
Looking good!

[image]

rosecitytransit, 2021-03-28:
Awesome!

PR #441 (2021-03-28): Finished unit logging script (fixes #181, can close #440)
I'm pretty sure this works. I have not built trunk-recorder with with the capability to call this script yet, however I did test out the script by giving it the same parameters trunk-recorder should.

Finishes #181, can close #440
commit a4e5cd28 (2021-03-28): Merge pull request #441 from rosecitytransit/master

Finished unit logging script (fixes #181, can close #440)