TrunkRecorder/trunk-recorder #450: docker armv7 - [ERROR] avahi_client_new() failed: Daemon not running

hayden-t, 2021-03-27:
First let me say thanks for this, also that im new to docker and trunk recorder, coming from op25 boatbod fork. 

ive set up a config.json which is being loaded and it seems to recognize my rtl.

but it only gets as far as

[ERROR] avahi_client_new() failed: Daemon not running
recorder: client.c:785: avahi_client_get_state: Assertion `client' failed.

not sure what it means or where to go next.

robotastic, 2021-05-05:
This looks like it is a general Docker issue. The problem is that gr-osmosdr package pulls in Soapy, which added a component called Soapy Remote. Soapy Remote dies if it can't connect to an avahi server (local DNS stuff). One option looks to be passing the paths to the avahi daemon that is running most host boxes. Adding these dir mappings to the docker command and the privlege flag seemed to do it:
`docker run -it   --privileged  -v /home/ubuntu/trunk-recorder-docker:/app  -v /var/run/dbus:/var/run/dbus -v /var/run/avahi-daemon/socket:/var/run/avahi-daemon/socket --device /dev/bus/usb robotastic/trunk-recorder:latest`

The other is to `apt remove soapysdr0.7-module-remote` as part of the Dockerfile.

@robbiet480  do you have thoughts on the best approach?

robbiet480, 2021-05-06:
Probably better to just remove the package

robotastic, 2021-05-16:
I think we maybe stuck with it since gr-osmosdr depends on soapy, which pulls in this module.... I did make some changes to the Dockerfile to set the user back to root. It seems to fix the problems around file mode privilege for access the SDR. I have been testing with the Ubuntu 21.04 Pi Image. I will make some updates to the Wiki

robotastic, 2021-05-20:
This should be fixed with the latest version. I will update some of the info on the wiki

closed 2021-05-20 by robotastic

hayden-t, 2021-09-05:
Hey, finally having a go at this again, surprised i hadn't till now considering your feedback and suggestions, thanks ! it seems your command above containing privileged gets past the error and might work ! I just have to test it some more to find out as ive never been this far in TR. where in the wiki is what you added ?

hayden-t, 2021-09-05:
ooh, had a look again and found it https://github.com/robotastic/trunk-recorder/wiki/Installation#docker

hayden-t, 2021-09-05:
so it seems fine now, can lock onto a control channel. now i can continue to set it up further.

hayden-t, 2021-09-05:
hey i found it was creating recordings owned as root (now privileged) which was a pain . i found you can use this command to make them as user instead --user "$(id -u):$(id -g)"  - i updated the wiki

hayden-t, 2022-02-21:
also not, avahi-daemon has to be installed and running, on host for docker

gokuors, 2022-12-14:
Adding the relevant voumes and devices from https://github.com/robotastic/trunk-recorder/issues/450#issuecomment-832369120 helped me.!