TrunkRecorder/trunk-recorder #454: refactor config parsing

x0ul, 2021-04-02:
Hey there, very cool project you've got happening here :)

I'd like to try out contributing a little to a public open source project and this one lines up with my interests somewhat. It's not something I've really done before, so let me know if I'm doing this wrong!

Since I'm just getting started I only have a really superficial idea of how the system design works, but one thing I can see is that `main.cc` has >400 lines of code for parsing the `config.json` file. IMO this config-related code should be pulled out into its own compilation unit with a single focus on parsing config and presenting it to the rest of the program.

Let me know if this is like a great idea, or a terrible idea.

robotastic, 2021-04-07:
Hi! Thanks for reaching out. Agreed! The config parsing is a beast, and it would be great to move it out of main.cc. I went down the path of doing this and got halfway through. That is way there is a config struct: https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/config.h

I did have it as a Class with parsing, but part of the parsing involves populating some of the high level data structures, so it became a bit of a mess. It is probably do able, but there a lot of landmines lurking. If you see a way, that would be awesome.

x0ul, 2021-04-20:
An update: I haven't forgotten about this, I've actually got something half-written that's looking alright. I've been preoccupied with business at my job lately, but will have something to show in a couple weeks.

robotastic, 2021-05-16:
Thanks @x0ul - I am going to close this Issue for now. Feel free to drop a PR as you get code working and we can discuss there. Or reopen this issue if you have questions.

closed 2021-05-16 by robotastic