TrunkRecorder/trunk-recorder #455: New trying to install Trunk Recorder with Docker Desktop on a Mac

CrewRest, 2021-04-02:
Hi all, I'm new to Docker Desktop and haven't had much programming experience put can sometime muddle my way through. I downloaded and installed Docker Desktop and it's working. When I try to create Trunk Recorder I get the following;

Last login: Fri Apr  2 15:20:35 on ttys000
MyMac@Me-MBP ~ % docker run -it \
  -v /home/pi/trunk-recorder/media:/app/media \
  -v /home/pi/trunk-recorder:/app/config \
  -v /home/pi/trunk-recorder/my_talkgroup.csv:/app/my_talkgroup.csv \
  --device /dev/bus/usb robotastic/trunk-recorder:latest
docker: Error response from daemon: Mounts denied: 
The path /home/pi/trunk-recorder/media is not shared from the host and is not known to Docker.
You can configure shared paths from Docker -> Preferences... -> Resources -> File Sharing.
See https://docs.docker.com/docker-for-mac for more info.
ERRO[0000] error waiting for container: context canceled 
MyMac@Me-MBP ~ % 

What I'm I not doing. It appears that docker isn't pulling an image from a Docker repository.

robotastic, 2021-04-03:
So -unfortunately, Docker on the Mac does not support passing through USB devices. I think there is some way to do it by using a VM, installing Docker on the Linux VM and then running it there.

You may want to just install it directly on the Mac.

CrewRest, 2021-04-03:
OK I'll give a try! Thanks!

CrewRest, 2021-04-04:
Still no good! I installed Ubuntu 16.x in a VM on a MAC running Big Sur. Get an error right off the batt

sudo apt install gnuradio gnuradio-dev libuhd-dev libgnuradio-uhd3.7.11
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libganuradioi-uhd3.7.11
E: Couldn't find any package by glob 'libgnuradioi-uhd3.7.11'
E: Couldn't find any package by regex 'libgnuradioi-uhd3.7.11'

What up?

JustApu, 2021-04-04:
Ubuntu 16 — try libgnuradio-uhd3.7.9 instead.

https://packages.ubuntu.com/search?keywords=uhd

CrewRest, 2021-04-04:
@robotastic I should have been more verbose in my post about trying to install in a VM. My Mac has an intel processor and Docker cannot run in a VM unless the processor is and AMD (at least not an intel) so I'm doing the longer way. The recommendation by @JustApu to use 3.7.9 is working for that step. On to the next :)

CrewRest, 2021-04-04:
@JustApu that worked. Now on the last step I get "E: Unable to locate package libccpunt-1.14-0"

What do you suggest for this one?

JustApu, 2021-04-04:
Same Ubuntu package search: https://packages.ubuntu.com/search?keywords=libcppunit

CrewRest, 2021-04-04:
@JustApu tried 1.13.2-2.1 from the search and same error

JustApu, 2021-04-04:
libcppunit-1.13-0v5 ?

CrewRest, 2021-04-04:
@JustApu progress is slow but progress non the same. 1.13-0v5 worked but now it can't find libcurl4

I searched for that and discovered their is a package for libcul4 but not for ver16. there is a libcurl4-openssl-dev for ver16. What would happen if I just took out that library from the line? or would there be another substitute for it?

CrewRest, 2021-04-04:
@JustApu I tried all the using different the different libcurl4's that were there but got errors. I think I'll try using Ubuntu 18 and see how that goes.

CrewRest, 2021-04-04:
@JustApu With Ubuntu 18 I've gotten to the Build Stage so your guidance has been great! Thanks, hopefully my success will continue!

CrewRest, 2021-04-04:
@robotastic Hi Luke! In the last part of the Build section
make -j 'nproc'  returns the following
make: *** No rule to make target 'proc'. Stop.

I tried using -k as shown in the not below the -j line and had the same result. I then tried cmake which seemed to work but at the 61% point I got c++: Internal compiler error: Killed (program cciplus)

any ideas?

CrewRest, 2021-04-04:
@robotastic so! I was able to get near the end of the make command before it failed. It had 
-- All libraries:
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GROSMOSDR_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /Users/emmetdandersonjr/trunk-recorder
   used as include directory in directory /Users/emmetdandersonjr/trunk-recorder
   used as include directory in directory /Users/emmetdandersonjr/trunk-recorder
   used as include directory in directory /Users/emmetdandersonjr/trunk-recorder
   used as include directory in directory /Users/emmetdandersonjr/trunk-recorder
   used as include directory in directory /Users/emmetdandersonjr/trunk-recorder
   used as include directory in directory /Users/emmetdandersonjr/trunk-recorder
   used as include directory in directory /Users/emmetdandersonjr/trunk-recorder/lib/op25_repeater
   used as include directory in directory /Users/emmetdandersonjr/trunk-recorder

robotastic, 2021-05-05:
Hmm....  I think the CMake file in Trunk Recorder maybe expecting the latest version of GnuRadio. It should be backwards compatible, but maybe it isn't. I would try using Ubuntu 20.04. I have things working fine on that. I am not sure why the Wiki says it doesn't work. I can try to dig into that.... but you should be good with that. Also - the Docker images should work fine on your machine. It says AMD64, but that actually means it should work with most modern machines.

robotastic, 2021-05-20:
Feel to re-open this if there are still problems.

closed 2021-05-20 by robotastic