TrunkRecorder/trunk-recorder #459: Build fails on macOS 11.2.3/gnuradio 3.9.0.0_3/cmake 3.20.1

bdurrow, 2021-04-28:
I used brew to install the dependencies as documented in the wiki but the build fails.  I think all the debugging information you might need is below.

```
# git show | cat
commit b0e96e44bd1fb2896c55d56ee40ad58e6a03c157
Merge: d104d64 5177bce
Author: Luke Berndt <luke@robotastic.com>
Date:   Tue Apr 6 21:42:54 2021 -0400

    Merge pull request #453 from x0ul/log-it

    allow specifying log file directory

# git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
# brew upgrade gnuradio uhd gr-osmosdr cmake pkgconfig cppunit openssl fdk-aac-encoder sox
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 4 taps (homebrew/cask-versions, homebrew/core, homebrew/cask and homebrew/cask-drivers).
==> Updated Formulae
Updated 8 formulae.
==> Updated Casks
Updated 29 casks.

Warning: Treating cmake as a formula. For the cask, use homebrew/cask/cmake
Warning: gnuradio 3.9.0.0_3 already installed
Warning: uhd 4.0.0.0_2 already installed
Warning: gr-osmosdr 0.2.3 already installed
Warning: cmake 3.20.1 already installed
Warning: pkgconfig 0.29.2_3 already installed
Warning: cppunit 1.15.1 already installed
Warning: openssl 1.1.1k already installed
Warning: fdk-aac-encoder 1.0.1 already installed
Warning: sox 14.4.2_3 already installed
# cmake --version
cmake version 3.20.1

CMake suite maintained and supported by Kitware (kitware.com
# cmake ../trunk-recorder -DOPENSSL_ROOT_DIR=/usr/local/opt/openssl
CMake Deprecation Warning at CMakeLists.txt:24 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Build type not specified: defaulting to release.
CMake Warning (dev) at /usr/local/Cellar/cmake/3.20.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (GnuradioRuntime).  This can
  lead to problems in calling code that expects `find_package` result
  variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.20.1/share/cmake/Modules/FindPkgConfig.cmake:70 (find_package_handle_standard_args)
  cmake/Modules/FindGnuradioRuntime.cmake:1 (INCLUDE)
  CMakeLists.txt:69 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/local/Cellar/cmake/3.20.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args`
  (GNURADIO_RUNTIME) does not match the name of the calling package
  (GnuradioRuntime).  This can lead to problems in calling code that expects
  `find_package` result variables (e.g., `_FOUND`) to follow a certain
  pattern.
Call Stack (most recent call first):
  cmake/Modules/FindGnuradioRuntime.cmake:36 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:69 (find_package)
[... 235 lines cut ...]
    ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:510:5: note: candidate template ignored: could not match 'unique_ptr' against 'shared_ptr'
    shared_ptr( std::unique_ptr< Y, D > && r ): px( r.get() ), pn()
    ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:526:5: note: candidate template ignored: could not match 'unique_ptr' against 'shared_ptr'
    shared_ptr( boost::movelib::unique_ptr< Y, D > r ): px( r.get() ), pn()
    ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:637:5: note: candidate template ignored: could not match 'boost::shared_ptr' against 'std::__1::shared_ptr'
    shared_ptr( shared_ptr<Y> && r, typename boost::detail::sp_enable_if_convertible<Y,T>::type = boost::detail::sp_empty() )
    ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:371:14: note: explicit constructor is not a candidate
    explicit shared_ptr( Y * p ): px( p ), pn() // Y must be complete
             ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:423:14: note: explicit constructor is not a candidate
    explicit shared_ptr( weak_ptr<Y> const & r ): pn( r.pn ) // may throw
             ^
/usr/local/include/boost/smart_ptr/shared_ptr.hpp:465:14: note: explicit constructor is not a candidate
    explicit shared_ptr( std::auto_ptr<Y> & r ): px(r.get()), pn()
             ^
1 error generated.
make[2]: *** [lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/ambe_encoder_sb_impl.cc.o] Error 1
make[1]: *** [lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/all] Error 2
make: *** [all] Error 2
#
```

chuot, 2021-06-10:
FYI, I'm hitting the exact same issue with Fedora 34

robotastic, 2021-09-02:
I will go give this a check. Do you know what version of Gnuradio is being used? I wonder if there could also be a change there.

haicenhacks, 2021-09-05:
I am also having an issue with compiling on Arch.

`/trunk-recorder/lib/op25_repeater/lib/ambe_encoder_sb_impl.cc:48:9: error: could not convert 'gnuradio::get_initial_sptr(T*) [with T = gr::op25_repeater::ambe_encoder_sb_impl]()' from 'std::shared_ptr<gr::op25_repeater::ambe_encoder_sb_impl>' to 'gr::op25_repeater::ambe_encoder_sb::sptr' {aka  boost::shared_ptr<gr::op25_repeater::ambe_encoder_sb>'}
   47 |       return gnuradio::get_initial_sptr
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~
   48 |         (new ambe_encoder_sb_impl(verbose_flag));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |
      |         std::shared_ptr<gr::op25_repeater::ambe_encoder_sb_impl>
make[2]: *** [lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/build.make:76: lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/ambe_encoder_sb_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:177: lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/all] Error 2`

I have tried downloading a few different releases as well as the current git repository, all with the same result.

The version of gnuradio installed is 3.9.2.0-3

toddejohnson, 2021-09-06:
Likely has something to do with https://github.com/boatbod/op25/issues/114

Sadly on ArchLinux they went 3.9 as an update AUR package not building with the same error as @haicenhacks

robotastic, 2021-09-15:
I have had a chance to test with gnuradio 3.9 yet, but I did improve the compile process for Ubuntu 21.04. Can you give the branch `4.0-beta` a try?

mldulaney, 2021-09-16:
Also hitting this with Fedora 35 Beta:

/home/mock/trunk/trunk-recorder-3.3.7/lib/op25_repeater/lib/dmr_bs_tx_bb_impl.cc: In static member function ‘static gr::op25_repeater::dmr_bs_tx_bb::sptr gr::op25_repeater::dmr_bs_tx_bb::make(int, const char*)’:
/home/mock/trunk/trunk-recorder-3.3.7/lib/op25_repeater/lib/dmr_bs_tx_bb_impl.cc:174:9: error: could not convert ‘gnuradio::get_initial_sptr(T*) [with T = gr::op25_repeater::dmr_bs_tx_bb_impl]()’ from ‘std::shared_ptr<gr::op25_repeater::dmr_bs_tx_bb_impl>’ to ‘gr::op25_repeater::dmr_bs_tx_bb::sptr’ {aka ‘boost::shared_ptr<gr::op25_repeater::dmr_bs_tx_bb>’}
  173 |       return gnuradio::get_initial_sptr
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~
  174 |         (new dmr_bs_tx_bb_impl(verbose_flag, config_file));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |
      |         std::shared_ptr<gr::op25_repeater::dmr_bs_tx_bb_impl>
make[2]: *** [lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/build.make:90: lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/dmr_bs_tx_bb_impl.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/mock/trunk/trunk-recorder-3.3.7/lib/op25_repeater/lib/ambe_encoder_sb_impl.cc: In static member function ‘static gr::op25_repeater::ambe_encoder_sb::sptr gr::op25_repeater::ambe_encoder_sb::make(int)’:
/home/mock/trunk/trunk-recorder-3.3.7/lib/op25_repeater/lib/ambe_encoder_sb_impl.cc:48:9: error: could not convert ‘gnuradio::get_initial_sptr(T*) [with T = gr::op25_repeater::ambe_encoder_sb_impl]()’ from ‘std::shared_ptr<gr::op25_repeater::ambe_encoder_sb_impl>’ to ‘gr::op25_repeater::ambe_encoder_sb::sptr’ {aka ‘boost::shared_ptr<gr::op25_repeater::ambe_encoder_sb>’}
   47 |       return gnuradio::get_initial_sptr
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~
   48 |         (new ambe_encoder_sb_impl(verbose_flag));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |
      |         std::shared_ptr<gr::op25_repeater::ambe_encoder_sb_impl>
make[2]: *** [lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/build.make:76: lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/ambe_encoder_sb_impl.cc.o] Error 1
/home/mock/trunk/trunk-recorder-3.3.7/lib/op25_repeater/lib/ysf_tx_sb_impl.cc: In static member function ‘static gr::op25_repeater::ysf_tx_sb::sptr gr::op25_repeater::ysf_tx_sb::make(int, const char*, bool)’:
/home/mock/trunk/trunk-recorder-3.3.7/lib/op25_repeater/lib/ysf_tx_sb_impl.cc:238:9: error: could not convert ‘gnuradio::get_initial_sptr(T*) [with T = gr::op25_repeater::ysf_tx_sb_impl]()’ from ‘std::shared_ptr<gr::op25_repeater::ysf_tx_sb_impl>’ to ‘gr::op25_repeater::ysf_tx_sb::sptr’ {aka ‘boost::shared_ptr<gr::op25_repeater::ysf_tx_sb>’}
  237 |       return gnuradio::get_initial_sptr
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~
  238 |         (new ysf_tx_sb_impl(verbose_flag, config_file, fullrate_mode));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |
      |         std::shared_ptr<gr::op25_repeater::ysf_tx_sb_impl>
make[2]: *** [lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/build.make:104: lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/ysf_tx_sb_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:177: lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

mepholic, 2021-09-17:
> I have had a chance to test with gnuradio 3.9 yet, but I did improve the compile process for Ubuntu 21.04. Can you give the branch `4.0-beta` a try?

Hey @robotastic, today I just made the critical mistake of _updating Arch_ and I realized that gnuradio updated when I did that, so I tried rebuilding against `master`, and saw similar issues to the users above. I found this thread and took your advice about building `4.0-beta`, which just finished compiling successfully. Is there anything I need to do to update my configuration from 3.3 before I restart trunk-recorder?

I see your commit adding gnuradio 3.9 support is from yesterday. Thank you for staying on top of this stuff!

robotastic, 2021-09-18:
@mepholic You are in uncharted lands! I got things to compile... but I wasn't able to do any testing. You shouldn't need to update your config file to use v4.0, but let me know if something goes sideways.

toddejohnson, 2021-09-18:
I'll work on porting the Arch Linux aur PKGBUILD to 4.0-beta tonight. I'll also post the list of packages to downgrade if you need. I just need to pull that history

mldulaney, 2021-09-18:
I should mention that my failure to build was on the 4.0beta branch.

toddejohnson, 2021-09-18:
Ok if you are on arch and need to downgrade these are the packages I  downgraded:
```
trunk-recorder-3.1.4-1-x86_64.pkg.tar.zst
gnuradio-3.8.2.0-1-x86_64.pkg.tar.zst
gnuradio-3.8.2.0-1-x86_64.pkg.tar.zst
libuhd-4.0.0.0-1-x86_64.pkg.tar.zst
libvolk-2.4.1-1-x86_64.pkg.tar.zst
gnuradio-iqbal-0.38.1-4-x86_64.pkg.tar.zst
soapysdr-0.7.2-3-x86_64.pkg.tar.zst
airspy-1.0.9-2-x86_64.pkg.tar.zst
boost-1.75.0-3-x86_64.pkg.tar.zst
Can be found at https://archive.archlinux.org/packages
```

I'm making progress with an aur package based upon branch 4.0beta. Just need to get the install working... Perhaps another hour? This take a bit to compile in my ramdisk chroot. I'm looking forward to updating my recorder!

toddejohnson, 2021-09-18:
I've just finished submitting [trunk-recorder-git to the aur](https://aur.archlinux.org/packages/trunk-recorder-git) and for me it is building in a clean chroot. Now for me to run it:

It almost starts up:
```
Decim: 100 Decim2:  4
[2021-09-18 00:04:43.361657] (info)      P25 Trunking two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 10000000
[2021-09-18 00:04:43.367310] (info)      P25 Trunking ARB - Initial Rate: 10000000 Resampled Rate: 25000 Initial Decimation: 100 System Rate: 24000 ARB Rate: 0.96
Allocating 15 zero-copy buffers
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::reference = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::reference = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```
I might just have an unclean install after my downgrade and I can work on that this weekend along with testing my config with docker/podman. 

Perhaps I got something [wrong in the build or dependencies that changed](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=trunk-recorder-git)?

robotastic, 2021-09-18:
This is the Dockerfile I was using to test if I could build under the latest version of Arch. I didn't get a chance to run it because I did it on my Mac and Docker doesn't support USB pass through on MacOS. I will move it over to my Linux box and test it there. There could be some subtle GR3.8 ->3.9 changes I missed
https://github.com/robotastic/trunk-recorder/blob/4.0-beta/utils/Dockerfile.arch-latest.dev

toddejohnson, 2021-09-18:
@robotastic Good Idea I pulled docker.io/robotastic/trunk-recorder:4.0-beta to verify I didn't have something else wrong. Had an issue about plugins so in the /app folder. I'll report that but after copying them. Been recording all night. Your dockerfile gave me an Idea instead of reinstalling arch I could just test again in docker. I'm not good at dockerfiles just yet and I should likely provide a pr one for Arch AUR pkg for your utils folder. 

```sh
useradd -m build
echo "build ALL=(ALL) NOPASSWD: ALL" > /etc/sudoers.d/build
pacman -Sy
pacman -S --needed base-devel git
su - build
git clone https://aur.archlinux.org/trunk-recorder-git.git
makepkg -sri
cd /app
recorder --config /app/config.json
```

Same output
```
[2021-09-18 10:22:52.593041] (info)   Creating decoder sink...
[2021-09-18 10:22:52.594592] (info)   Decoder sink created!
[2021-09-18 10:22:52.595436] (info)   [SYS]    2C      TG:                       - (         3)        Freq: 451.000000 MHz    Starting Analog Recorder Num [22]       Squelch: -40
[2021-09-18 10:22:52.595669] (info)   [SYS]    TG:                       - (         3)        Freq: 451.000000 MHz
Allocating 15 zero-copy buffers
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::reference = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```

So I'm trying with your dockerfile from the utils folder and  it worked! Well missing sox and fdkaac but it starts recording without segfault. 
```sh
docker build -t tr-arch -f utils/Dockerfile.arch-latest.dev  .
docker run -it --privileged -e TZ=America/Chicago --user 0:0 -v $(pwd)/app:/app -v $(pwd):/src --tmpfs /app/media tr-arch /bin/bash
mkdir /trunk-build
cd /trunk-build/
cmake /src
make
./recorder --config /app/config.json 
```

So feel free to ignore my comments about my package I can work on that separate and this can go back to focus on Mac. Sorry @bdurrow for hijacking your thread.

robotastic, 2024-12-14:
Reopen if this is still an issue

closed 2024-12-14 by robotastic