TrunkRecorder/trunk-recorder #461: Docker usb permissions

jp83, 2021-04-30:
Opening a separate issue for tracking and informing @robbiet480 .  I can't get docker usb permissions to work with the latest Docker images. I had my own really old Docker build based on previous image that I overlaid and copied in my own config a year or two ago, but something underlying must have changed. I previously had --privileged -v /dev/bus/usb:/dev/bus/usb. I've tried --device /dev/bus/usb and --device-cgroup-rule 'a : rwm'. I keep getting the usual:

Using device #0 Generic RTL2832U OEM
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules

I understand from the FAQ to git rtl-sdr and build it, but shouldn't that be built into the image from Dockerfile? What am I missing? I happen to be running on a small thin client machine with Rancher OS, lightweight operating system. For reference I just got openwebrx docker image running on my setup with my rtl-sdr out-of-the-box like so...

docker run -d \
  --restart=always \
  --name openwebrx \
  --device /dev/bus/usb \
  -p 8073:8073 \
  -v $PWD/_data:/etc/openwebrx/ \
  --tmpfs=/tmp/openwebrx \
  jketterl/openwebrx:stable

Of course I stopped and removed it to free up the usb before coming back to this.  A little more debug...  I tried to append the rtl-sdr in my own Dockerfile to build upon the one here in this repo, realized it didn't have permissions due to USER nobody ( https://github.com/robotastic/trunk-recorder/blob/b0e96e44bd1fb2896c55d56ee40ad58e6a03c157/Dockerfile#L44 ).  The Dockerfile structure is different for openwebrx but don't see a USER command.  For lack of anything else I added "USER root" and then realized I probably don't even need rtl-sdr.

FROM robotastic/trunk-recorder:latest
USER root
#RUN apt-get install -y rtl-sdr && apt install -y gr-osmosdr
CMD ["./recorder","--config=/app/config.json"]

And now this works (doesn't without --privileged)

docker run --rm -it -v $PWD/config/config.json:/app/config.json --privileged -v /dev/bus/usb:/dev/bus/usb trunk-recorder

This actually does too...

docker run --rm -it -v $PWD/config/config.json:/app/config.json --device /dev/bus/usb trunk-recorder

So I believe the culprit is the "USER nobody", I can appreciate this is good practice, and perhaps I'm missing something fundamental in my setup, but it doesn't work out of the box.  Should permissions be opened up before change user or something?  Thanks.

jp83, 2021-04-30:
Ok, thank you!  That seemed to do the trick for me.  Since there's a variety of different examples here's my docker-compose.yaml for reference.

```
version: '3'
services:
  recorder:
    image: robotastic/trunk-recorder:latest
    container_name: trunk-recorder
    restart: always
    user: root:root
    devices:
      - /dev/bus/usb:/dev/bus/usb
    volumes:
      - ./media:/app/media
      - ./config/config.json:/app/config.json
      - ./ChanList.csv:/app/ChanList.csv
  rdio:
    image: chuot/rdio-scanner:latest
    container_name: rdio-scanner
    restart: always
    ports:
      - "3000:3000"
    volumes:
      - ./rdio-scanner:/app/data
      - ./media:/app/data/media

```

robotastic, 2021-05-05:
I am seeing the same thing. I originally thought the issue was that a modeprobe.d blacklist was needed to block the wrong drivers from being loaded... but it turns out the only reason that worked is that I switched the user back to root.

Should we just make the User Root? Is some of the problem the user mapping? 

It doesn't look like the udev rules get installed, we could try doing that:

`sudo cp ../rtl-sdr.rules /etc/udev/rules.d/`

https://github.com/keenerd/rtl-sdr/blob/master/rtl-sdr.rules

robotastic, 2021-05-16:
I changed the user back to **root** in the dockerfile. It seems like that is the most robust way to make the docker image play well with SDRs.

closed 2021-05-16 by robotastic